SpikeInterface/spikeinterface #3594: about plot waveforms

ohkoibln, 2024-12-20:
I already had my sorting results  unit_1_waveforms.npy,unit_2_waveforms.npy .How could I use the sw.plot_unit_waveforms(sorting_analyzer, unit_ids=unit_ids, figsize=(16, 4)) directly?

ohkoibln, 2024-12-20:
> 您好，我们尝试解决了这个问题。
> 
> 我们是这样做的：
> 
> 修改了 UnitWaveformsWidget，以接受预先计算的波形作为输入。这允许用户直接将 plot_unit_waveforms 与自己的波形文件一起使用，而无需 SortingAnalyzer 或 Templates 对象。
> 
> 您可以在此提交中查看更改：[wwqwq2313@c2f8448](https://github.com/wwqwq2313/SpikeInterface-spikeinterface-3594/commit/c2f8448dc5c51594ab2a65402b2d463706b2870c)。
> 
> 谨慎
> 
> 免责声明：解决方案的概念是由 AI 创建的，在检查生成的代码的正确性之前，切勿复制粘贴此代码。解决方案可能不完整，您应该仅将此代码用作灵感。
> 
> 如果您不再希望 Latta AI 尝试解决存储库中的问题，则可以阻止此账户。
Thanks, I will try!

ohkoibln, 2024-12-20:
could I draw waveforms of all units in a figure

ohkoibln, 2024-12-20:
> You need to load the sorting analyzer with the `si.load_sorting_analyzer()` function. Then you can use the `sw.plot_unit_waveforms(sorting_analyzer)`, which will plot all units!

ok, I also want to plot some units with different colors in one axis，not in different axises.

ohkoibln, 2024-12-21:
> load_sorting_analyzer()

Thanks a lot!

zm711, 2025-03-13:
It seems like this has been sorted out. I'll close but open more issues if needed :)

closed 2025-03-13 by zm711