SpikeInterface/spikeinterface #3595: create_sorting_analyzer  fail, I used tridesclous,spikeinterface Version: 0.101.2

didi226, 2024-12-29:
```
tridesclous_para={'sorter_name':'tridesclous','freq_min': 300,'freq_max':6000, 'n_jobs': 1,
							  'detect_sign': -1,'detect_threshold':3 * np.std(signals)}
sorting = run_sorter( recording=recording_preprocessed, verbose=True,
									  folder=folder,
									  **tridesclous_para)
	analyzer=create_sorting_analyzer(recording_preprocessed, sorting, folder=save_sorting_wave_fold, sparse=False,return_scaled=False)
```
error:
```
  File "C:\software\Anaconda\envs\py3.9\lib\site-packages\spikeinterface\core\sortinganalyzer.py", line 322, in create
    all_probes = recording.get_probegroup().probes
AttributeError: 'TridesclousSortingExtractor' object has no attribute 'get_probegroup'
```

chrishalcrow, 2025-01-06:
Hey @didi226 . Your `recording` and `sorting` are the wrong way around in the `create_sorting_analyzer`. So you should have:

```
analyzer=create_sorting_analyzer(sorting, recording_preprocessed, folder=save_sorting_wave_fold, sparse=False,return_scaled=False)
```

The safest way to avoids this error (which I've also made many times!) is to always explicitly include the keywords, e.g.:

```
analyzer=create_sorting_analyzer(recording=recording_preprocessed, sorting=sorting, folder=save_sorting_wave_fold, sparse=False,return_scaled=False)
```

Let me know if that fixes it

didi226, 2025-01-08:
Yes, you are right. I made a mistake. Thank you for your reply.

closed 2025-01-08 by chrishalcrow