SpikeInterface/spikeinterface #3596: Help me understand why my unit locations don't make sense

fruce-ki, 2025-01-03:
Hello and happy new year!

Let me preface my problem by saying that I am not a neuroscientist and do not have a background in signal processing. I am a bioinformatician trying to set up a pipeline for my lab.

I am using spike interface 0.101.2 to interface between Maxwell Two data and spyking-circus 1 & 2.

After sorting (with either version of spyking circus) and creating the analyzer, I want to assess the locations of the sorted neurons on the sensor array. However, all the neurons (up to a 1000 per well) are plotted right on top of each other in just one location or at most in a handful of locations. This happens with both the built-in report, and with custom export and plot, with both sorter versions, with two of the three localisation methods (the third doesn't put them so much on top of each other, but their distribution is still suspicious and unlikely), through two entirely independent pipelines made by separate people, and for all the wells across multiple plates.

This can't be right. Even physically impossible. I'd expect neurons to be spread out into a single-layer network over the array. Microscopy and the spread of selected electrodes suggests that they are indeed spread out, and we do get normal looking recordings. So the inferred unit locations do not represent the cell cultures. The question is why.

Am I executing it wrong? Are my parameters way off? Is the sorter unable to sort network bursts? Something else? I feel completely out of my depth...

I am including my code to run with spyking circus 2, omitting some  parts that deal with file logistics. Any help is appreciated!

```
#!/user/bin/env python

### Developed with python 3.12.8

import os
#import sys
import subprocess
import argparse
import re
import numpy as np

import spikeinterface as si
import spikeinterface.extractors as sixtr
#import spikeinterface.preprocessing as sipre
import spikeinterface.sorters as siso
import spikeinterface.qualitymetrics as siqm
import spikeinterface.exporters as sixpo

#import generate_mapping as mgm      ## the function throws an error if used from import

def main(args):

	## Options
	parser = argparse.ArgumentParser(description="Extract spike-sorted data.")
	parser.add_argument("--dataPath", required=False, type=str, default="/SCRATCH/mea/data/", help="Base path for raw recordings.")
	parser.add_argument("--experiment", required=True, type=str, help="Experiment name (== data subdirectory).")
	parser.add_argument("--measurement", required=True, type=str, help="Usually date & condition.")
	parser.add_argument("--plate", required=True, type=str, help="Plate ID.")
	parser.add_argument("--intermediatePath", required=False, type=str, default="Network/", help="Fixed sub-dir in the plate.")
	parser.add_argument("--recording", required=True, type=str, help="6-digit recording id.")
	parser.add_argument("--sortedPath", required=False, type=str, default="/SCRATCH/mea/sorted/", help="Base path for sorted recordings.")
	parser.add_argument("--analysisPath", required=False, type=str, default="/SCRATCH/mea/analysis/", help="Base path for analysis results.")

	parser.add_argument("--numofWells", required=False, type=int, default=None, help="6 or 24.")
	parser.add_argument("--subsample", required=False, type=int, default=1000, help="Number of units to subset for calculations.")

	parser.add_argument("--gm_path", required=False, type=str, default='/BIOINFORMATICS/GH_repos/MEA_ghrepo/generate_mapping.py', help="Maxwell's well map script.")
	parser.add_argument("--hdf5", required=False, type=str, default='/SCRATCH/REFERENCES/maxwell_plugin', help="Maxwell HDF5 plugin.") # /BIOINFORMATICS/GH_repos/MEA_ghrepo/maxwell_plugin

	parser.add_argument("--cores", required=False, type=int, default=1, help="CPU cores.")
	parser.add_argument("--memory", required=False, type=str, default='280G', help="RAM.")
	

	# Hard-code some values for easy interactive development
	args = ['sort.py',
		'--experiment', 'DoseRes',
		'--measurement', '240910_7W',
		'--plate', 'M07100',
		'--recording', '000022'
	]

	params = parser.parse_args(args[1:])

	os.environ['HDF5_PLUGIN_PATH'] = params.hdf5
	global_job_kwargs = dict(n_jobs = params.cores, total_memory = params.memory, mp_context = 'spawn', progress_bar = False)
[... 113 lines cut ...]
		analyzer.compute("correlograms", window_ms=100, bin_ms=5)
		analyzer.compute("template_metrics", include_multi_channel_metrics = False,
								metric_kwargs = {"recovery_window_ms" : 0.7, "peak_relative_threshold" : 0.2, "peak_width_ms" : 0.2,
												"min_channels_for_velocity" : 5, "min_r2_velocity" : 0.7, "exp_peak_function" : "ptp",
												"min_r2_exp_decay" : 0.5, "spread_threshold" : 0.2, "spread_smooth_um" : 20})
		analyzer.compute("principal_components", n_components=3, mode="by_channel_local", whiten=True) 
		try:
			analyzer.compute("amplitude_scalings") 
		#except LinAlgError:     ## Don't know where LinAlgError is defined, clearly not in my current imports.
		except: 
			raise Warning("An error, possibly LinAlgError due to expected matrix being singular, occurred. Computation of Amplitude Scalings aborted.")
			## I have no idea how to prevent it or how to fix it. I don't even know if it is a useful metric for us. For now I'm passing the ball downstream, to the interpretation of any metrics.
		analyzer.compute("spike_locations", ms_before = 0.5, method = "monopolar_triangulation")
		analyzer.compute("template_similarity", method = "cosine", max_lag_ms = 0, support = "union")
		analyzer.compute("quality_metrics", skip_pc_metrics = False)
		

		
		## Extract analyzer metrics
		sixpo.export_report(analyzer, os.path.join(params.analysisPath, params.experiment, w + '_sorting_report'),
						remove_if_exists=True, format='pdf', show_figures=False)

if __name__ == '__main__':
    sys.exit(main())
```

alejoe91, 2025-01-03:
Hi! Happy new year to you too!

Can you show us some of the plots?

samuelgarcia, 2025-01-06:
Hi.
@cwindolf will be our best expert on this.

Basically when the method used is "monopolar_trianglulation" you can have strange result if the waveform is not 'smotth" in space. In that case you can use "grid_convolution" or "center_of_mass" to have something that "look more normal".

fruce-ki, 2025-01-07:
Thank you all for responding!

Here are two examples of the center of mass localisation, from the built-in report.
[unit_locations.pdf](https://github.com/user-attachments/files/18329575/unit_locations.pdf)
[unit_locations.pdf](https://github.com/user-attachments/files/18329576/unit_locations.pdf)

I don't have any template plots at hand. I need to look them up in the docs first and I will reply as soon as I have something.

alejoe91, 2025-01-07:
Thanks. Yeah definitely something looks off... can you try to `sw.plot_unit_templates(analyzer, backend="ipywidgets")` from a notebook and check how the units look? 

One possibility is that the electrode that all units are found on uis broken and thus very noisy, and this shadows the "real" signal from other electrodes. 

In this case, you should perform a bad channel detection/removal step prior to spike sorting and downstream analysis.

fruce-ki, 2025-01-07:
That would mean that every well on every electrode has one or two very noisy electrodes?

Thank you for suggesting the plotting command, that saved me some digging. Unfortunately it produces an error : 

`"name": "ValueError",
	"message": "Key backend: 'module://ipympl.backend_nbagg' is not a valid value for backend; supported values are ['gtk3agg', 'gtk3cairo', 'gtk4agg', 'gtk4cairo', 'macosx', 'nbagg', 'notebook', 'qtagg', 'qtcairo', 'qt5agg', 'qt5cairo', 'tkagg', 'tkcairo', 'webagg', 'wx', 'wxagg', 'wxcairo', 'agg', 'cairo', 'pdf', 'pgf', 'ps', 'svg', 'template', 'inline']"`

If I drop the backend parameter it kinda works, but the plots I get are illegible. Too many panels, too much overflowing text.

alejoe91, 2025-01-07:
I think the error is due to an old jupyter installation. You can plot a subset of units with the `unit_ids=[...]` argument

fruce-ki, 2025-01-07:
Could be. I'm using python 3.9.21 for this environment as it is the only one where all the dependencies are met for both spykingcircus1 and spikeinterface.

Even for my inexperienced eyes, I think it is safe to say that these are not what spikes should look like...

[image]
[image]
[image]

Do you have any recommendations on what pre-processing to apply?

alejoe91, 2025-01-07:
yep, that's for sure! You should also check the raw data to make sure you do have spikes. It might be that the culture/slice you're testing is just "dead". Do you see many individual spikes in the Maxwell software? 

Maybe you could share a session with us. @yger is an absolute expert here :) He will also recommend to use SpykingCircus2 instead of SC1, since the latter is not maintained anymore.

fruce-ki, 2025-01-07:
This data is from SC2. 

I am only implicating SC1 because all of the data prior to my involvement was with SC1 and troubleshooting an issue when multiple parameters were changed at once is not the way to go.

In general the Maxwell software shows spikes and network bursts that we consider good, at least for some of the cultures. Some wells do have problems, but this localisation issue has occurred on all the wells and all the plates that I analyzed, so that certainly includes good cultures.

I have some meetings now, I'll look into the channels later.

yger, 2025-01-07:
Hi. Jumping into the discussion, I would suspect that you have something strange with the data and/or preprocessing. Can you share some small file (even a temporal slice) with me?

fruce-ki, 2025-01-07:
It's getting late today,I'll get back to you tomorrow with hopefully a slice (I don't know how to do it off the top of my head, so I need a bit of time).

With regards to pre-processing and processing, I am not doing anything outside the code that I showed at the beginning. Which is to say, I am not doing any pre-processing at the current stage of development. I think there are some preprocessing parameters integrated in the sorter execution though... I went with the default values or what was shown in doc examples.

PS. I verified with the lab and the wells I am using as examples definitely have network-bursting neuron cultures according to the Maxwell software. However, it occurred to me that the first 10 channels of each well that I subset and showed may not necessarily be the most representative of the plate... I'll try to pick some other subsets from a known good well.

alejoe91, 2025-01-08:
@fruce-ki I'm thinking of another reason why you're getting weird results.

While it's true that each sorter will apply its own preprocessing, you are then constructing the analyzer using the raw data. The Maxwell could experience slow drifts over time and that will completely mess up postprocessing. I recommend adding at least a bandpass filter to the recording before constructing the analyzer. Can you try that out?

yger, 2025-01-08:
Arg yes, could be an option. SC2 will preprocess the recording with a bandpass filter + CMR + Whitening. I think the recording is saved internally in the sorting folder and could be reloaded if needed, but that is what you should give to the analyzer if you want to properly visualize what has been found by SC2. Otherwise, as @alejoe91  is saying, maybe some drifts and/or offset can mess up everything

alejoe91, 2025-01-08:
I would suggest just trying to add a bandpass for now to isolate the issue :) SC2 has some custom processing (e.g. whitening) which is great for spike sorting performance, but it could be less interpretable for postprocessing

fruce-ki, 2025-01-09:
I believe you guys nailed it with the unsorted/sorted mismatch! Things are looking up!

I've added this pre-processing step (please, feel free to recommend improvements, I am still not a signal specialist):

```
recording = sipre.bandpass_filter(recording, freq_min = 100.0, freq_max = 3500, margin_ms = 5.0, dtype = None)

bad_ids = sipre.detect_bad_channels(recording, method = 'coherence+psd', std_mad_threshold = 5, psd_hf_threshold = 0.02, dead_channel_threshold = -0.5, noisy_channel_threshold = 1.0, outside_channel_threshold = -0.75, outside_channels_location = 'top', n_neighbors = 11, nyquist_threshold = 0.8, direction = 'y', chunk_duration_s = 0.3, num_random_chunks = 100, welch_window_ms = 10.0, neighborhood_r2_threshold = 0.9, neighborhood_r2_radius_um = 30.0, seed = None)

recording.remove_channels(bad_ids)	
	
recording = sipre.correct_motion(recording, preset = 'dredge_fast', folder = None, output_motion = False, output_motion_info = False)
```

and changed my sorting prameters to this (again, suggestions are welcome):

```
joblist = [
	{
		'sorter_name': 'spykingcircus2', 
		'recording': recordings[wells[i]], 
		'output_folder': os.path.join(params.sortedPath, params.experiment, 'spykingcircus2', well_ids[wells[i]]),
		'remove_existing_folder' : True,
		'verbose' : False,
		'raise_error' : False,
		'general' : {'ms_before': 2, 'ms_after': 2, 'radius_um': 100}, 
		'sparsity' : {'method': 'snr', 'amplitude_mode': 'peak_to_peak', 'threshold': 0.25}, 
		'detection' : {'peak_sign': 'neg', 'detect_threshold': 4}, 
		'selection' : {'method': 'uniform', 'n_peaks_per_channel': 5000, 'min_n_peaks': 10000, 'select_per_channel': False}, 
		'apply_motion_correction' : False, 
		'merging' : {'similarity_kwargs': {'method': 'cosine', 'support': 'union', 'max_lag_ms': 0.2}, 
		'correlograms_kwargs': {}, 
		'auto_merge': {'min_spikes': 10, 'corr_diff_thresh': 0.25}}, 
		'clustering' : {'legacy': True}, 
		'apply_preprocessing' : False, 
		'matched_filtering' : False, 
		'cache_preprocessing' : {'mode': 'memory', 'memory_limit': 0.9, 'delete_cache': True}, 
		'multi_units_only' : False, 
		'job_kwargs' : {'n_jobs': 1}, 
		'debug' : True
	}
	for i in list(range(0, len(wells)))
]

sortings = siso.run_sorter_jobs(job_list=joblist, engine='joblib', return_output = False)
```

and the localisations are starting to look like this, which looks very plausible:

[unit_locations.pdf](https://github.com/user-attachments/files/18363057/unit_locations.pdf)

Thank you so much for your help!

fruce-ki, 2025-01-10:
I guess this issue can be considered closed.

closed 2025-01-10 by fruce-ki