SpikeInterface/spikeinterface #3599: How to save sorted spikes time stamp in spikeinterface

venkatbits, 2025-01-08:
Hi All

I have searched extensively in the spike interface documentation and web but unable to get this information on how to save all the sorted spike times hence raising this issue

I am running the following commands on my data

sorting_spycir2 = ss.run_sorter(sorter_name="spykingcircus2", recording=recording_seg, output_folder="C:/Users//Desktop/spike/folder_spykingcircus2_all_chanels") 

folder = ' C:/Users//Desktop/spike/waveforms_spycir2_all'
we_spycir2_all= si.extract_waveforms(recording_seg, sorting_spycir2, folder, load_if_exists=None,  ms_before=1, ms_after=2., max_spikes_per_unit=500, n_jobs=1, chunk_size=30000)
print(we_spycir2_all)

sorting_analyzer = si.create_sorting_analyzer(sorting=sorting_spycir2, recording=recording_seg)
sorting_analyzer.compute(['random_spikes', 'waveforms', 'templates', 'noise_levels'])  
sorting_analyzer.compute(['spike_locations'])

export_report(sorting_analyzer=sorting_analyzer, output_folder='C:/Users/Desktop/ spike  /folder_spykingcircus_allchanels_report')
In the above generated report I could not find the time stamps of each detected spike

I want to save all the detected/sorted spikes time stamps. Can you kindly help with this
 thanks 
Venkat

yger, 2025-01-08:
the time/sorted spikes are stored in the sorting objects. You can get them as numpy arrays by doing

sorting_spycir2.get_all_spike_trains()

venkatbits, 2025-01-08:
Thanks very much, it worked, I am sorry to bother can you kindly help me with how to save this as a text or excel sheet I am new to python

zm711, 2025-01-08:
Does it have to be an text or csv. You can save the sorting object and then you'll always have the spike times. Are you planning to switch to a different programming environment?

venkatbits, 2025-01-08:
Hi Zach

Can you please tell me how to save it as text or csv as well as saving it
as a sorting object .
*Are you planning to switch to a different programming environment? *
No wanted to send it to a collaborator who can read only text or excel files

Thanks very much

-- 
The information contained in this electronic communication is intended 
solely for the individual(s) or entity to which it is addressed. It may 
contain proprietary, confidential and/or legally privileged information. 
Any review, retransmission, dissemination, printing, copying or other use 
of, or taking any action in reliance on the contents of this information by 
person(s) or entities other than the intended recipient is strictly 
prohibited and may be unlawful. If you have received this communication in 
error, please notify us by responding to this email or telephone and 
immediately and permanently delete all copies of this message and any 
attachments from your system(s). The contents of this message do not 
necessarily represent the views or policies of BITS Pilani.

zm711, 2025-01-08:
To save it as a sorting object all you have to do is type:

```python
sorting.save(xx)
```
where the xx will be the format you want and the location to save it to.

For the excel or text how do you need the data organized? The easiest would honestly be just a couple columns of the label and the spike time and then you could index into the excel to get the spike train for each neuron. Again that would require some programming experience. 

I guess the other way would be to save each neuron as a separate column/row of an excel/txt file. This would be super messy and not as storage friendly. What is the experience level of your collaborators? Or how do they want to interact with the data?

venkatbits, 2025-01-09:
Dear Zach

Thanks very much I will give details here

I ran spykingcircus2, now first I want to determine whether the algorithm
has detected the spikes in the data, for this purpose for example as given
below my collaborator has marked a spike in the raw data, now I want to
give him the spikes detected for channel 4 by the algorithm along with
their spike timings so that he can determine along the channel 4 data
whether the algorithm can detect all the spikes he is looking for. For this
purpose can you help me in saving the results in a excel sheet so that he
can open the excel sheet and see these details, I am new to python and my
collaborator is comfortable with excel only

[image: image.png]

-- 
The information contained in this electronic communication is intended 
solely for the individual(s) or entity to which it is addressed. It may 
contain proprietary, confidential and/or legally privileged information. 
Any review, retransmission, dissemination, printing, copying or other use 
of, or taking any action in reliance on the contents of this information by 
person(s) or entities other than the intended recipient is strictly 
prohibited and may be unlawful. If you have received this communication in 
error, please notify us by responding to this email or telephone and 
immediately and permanently delete all copies of this message and any 
attachments from your system(s). The contents of this message do not 
necessarily represent the views or policies of BITS Pilani.

zm711, 2025-01-09:
@venkatbits thanks for that info. It actually changes a lot.

So at a fundamental level a sorting contains two vectors of information-- the spike times (ie when the spikes occurred) and the spike labels (we call them unit ids in spikeinterface, but some people prefer cluster ids or calling them neuron ids). Our sorting object also has other information like segment info (which doesn't matter for monosegment things) as well as some metadata. What we don't explicitly have is "channel 4". This is because one unit/neuron can generate spikes on multiple channels. So if you're just counting spikes on channel 4 and on channel 3 then they will likely be some of the same neurons and so you would be double counting spikes unfairly (unless you channels are super isolated). 

So our strategy is to generate unit locations based on one of three computational strategies during post processing to give you the location of your unit. But this isn't spikes/channel. Rather the unit location is one more piece of information to go along with each unit it. (I can share docs on how to use our analyzer if you are interested).

But in your case if you care about spikes/channel then we really need to ask why? Are you're channels completed isolated such that each channel can't "see" what is on other channels? In this case you don't even necessarily need to spike sort. You could just threshold the data yourself (or use our peak detection tools). 

Or rather than just think about spikes along channel four we could just give you all the spike times and then your collaborator can look at his spike times and see which ones match? So you are trying to manually validate SC2? 

Sorry we have to ask so many questions but how we organize the data for your collaborator really depends on what you want to do with the data.

zm711, 2025-01-09:
yeah of course. @samuelgarcia or @alejoe91 will `detect_peaks` work on one channel only? Or @yger do you have any opinions on whether SC2 will actually work with monotrode data?

In this case though your current sorting object should all be on channel 4, so if you want to save whatever you do have then you could take the spike_vector with:

```python

import pandas as pd

# this collects the information as two long balanced vectors
spike_vector = sorting.to_spike_vector()
spike_times = spike_vector["sample_index"]
spike_indices = spike_vector["unit_index"]

# this makes sure the numbers are coordinated between the sorting object and the excel
spike_unit_ids = np.array([sorting.unit_ids[unit] for unit in spike_indices])

# dataframes are similar to excels
df = pd.DataFrame({"Spike Times": spike_times, "Unit Id": spike_unit_ids})
# this will create a csv, but you will need to fill out the location and name to fit with what you want
# for example name of file and location. See the Pandas documentation for arguments you need to add
df.to_csv() 
```

venkatbits, 2025-01-09:
Dear Zach

Many thanks again

Can you also suggest to me which of the sorting algorithms is suitable for
single channel data as you have rightly mentioned SC2 failed to detect any
spikes when I used only  channel 4

Thanks for the script I will check it tomorrow morning when I go to work
and will let you know

Thanks very much again

Venkat

-- 
The information contained in this electronic communication is intended 
solely for the individual(s) or entity to which it is addressed. It may 
contain proprietary, confidential and/or legally privileged information. 
Any review, retransmission, dissemination, printing, copying or other use 
of, or taking any action in reliance on the contents of this information by 
person(s) or entities other than the intended recipient is strictly 
prohibited and may be unlawful. If you have received this communication in 
error, please notify us by responding to this email or telephone and 
immediately and permanently delete all copies of this message and any 
attachments from your system(s). The contents of this message do not 
necessarily represent the views or policies of BITS Pilani.

jakeswann1, 2025-01-11:
I have used the `detect_peaks()` function with the `'by_channel'` method to look at all detected peaks channel-by-channel in the past, to get a rough sense for how much a spike sorter might be missing, then created a sorting object from this and exported to phy to do some super manual cluster cutting, where essentially each channels detected peaks were treated as a 'unit' initially. This seemed to work well for that particular purpose, and I can share some code if that would be useful

venkatbits, 2025-01-11:
Dear Jake

Thanks so much, if you can share the code it will be great

Thanks a lot for your help

Venkat

-- 
The information contained in this electronic communication is intended 
solely for the individual(s) or entity to which it is addressed. It may 
contain proprietary, confidential and/or legally privileged information. 
Any review, retransmission, dissemination, printing, copying or other use 
of, or taking any action in reliance on the contents of this information by 
person(s) or entities other than the intended recipient is strictly 
prohibited and may be unlawful. If you have received this communication in 
error, please notify us by responding to this email or telephone and 
immediately and permanently delete all copies of this message and any 
attachments from your system(s). The contents of this message do not 
necessarily represent the views or policies of BITS Pilani.

samuelgarcia, 2025-01-13:
Yes detect_peaks works with one unique channel.

And this should be faster
```python
unit_indices = spike_vector["unit_index"]
spike_unit_ids = sorting.unit_ids[unit_indices]
```

venkatbits, 2025-01-13:
Dear Gracia

Thanks a lot

Venkat

-- 
The information contained in this electronic communication is intended 
solely for the individual(s) or entity to which it is addressed. It may 
contain proprietary, confidential and/or legally privileged information. 
Any review, retransmission, dissemination, printing, copying or other use 
of, or taking any action in reliance on the contents of this information by 
person(s) or entities other than the intended recipient is strictly 
prohibited and may be unlawful. If you have received this communication in 
error, please notify us by responding to this email or telephone and 
immediately and permanently delete all copies of this message and any 
attachments from your system(s). The contents of this message do not 
necessarily represent the views or policies of BITS Pilani.

zm711, 2025-02-05:
So it seems like we've answered this one. So I'll close but feel free to open new issues if they come up.

closed 2025-02-05 by zm711

venkatbits, 2025-02-06:
Dear Zach

Thanks very much for all your kind help, I am waiting for a reply from Jake
as he mentioned he will share his code using which I thought of verifying
the results. Also, when you use detect_peaks it looks like I cannot use
different algorithms like waveclus, ironclust, herdingspikes etc meaning is
there a possibility to specify "by channel" option in waveclus, ironclust,
herdingspikes for single channel spike sorting or am I completely wrong in
my understanding? Kindly correct me if I am wrong

Thanks very much again for all your kind help

-- 
The information contained in this electronic communication is intended 
solely for the individual(s) or entity to which it is addressed. It may 
contain proprietary, confidential and/or legally privileged information. 
Any review, retransmission, dissemination, printing, copying or other use 
of, or taking any action in reliance on the contents of this information by 
person(s) or entities other than the intended recipient is strictly 
prohibited and may be unlawful. If you have received this communication in 
error, please notify us by responding to this email or telephone and 
immediately and permanently delete all copies of this message and any 
attachments from your system(s). The contents of this message do not 
necessarily represent the views or policies of BITS Pilani.

venkatbits, 2025-02-06:
> For each sorter you would have to look that up yourself. In general most
> sorters do not give you an option to do something "by channel". But the
> best way to confirm that is ask the developers of the sorters since we only
> maintain the convenience wrappers for the sorters (except for TDC2 and
> SC2). So it is hard to say for sure.
>
> detect_peaks is an internal function written by the spikeinterface team.
> Its goal is to detect threshold crossing that could be extracellular
> potentials so it is not a spike sorter by itself but rather a component to
> build your own sorter. That's why it is in the sorting components module.
> It needs to be combined to make a full sorting pipeline. For your purposes
> however, you might not need a full sorter and so that's why it was
> suggested to try detect_peaks by itself. For low channel counts or
> isolated channels you might not need a fancy spike sorter you might just
> need to find your spikes. We don't have all of the docs for sorting
> components yet so it is a little harder to use, but we plan to do fuller
> docs soon!

Dear Zach

Thanks very much I will look into how to use the other sorting algorithms
for my single channel data

You can close this issue.

Thanks very much again

Venkat

-- 
The information contained in this electronic communication is intended 
solely for the individual(s) or entity to which it is addressed. It may 
contain proprietary, confidential and/or legally privileged information. 
Any review, retransmission, dissemination, printing, copying or other use 
of, or taking any action in reliance on the contents of this information by 
person(s) or entities other than the intended recipient is strictly 
prohibited and may be unlawful. If you have received this communication in 
error, please notify us by responding to this email or telephone and 
immediately and permanently delete all copies of this message and any 
attachments from your system(s). The contents of this message do not 
necessarily represent the views or policies of BITS Pilani.