SpikeInterface/spikeinterface #3602: NPX2.0 with OpenEphys files

castro-lab, 2025-01-08:
Hello, I see in the documentation an example of how to analyze NeuroPixels data recorded using spikeglx. Do you have an example using OpenEphys files? If not, any differences to consider vs the spikeglx code? 
Thanks

JoeZiminski, 2025-01-09:
Hi @castro-lab, you can use the [`read_openephys`](https://spikeinterface.readthedocs.io/en/latest/api.html#spikeinterface.extractors.read_openephys) function (instead of `read_spikeglx`) to read OpenEphys. Otherwise, the preprocessing / sorting function calls should be the same, let us know if you run into any problems.

castro-lab, 2025-01-10:
Hi @JoeZiminski, Thanks for the tip! As per the image, I could load it by specifying the 0-based block_index but not using the experiment_names. The folder has 4 experiments. How does the experiment_names work? 

[image]

zm711, 2025-01-13:
@castro-lab we probably need to see what the actual error is. Also are the names actually experiment1 etc. You would need to use the actual names to load them if they exist. block_index according to the doc also seems to load only one experiment at the time.

So basically could you post your stack trace for us so we can help you more :)

alejoe91, 2025-01-14:
I think it would be nice to add the experiment name as an annotation. I'll make a PR to add it :)

PR #3624 (2025-01-17): Add experiment_name annotation to open ephys
Fixes #3602
closed 2025-01-17 by alejoe91