SpikeInterface/spikeinterface #3603: Should we fail doc build on broken example moving forward?

zm711, 2025-01-09:
Just a discussion based on the fact @chrishalcrow recently had to fix the examples because of a change in generator. I think the project is big enough that we need to make sure our changes don't break the docs to keep things look as professional as possible. Should we make it so rtd does the red check if the examples fail to build rather than allow them to pass with failed examples to alert us to these changes at the time of the PR instead of later when one of us checks? ie whoever broke it should fix it at the time?

zm711, 2025-01-09:
I guess we could ask the devs how many people build locally. (I don't I just let the CI test it....) but I think maybe you and @JoeZiminski build locally most times first?

zm711, 2025-01-10:
@samuelgarcia I vaguely remember you being against this ages ago. Is that the case or am I wrong?

samuelgarcia, 2025-01-13:
I think I prefer that readthe doc build is building always.
We could have a github action testing the doc sphinx galery building, that would be faster that waiting the RTD send back the build is OK ofr not.

JoeZiminski, 2025-01-13:
I agree @zm711 this is a good idea, as a new user, nothing is worse than the documentation being incorrect - better to have no documentation than wrong documentation. Agree it looks unprofessional, and degrades trust in the package. 
It's very easy to introduce changes that lead to the doc examples becoming incorrect, the only way to protect against it is to ensure these are caught in CI before merge. 

I think regarding running locally, this should not affect things (?) as you can build the relevant documentation locally as we do now (i.e. build will encounter and flag errors but continue building), but in the CI RTD will fail build on error to ensure these cannot be merged until fixed. In an unfortunate case where someone breaks docs they cannot run locally, they can either a) use the CI for troubleshooting and fixing the issue b) request help from someone who can build locally if they are stuck. I think both are preferable to merging broken docs.

zm711, 2025-01-13:
I would be curious what @alejoe91 and @h-mayorquin think? Any strong opinions on your sides?

h-mayorquin, 2025-01-13:
The problem is that the docs take too long to build. But do they take longer than the full tests? If they do not then I guess it should be OK.

I am + 1 on this.

alejoe91, 2025-01-13:
I think that depends on the PR and the files changed. I'm actually ok in failing a PR if docs fail too. Let's discuss tomorrow

PR #3619 (2025-01-14): Switch RTD build to fail on gallery example failure
I included our options in a comment to make it easier to edit in the future.

Fixes #3603
closed 2025-01-14 by alejoe91