SpikeInterface/spikeinterface #3606: Flat ephys signal when loading data with read_openephys

slew000, 2025-01-10:
Hello,

I'm using spikeinterface to load and visualise OpenEphys data before spike sorting, but I'm having issues with the visualisation. I'm loading my files using auxrec = si.read_openephys(folder_path=file_path), with the file_path being this open ephys folder: 
[image]

When I try to plot the imported data using sw.plot_traces(auxrec, time_range=[0,10]), I can only see the TTL channel, but the ephys signal seems flat (I have checked in OE and there is data, it is not actually flat):

[image]

However, if I plot the data using using .get_traces() or if I apply auxrec.channel_slice(channel_ids=chan_ids), and I plot the data using the same functions, I can see the signal:
 
[image]

Does anyone have any idea on why the imported signal displays as flat?

Thanks a lot!

alejoe91, 2025-01-10:
Hi @slew000 

I think it's just a scaling issue in the visualization! If you try the `sw.plot_traces(auxrec, time_range=[0,10], channel_ids=chan_ids) you should get a similar plot :)

slew000, 2025-01-10:
> Hi @slew000
> 
> I think it's just a scaling issue in the visualization! If you try the `sw.plot_traces(auxrec, time_range=[0,10], channel_ids=chan_ids) you should get a similar plot :)

Hi Alejo, 

Thanks, that worked! However, I don't think I fully understand the scaling. In the flat plot, if I zoom in or out, I still don't see anything (?), whereas the TTL signal seems fine. (For reference, I took out the TTL channels of the chan_ids)

Thanks!

alejoe91, 2025-01-14:
I think that's simply because the ephys signals are in the uV scale, while the TTL signals are full range, so the events can be a few Volts

slew000, 2025-01-20:
That makes sense, thanks for the explanation! :)

zm711, 2025-02-05:
I'll close this since it seems like we figured it out. Feel free to reopen if the concern is still present :)