SpikeInterface/spikeinterface #3609: 'spikeinterface' is not a package

DavidLucha, 2025-01-13:
Hi,

I've installed spikeinterface on Windows 11 as per the instructions and I keep running into the same issue whenever I try to call any of the sub-modules. Regardless of the sub-module that I call, I get this error that 'spikeinterface' is not a package.

[image]

(spikeinterface) C:\Users\david>pip list
Package            Version      Editable project location
------------------ ------------ -----------------------------
asciitree          0.3.3
asttokens          3.0.0
colorama           0.4.6
comm               0.2.2
contourpy          1.3.1
cuda-python        12.6.2.post1
cycler             0.12.1
debugpy            1.8.11
decorator          5.1.1
distinctipy        1.3.4
exceptiongroup     1.2.2
executing          2.1.0
fasteners          0.19
fonttools          4.55.3
h5py               3.12.1
importlib_metadata 8.5.0
ipykernel          6.29.5
ipython            8.31.0
jedi               0.19.2
joblib             1.4.2
jupyter_client     8.6.3
jupyter_core       5.7.2
kiwisolver         1.4.8
llvmlite           0.43.0
matplotlib         3.10.0
matplotlib-inline  0.1.7
neo                0.13.4
nest_asyncio       1.6.0
networkx           3.4.2
numba              0.60.0
numcodecs          0.13.1
numpy              1.26.4
packaging          24.2
pandas             2.2.3
parso              0.8.4
pickleshare        0.7.5
pillow             11.1.0
pip                24.2
platformdirs       4.3.6
probeinterface     0.2.24
prompt_toolkit     3.0.48
psutil             6.1.1
pure_eval          0.2.3
Pygments           2.19.1
pyparsing          3.2.1
python-dateutil    2.9.0.post0
pytz               2024.2
pywin32            307
pyzmq              26.2.0
quantities         0.16.1
scikit-learn       1.6.1
scipy              1.15.1
setuptools         75.1.0
six                1.17.0
spikeinterface     0.102.0
stack_data         0.6.3
threadpoolctl      3.5.0
tornado            6.4.2
tqdm               4.67.1
traitlets          5.14.3
typing_extensions  4.12.2
tzdata             2024.2
wcwidth            0.2.13
wheel              0.44.0
zarr               2.17.2
zipp               3.21.0

Any help would be greatly appreciated.

samuelgarcia, 2025-01-13:
In python, you should have a local file name `spikeinterface.py` which is also a package name `spikeinterface`.
Because the import is using the file in the folder that is masking the package.

zm711, 2025-01-14:
Hey @DavidLucha were you able to test out Sam's advice to name your sciprt file something else?

DavidLucha, 2025-01-14:
Hi Zach and Sam,

I will be able to test this later tonight when I’m back in the office. I’ll
keep you posted once I’ve given it a go.

Thanks,

zm711, 2025-01-14:
Perfect. Thanks :) Good luck and definitely keep reaching out with any issues!

DavidLucha, 2025-01-14:
That seems to be working now. Thanks for the quick fix!

closed 2025-01-14 by DavidLucha