SpikeInterface/spikeinterface #3611: Suggestion to default to `use_binary_file=True` for KS4 sorting

guptadivyansh, 2025-01-13:
Continuing the discussion from the merged PR here..

The suggestion is to make change the default value for the parameter. 

 The flag `use_binary_file` makes a big difference to KS4 sorting performance. Writing the binary does take a while but overall I managed to sort a ~2 hour recording in ~4 hours with `use_binary_file = True` compared to ~22 hours with the default `use_binary_file = False` for the same recording. So, thank you for adding this param!

Workstation: Intel i7 8700K, Nvidia 3060 12GB, 64GB RAM, SSD

It would be good to hear if others are also seeing such a big performance difference. 

Although as @sameulgarcia mentioned, it is weird that KS4 is so slow when passed the SI recording internally. If someone can get to the bottom of that, then perhaps we could have the best of both worlds; fast sorting without explicitly writing to disk.

In the meantime, `use_binary_file = True` might be a better default

_Originally posted by @guptadivyansh in https://github.com/SpikeInterface/spikeinterface/issues/3339#issuecomment-2585290208_

zm711, 2025-01-13:
This was also found to be true for Mountainsort5 which started caching the recording (ie writing to disk) before sorting which leads to great speed-ups to the sorting itself. So we see this in multiple sorters. One hypothesis that may contribute in the case of KS4 is that although they use spikeinterface objects natively they do not make use of any of our multiprocessing. Maybe their binary support has speed-ups that they have not applied to our native objects. They directly say this in the comments of their code (the comment about not using spikeinterface multiprocessing for their wrapper of a recording).

alejoe91, 2025-01-13:
Given that this came up several times, I think we should have `use_binary_file` true by default. From Sam's comment https://github.com/SpikeInterface/spikeinterface/pull/3339#issuecomment-2586620184 he also agrees :)

zm711, 2025-01-13:
That's a really chill PR @guptadivyansh any interest in trying it yourself or do you want us to just deal with it?

guptadivyansh, 2025-01-13:
Sure, I'll give it a go! (_excited for first contribution_)

But first, do others also see such a big difference in performance between the two?

closed 2025-01-14 by alejoe91