SpikeInterface/spikeinterface #3618: `ValueError: codec not available: 'wavpack'`

a-zmz, 2025-01-14:
Hi,

I have used `wavpack` to compress my preprocessed recording:
```
    64 wv_compressor = WavPack(
    65     level=3, # high compression
    66     bps=None, # lossless
    67 )
```
and saved it to `.zarr`:

```
preprocessed.save(
    format="zarr",
    folder=output,
    compressor=wv_compressor,
)
```
and now i could load the preprocessed:
```
rec = si.load_extractor(preprocessed)
```
but when i try to sort:
```
# set ks4 parameters
ks4_params = {
             "do_correction": False, # do not motion correct
             "do_CAR": False, # do not common average reference
             "save_preprocessed_copy": True, # save ks4 preprocessed data
 }
 
# run sorter
sorting = ss.run_sorter(
             sorter_name='kilosort4',
             recording=rec,
             folder=output,
             singularity_image=True,
             remove_existing_folder=True,
             verbose=True,
             **ks4_params,
)
```

i got the following error:

```
SpikeSortingError: Spike sorting in singularity failed with the following error:                                                                                                                                                              
Traceback (most recent call last):                                                                                                                                                                                                            
  File "/home/amz/processed/npx/processed/20240722_az_VDCN09/in_container_sorter_script.py", line 13, in <module>                                                                                                                             
    recording = load_extractor(json_rec)                                                                                                                                                                                                      
                ^^^^^^^^^^^^^^^^^^^^^^^^                                                                                                                                                                                                      
  File "/home/amz/processed/npx/processed/20240722_az_VDCN09/in_container_python_base/lib/python3.11/site-packages/spikeinterface/core/base.py", line 1196, in load_extractor                                                                 
    return BaseExtractor.load(file_or_folder_or_dict, base_folder=base_folder)                                                                                                                                                                
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^                                                                                                                                                                
  File "/home/amz/processed/npx/processed/20240722_az_VDCN09/in_container_python_base/lib/python3.11/site-packages/spikeinterface/core/base.py", line 783, in load                                                                            
    extractor = BaseExtractor.from_dict(d, base_folder=base_folder)                                                                                                                                                                           
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^                                                                                                                                                                           
  File "/home/amz/processed/npx/processed/20240722_az_VDCN09/in_container_python_base/lib/python3.11/site-packages/spikeinterface/core/base.py", line 529, in from_dict                                                                       
    extractor = _load_extractor_from_dict(dictionary)                                                                                                                                                                                         
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^                                                                                                                                                                                         
  File "/home/amz/processed/npx/processed/20240722_az_VDCN09/in_container_python_base/lib/python3.11/site-packages/spikeinterface/core/base.py", line 1135, in _load_extractor_from_dict                                                      
    extractor = extractor_class(**new_kwargs)                                                                                                                                                                                                 
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^                                                                                                                                                                                                 
  File "/home/amz/processed/npx/processed/20240722_az_VDCN09/in_container_python_base/lib/python3.11/site-packages/spikeinterface/core/zarrextractors.py", line 67, in __init__                                                               
    dtype = self._root["traces_seg0"].dtype                                                                                                                                                                                                   
            ~~~~~~~~~~^^^^^^^^^^^^^^^                                                                                                                                                                                                         
  File "/home/amz/processed/npx/processed/20240722_az_VDCN09/in_container_python_base/lib/python3.11/site-packages/zarr/hierarchy.py", line 467, in __getitem__                                                                               
    return Array(                                                                                                                                                                                                                             
           ^^^^^^                                                                                                                                                                                                                             
  File "/home/amz/processed/npx/processed/20240722_az_VDCN09/in_container_python_base/lib/python3.11/site-packages/zarr/core.py", line 170, in __init__                                                                                       
    self._load_metadata()                                  
  File "/home/amz/processed/npx/processed/20240722_az_VDCN09/in_container_python_base/lib/python3.11/site-packages/zarr/core.py", line 193, in _load_metadata
    self._load_metadata_nosync()
  File "/home/amz/processed/npx/processed/20240722_az_VDCN09/in_container_python_base/lib/python3.11/site-packages/zarr/core.py", line 240, in _load_metadata_nosync
    self._compressor = get_codec(compressor)
                       ^^^^^^^^^^^^^^^^^^^^^
  File "/home/amz/processed/npx/processed/20240722_az_VDCN09/in_container_python_base/lib/python3.11/site-packages/numcodecs/registry.py", line 53, in get_codec
    raise ValueError(f'codec not available: {codec_id!r}')
ValueError: codec not available: 'wavpack'

> /home/amz/.conda/envs/ks4/lib/python3.11/site-packages/spikeinterface/sorters/runsorter.py(648)run_sorter_container()
    646     if run_error:                                  
    647         if raise_error:
--> 648             raise SpikeSortingError(f"Spike sorting in {mode} failed with the following error:\n{run_sorter_output}")
    649     else:                                          
    650         if with_output:
```

I have spikeinterface version `0.101.2` running in conda environment with python 3.11, os fedora 41.

Could anyone please help? Thanks very much!

zm711, 2025-01-14:
Does your container also have wavpack installed? You're doing this in a container so your container also needs to be able to handle zarr in this case.

a-zmz, 2025-01-14:
ah right, i did not explicitly install wavpack in the container. i use `singularity` and i only did `singularity_image=True`. could you please tell me how to do that? thanks!

alejoe91, 2025-01-14:
Yes. you can do this:
```
rec.extra_requirements.append("wavpack_numcodecs")
```

a-zmz, 2025-01-14:
> Yes. you can do this:
> 
> ```
> rec.extra_requirements.append("wavpack_numcodecs")
> ```
then i got:
```
SpikeSortingError: Spike sorting in singularity failed with the following error:                                                                                                                                           16:52:09 [15/23701]
Traceback (most recent call last):                                                                                                                                                                                                            
  File "/home/amz/processed/npx/processed/20240722_az_VDCN09/in_container_sorter_script.py", line 13, in <module>                                                                                                                             
    recording = load_extractor(json_rec)                                                                                                                                                                                                      
                ^^^^^^^^^^^^^^^^^^^^^^^^                                                                                                                                                                                                      
  File "/home/amz/processed/npx/processed/20240722_az_VDCN09/in_container_python_base/lib/python3.11/site-packages/spikeinterface/core/base.py", line 1196, in load_extractor                                                                 
    return BaseExtractor.load(file_or_folder_or_dict, base_folder=base_folder)                                                                                                                                                                
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^                                                                                                                                                                
  File "/home/amz/processed/npx/processed/20240722_az_VDCN09/in_container_python_base/lib/python3.11/site-packages/spikeinterface/core/base.py", line 783, in load                                                                            
    extractor = BaseExtractor.from_dict(d, base_folder=base_folder)                                                                                                                                                                           
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^                                                                                                                                                                           
  File "/home/amz/processed/npx/processed/20240722_az_VDCN09/in_container_python_base/lib/python3.11/site-packages/spikeinterface/core/base.py", line 529, in from_dict                                                                       
    extractor = _load_extractor_from_dict(dictionary)                                                                                                                                                                                         
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^                                                                                                                                                                                         
  File "/home/amz/processed/npx/processed/20240722_az_VDCN09/in_container_python_base/lib/python3.11/site-packages/spikeinterface/core/base.py", line 1135, in _load_extractor_from_dict                                                      
    extractor = extractor_class(**new_kwargs)                                                                                                                                                                                                 
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/amz/processed/npx/processed/20240722_az_VDCN09/in_container_python_base/lib/python3.11/site-packages/spikeinterface/core/zarrextractors.py", line 67, in __init__
    dtype = self._root["traces_seg0"].dtype
            ~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/home/amz/processed/npx/processed/20240722_az_VDCN09/in_container_python_base/lib/python3.11/site-packages/zarr/hierarchy.py", line 467, in __getitem__
    return Array(                                          
           ^^^^^^                                          
  File "/home/amz/processed/npx/processed/20240722_az_VDCN09/in_container_python_base/lib/python3.11/site-packages/zarr/core.py", line 170, in __init__
    self._load_metadata()                                  
  File "/home/amz/processed/npx/processed/20240722_az_VDCN09/in_container_python_base/lib/python3.11/site-packages/zarr/core.py", line 193, in _load_metadata
    self._load_metadata_nosync()
  File "/home/amz/processed/npx/processed/20240722_az_VDCN09/in_container_python_base/lib/python3.11/site-packages/zarr/core.py", line 240, in _load_metadata_nosync
    self._compressor = get_codec(compressor)
                       ^^^^^^^^^^^^^^^^^^^^^
  File "/home/amz/processed/npx/processed/20240722_az_VDCN09/in_container_python_base/lib/python3.11/site-packages/numcodecs/registry.py", line 49, in get_codec
    cls = entries[codec_id].load()
          ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/miniconda3/lib/python3.11/importlib/metadata/__init__.py", line 202, in load
[... cut ...]
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/amz/processed/npx/processed/20240722_az_VDCN09/in_container_python_base/lib/python3.11/site-packages/wavpack_numcodecs/__init__.py", line 1, in <module>
    from wavpack_numcodecs.wavpack import WavPack, wavpack_version
ImportError: libwavpack.so.1: cannot open shared object file: No such file or directory
```

a-zmz, 2025-01-14:
sorry but what do you mean by `start the singularity container externally`?

alejoe91, 2025-01-14:
I can test on my end ;)

alejoe91, 2025-01-14:
ok, this is a bit tricky. I tested locally with `singularity shell docker://spikeinterface/kilosort4-base:lates` and then ran `pip install wavpack-numcodecs` in it and got this error:
```
Singularity> pip install wavpack-numcodecs
Defaulting to user installation because normal site-packages is not writeable
Collecting wavpack-numcodecs
  Using cached wavpack-numcodecs-0.2.2.tar.gz (2.6 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: numpy in /home/miniconda3/lib/python3.11/site-packages (from wavpack-numcodecs) (1.26.3)
Collecting numcodecs (from wavpack-numcodecs)
  Using cached numcodecs-0.14.1-cp311-cp311-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (3.0 kB)
Requirement already satisfied: packaging in /home/miniconda3/lib/python3.11/site-packages (from wavpack-numcodecs) (23.1)
Using cached numcodecs-0.14.1-cp311-cp311-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (8.9 MB)
Building wheels for collected packages: wavpack-numcodecs
  Building wheel for wavpack-numcodecs (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for wavpack-numcodecs (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [197 lines of output]
      /tmp/pip-build-env-djfntjx4/overlay/lib/python3.11/site-packages/setuptools/config/_apply_pyprojecttoml.py:74: _MissingDynamic: `entry-points` defined outside of `pyproject.toml` is ignored.
      !!
      
              ********************************************************************************
              The following seems to be defined outside of `pyproject.toml`:
      
              `entry-points = {'numcodecs.codecs': ['wavpack = wavpack_numcodecs:WavPack']}`
      
              According to the spec (see the link below), however, setuptools CANNOT
              consider this value unless `entry-points` is listed as `dynamic`.
      
              https://packaging.python.org/en/latest/specifications/pyproject-toml/#declaring-project-metadata-the-project-table
      
              To prevent this problem, you can list `entry-points` under `dynamic` or alternatively
              remove the `[project]` table from your file and rely entirely on other means of
              configuration.
              ********************************************************************************
      
      !!
        _handle_missing_dynamic(dist, project_table)
      Using shipped libraries
      running bdist_wheel
      running build
      running build_py
      creating build/lib.linux-x86_64-cpython-311/wavpack_numcodecs
      copying src/wavpack_numcodecs/__init__.py -> build/lib.linux-x86_64-cpython-311/wavpack_numcodecs
      copying src/wavpack_numcodecs/globals.py -> build/lib.linux-x86_64-cpython-311/wavpack_numcodecs
      running egg_info
      writing src/wavpack_numcodecs.egg-info/PKG-INFO
      writing dependency_links to src/wavpack_numcodecs.egg-info/dependency_links.txt
      writing entry points to src/wavpack_numcodecs.egg-info/entry_points.txt
      writing requirements to src/wavpack_numcodecs.egg-info/requires.txt
      writing top-level names to src/wavpack_numcodecs.egg-info/top_level.txt
      reading manifest file 'src/wavpack_numcodecs.egg-info/SOURCES.txt'
      adding license file 'LICENSE'
      writing manifest file 'src/wavpack_numcodecs.egg-info/SOURCES.txt'
      /tmp/pip-build-env-djfntjx4/overlay/lib/python3.11/site-packages/setuptools/command/build_py.py:212: _Warning: Package 'wavpack_numcodecs.include.wavpack' is absent from the `packages` configuration.
      !!
      
              ********************************************************************************
              ############################
              # Package would be ignored #
              ############################
              Python recognizes 'wavpack_numcodecs.include.wavpack' as an importable package[^1],
              but it is absent from setuptools' `packages` configuration.
      
              This leads to an ambiguous overall configuration. If you want to distribute this
              package, please make sure that 'wavpack_numcodecs.include.wavpack' is explicitly added
              to the `packages` configuration field.
[... 133 lines cut ...]
      copying src/wavpack_numcodecs/libraries/5.7.0/linux-x86_64-glibc2.39/libwavpack.so -> build/lib.linux-x86_64-cpython-311/wavpack_numcodecs/libraries/5.7.0/linux-x86_64-glibc2.39
      creating build/lib.linux-x86_64-cpython-311/wavpack_numcodecs/libraries/5.7.0/windows-x86_32
      copying src/wavpack_numcodecs/libraries/5.7.0/windows-x86_32/wavpackdll.dll -> build/lib.linux-x86_64-cpython-311/wavpack_numcodecs/libraries/5.7.0/windows-x86_32
      copying src/wavpack_numcodecs/libraries/5.7.0/windows-x86_32/wavpackdll.lib -> build/lib.linux-x86_64-cpython-311/wavpack_numcodecs/libraries/5.7.0/windows-x86_32
      creating build/lib.linux-x86_64-cpython-311/wavpack_numcodecs/libraries/5.7.0/windows-x86_64
      copying src/wavpack_numcodecs/libraries/5.7.0/windows-x86_64/wavpackdll.dll -> build/lib.linux-x86_64-cpython-311/wavpack_numcodecs/libraries/5.7.0/windows-x86_64
      copying src/wavpack_numcodecs/libraries/5.7.0/windows-x86_64/wavpackdll.lib -> build/lib.linux-x86_64-cpython-311/wavpack_numcodecs/libraries/5.7.0/windows-x86_64
      creating build/lib.linux-x86_64-cpython-311/wavpack_numcodecs/src
[... cut ...]
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for wavpack-numcodecs
Failed to build wavpack-numcodecs
```

So the problem is that you need `gcc` installed. I think that the easiest option would be to make a Dockerfile that starts from the `spikeinterface:kilosort4-base` and installs `build-essentials` in it.

Do you have docker installed as well? The process would be much easier in this case, because you wouldn't need to push to dockerhub.

alejoe91, 2025-01-14:
@a-zmz here's a solution.

You can build a singularity image from scratch. Create the following `kilosort4_with_gcc` file:
```
Bootstrap: docker
From: spikeinterface/kilosort4-base:latest

%post
    apt update
    apt install -y build-essential
```
Then build the dingularity image with:
```
sudo singularity build kilosort4_with_gcc.img kilosort4_wigh_gcc
```
This will generate the following `kilosort4_with_gcc.img` file in your current location:

Then run spike sorting by specifying the local path to the sif file:
```
# run sorter
rec.extra_requirements.append("wavpack-numcodecs")
sorting = ss.run_sorter(
             sorter_name='kilosort4',
             recording=rec,
             folder=output,
             singularity_image="path-to-kilosort4_with_gcc.img",
             remove_existing_folder=True,
             verbose=True,
             **ks4_params,
)
```

I know it's a bit convoluted, but it works :)

a-zmz, 2025-01-14:
cool thanks very much, i will test now!

a-zmz, 2025-01-14:
what worked for me:

from the previously pulled `kilosort4-base.sif`, i created a sandbox:
`singularity build --sandbox ks4/ kilosort4-base.sif `
and went into the image:
` sudo singularity shell --writable ks4/`

then in the image, i did:
```
apt-get update
apt install -y build-essential
apt install wget
apt install -y gettext
wget "https://www.wavpack.com/wavpack-5.7.0.tar.bz2"
tar -xf wavpack-5.7.0.tar.bz2
cd wavpack-5.7.0
./configure
make install
pip install wavpack-numcodecs
pip install "spikeinterface[full]"
```

basically the same way of installing `wavpack` as you mentioned previously here: https://github.com/AllenNeuralDynamics/wavpack-numcodecs/issues/13.

then build the new `.sif` file:
`sudo singularity build ks4_with_wavpack.sif ks4/`,
and use it in `run_sorter`:
```
sorting = ss.run_sorter(
             sorter_name='kilosort4',
             recording=rec,
             folder=output,
             singularity_image="path-to-ks4_with_wavpack.sif",
             remove_existing_folder=True,
             verbose=True,
             **ks4_params,
)
```

thanks again!