SpikeInterface/spikeinterface #3625: incorrect unit amplitude and location after `spre.astype(recording, np.int16)`

a-zmz, 2025-01-16:
Hi,

after preprocessing with `dredge` motion correction, the recording data type became `float32`. i want to change dtype back to `int16` to save some space, so i did:
`int_rec = spre.astype(recording=recording, dtype=np.int16)`

however, after kilosort 4 and `export_report`:
- in `unit list.csv`, all units have `max_on_channel_id` being `imec0.ap#AP174`
- amplitudes are centred at 4000
- unit location are off

[image]

[unit list.csv](https://github.com/user-attachments/files/18440438/unit.list.csv)

[image]

but before changing dtype, they look like this:

[image]
[unit list.csv](https://github.com/user-attachments/files/18440459/unit.list.csv)
[image]

could someone please help? thanks!

samuelgarcia, 2025-01-17:
Hi.
I think you need to propagate 2 properties from the original recording after going back to int16
`"gain_to_uV"` and `"offset_to_uV"`.

They must have been lost in the correct motion preprocessing step or somwhere.

The amplitude alsod epend on theses 2 factor if they are present in the recording properties

a-zmz, 2025-01-17:
hi,

the preprocessed recording does have these two properties from the raw recording, and so does the int16 converted one.

alejoe91, 2025-01-17:
Could this be an overflow issue? Can you try to convert to `int32` instead and see how the locations look?

zm711, 2025-01-24:
@a-zmz were you able to check it with `int32` to see if it is an overflow. That is important for us to know if we need to explore this more or not :)

a-zmz, 2025-01-24:
Hi @zm711, I was gonna do it tmr, will post the results!

zm711, 2025-01-24:
Perfect! Thanks.

a-zmz, 2025-01-26:
hello, i can confirm that when converting to `int32`, amplitude and unit locations look correct. i also tried to convert again to `int16` after dtype being `int32`, the problem did not appear. i wonder why...

zm711, 2025-02-05:
It's tricky for me to imagine all the ways downcasting could mess things up and sometimes the behavior is different between OSs etc so maybe something like that. 

Happy it fixed it using a bigger dtype. I'll close for now!

closed 2025-02-05 by zm711