SpikeInterface/spikeinterface #3626: LFP channels vs AP channels

taningh86, 2025-01-17:
Hi,
This may be more of a general question and possibly more suited for Neuropixels slack channel. But, I figured I would still ask here because I am sure this question must have been raised in the past in the forum and also I am using spikeinterface to process LFP data.
Looks like LFP channels are different than AP channels. I haven't looked in depth on 'Why' but i could make sume asumptions why that would be. But If anyone knows about it in certain terms then please feel free to share. I use the code below to process LFP data and get  the LFP channel ID list. The ID's are long and seems to span multiple channels. Am I getting something wrong here or this is actually the case? if so, then why only these channels have LFP signals and not others.
`lfprec2 = si.bandpass_filter(
    raw_rec,
    freq_min=50,
    freq_max=150,

    margin_ms=1850.,
    filter_order=3,
    dtype="float32",
    add_reflect_padding=True,
)
lfprec2 = si.phase_shift(lfprec2)
lfprec2 = si.resample(lfprec2, resample_rate=150, margin_ms=100)

lfprec2 = si.directional_derivative(lfprec2, order=2, edge_order=1)
lfprec2 = si.average_across_direction(lfprec2)

print(lfprec2)`

`print(lfprec2.get_channel_ids())`

And below is the list of LFP channel it outputs:

`['imec0.lf#LF48-imec0.lf#LF49' 'imec0.lf#LF50-imec0.lf#LF51'
 'imec0.lf#LF52-imec0.lf#LF53' 'imec0.lf#LF54-imec0.lf#LF55'
 'imec0.lf#LF56-imec0.lf#LF57' 'imec0.lf#LF58-imec0.lf#LF59'
 'imec0.lf#LF60-imec0.lf#LF61' 'imec0.lf#LF62-imec0.lf#LF63'
 'imec0.lf#LF64-imec0.lf#LF65' 'imec0.lf#LF66-imec0.lf#LF67'
 'imec0.lf#LF68-imec0.lf#LF69' 'imec0.lf#LF70-imec0.lf#LF71'
 'imec0.lf#LF72-imec0.lf#LF73' 'imec0.lf#LF74-imec0.lf#LF75'
 'imec0.lf#LF76-imec0.lf#LF77' 'imec0.lf#LF78-imec0.lf#LF79'
 'imec0.lf#LF80-imec0.lf#LF81' 'imec0.lf#LF82-imec0.lf#LF83'
 'imec0.lf#LF84-imec0.lf#LF85' 'imec0.lf#LF86-imec0.lf#LF87'
 'imec0.lf#LF88-imec0.lf#LF89' 'imec0.lf#LF90-imec0.lf#LF91'
 'imec0.lf#LF92-imec0.lf#LF93' 'imec0.lf#LF94-imec0.lf#LF95'
 'imec0.lf#LF144-imec0.lf#LF145' 'imec0.lf#LF146-imec0.lf#LF147'
 'imec0.lf#LF148-imec0.lf#LF149' 'imec0.lf#LF150-imec0.lf#LF151'
 'imec0.lf#LF152-imec0.lf#LF153' 'imec0.lf#LF154-imec0.lf#LF155'
 'imec0.lf#LF156-imec0.lf#LF157' 'imec0.lf#LF158-imec0.lf#LF159'
 'imec0.lf#LF160-imec0.lf#LF161' 'imec0.lf#LF162-imec0.lf#LF163'
 'imec0.lf#LF164-imec0.lf#LF165' 'imec0.lf#LF166-imec0.lf#LF167'
 'imec0.lf#LF168-imec0.lf#LF169' 'imec0.lf#LF170-imec0.lf#LF171'
 'imec0.lf#LF172-imec0.lf#LF173' 'imec0.lf#LF174-imec0.lf#LF175'
 'imec0.lf#LF176-imec0.lf#LF177' 'imec0.lf#LF178-imec0.lf#LF179'
 'imec0.lf#LF180-imec0.lf#LF181' 'imec0.lf#LF182-imec0.lf#LF183'
 'imec0.lf#LF184-imec0.lf#LF185' 'imec0.lf#LF186-imec0.lf#LF187'
 'imec0.lf#LF188-imec0.lf#LF189' 'imec0.lf#LF190-imec0.lf#LF191'
 'imec0.lf#LF240-imec0.lf#LF241' 'imec0.lf#LF242-imec0.lf#LF243'
 'imec0.lf#LF244-imec0.lf#LF245' 'imec0.lf#LF246-imec0.lf#LF247'
 'imec0.lf#LF248-imec0.lf#LF249' 'imec0.lf#LF250-imec0.lf#LF251'
 'imec0.lf#LF252-imec0.lf#LF253' 'imec0.lf#LF254-imec0.lf#LF255'
 'imec0.lf#LF256-imec0.lf#LF257' 'imec0.lf#LF258-imec0.lf#LF259'
 'imec0.lf#LF260-imec0.lf#LF261' 'imec0.lf#LF262-imec0.lf#LF263'
 'imec0.lf#LF264-imec0.lf#LF265' 'imec0.lf#LF266-imec0.lf#LF267'
 'imec0.lf#LF268-imec0.lf#LF269' 'imec0.lf#LF270-imec0.lf#LF271'
 'imec0.lf#LF272-imec0.lf#LF273' 'imec0.lf#LF274-imec0.lf#LF275'
 'imec0.lf#LF276-imec0.lf#LF277' 'imec0.lf#LF278-imec0.lf#LF279'
 'imec0.lf#LF280-imec0.lf#LF281' 'imec0.lf#LF282-imec0.lf#LF283'
 'imec0.lf#LF284-imec0.lf#LF285' 'imec0.lf#LF286-imec0.lf#LF287'
 'imec0.lf#LF336-imec0.lf#LF337' 'imec0.lf#LF338-imec0.lf#LF339'
 'imec0.lf#LF340-imec0.lf#LF341' 'imec0.lf#LF342-imec0.lf#LF343'
 'imec0.lf#LF344-imec0.lf#LF345' 'imec0.lf#LF346-imec0.lf#LF347'
 'imec0.lf#LF348-imec0.lf#LF349' 'imec0.lf#LF350-imec0.lf#LF351'
 'imec0.lf#LF352-imec0.lf#LF353' 'imec0.lf#LF354-imec0.lf#LF355'
 'imec0.lf#LF356-imec0.lf#LF357' 'imec0.lf#LF358-imec0.lf#LF359'
 'imec0.lf#LF360-imec0.lf#LF361' 'imec0.lf#LF362-imec0.lf#LF363'
 'imec0.lf#LF364-imec0.lf#LF365' 'imec0.lf#LF366-imec0.lf#LF367'
 'imec0.lf#LF368-imec0.lf#LF369' 'imec0.lf#LF370-imec0.lf#LF371'
 'imec0.lf#LF372-imec0.lf#LF373' 'imec0.lf#LF374-imec0.lf#LF375'
 'imec0.lf#LF376-imec0.lf#LF377' 'imec0.lf#LF378-imec0.lf#LF379'
 'imec0.lf#LF380-imec0.lf#LF381' 'imec0.lf#LF382-imec0.lf#LF383'
 'imec0.lf#LF100-imec0.lf#LF101-imec0.lf#LF292-imec0.lf#LF293'
 'imec0.lf#LF102-imec0.lf#LF103-imec0.lf#LF294-imec0.lf#LF295'
 'imec0.lf#LF104-imec0.lf#LF105-imec0.lf#LF296-imec0.lf#LF297'
 'imec0.lf#LF106-imec0.lf#LF107-imec0.lf#LF298-imec0.lf#LF299'
 'imec0.lf#LF108-imec0.lf#LF109-imec0.lf#LF300-imec0.lf#LF301'
 'imec0.lf#LF110-imec0.lf#LF111-imec0.lf#LF302-imec0.lf#LF303'
 'imec0.lf#LF112-imec0.lf#LF113-imec0.lf#LF304-imec0.lf#LF305'
 'imec0.lf#LF114-imec0.lf#LF115-imec0.lf#LF306-imec0.lf#LF307'
 'imec0.lf#LF116-imec0.lf#LF117-imec0.lf#LF308-imec0.lf#LF309'
 'imec0.lf#LF118-imec0.lf#LF119-imec0.lf#LF310-imec0.lf#LF311'
 'imec0.lf#LF120-imec0.lf#LF121-imec0.lf#LF312-imec0.lf#LF313'
 'imec0.lf#LF122-imec0.lf#LF123-imec0.lf#LF314-imec0.lf#LF315'
 'imec0.lf#LF124-imec0.lf#LF125-imec0.lf#LF316-imec0.lf#LF317'
 'imec0.lf#LF126-imec0.lf#LF127-imec0.lf#LF318-imec0.lf#LF319'
 'imec0.lf#LF128-imec0.lf#LF129-imec0.lf#LF320-imec0.lf#LF321'
 'imec0.lf#LF130-imec0.lf#LF131-imec0.lf#LF322-imec0.lf#LF323'
 'imec0.lf#LF132-imec0.lf#LF133-imec0.lf#LF324-imec0.lf#LF325'
 'imec0.lf#LF134-imec0.lf#LF135-imec0.lf#LF326-imec0.lf#LF327'
 'imec0.lf#LF136-imec0.lf#LF137-imec0.lf#LF328-imec0.lf#LF329'
 'imec0.lf#LF138-imec0.lf#LF139-imec0.lf#LF330-imec0.lf#LF331'
 'imec0.lf#LF140-imec0.lf#LF141-imec0.lf#LF332-imec0.lf#LF333'
 'imec0.lf#LF142-imec0.lf#LF143-imec0.lf#LF334-imec0.lf#LF335'
 'imec0.lf#LF0-imec0.lf#LF1-imec0.lf#LF192-imec0.lf#LF193'
 'imec0.lf#LF2-imec0.lf#LF3-imec0.lf#LF194-imec0.lf#LF195'
 'imec0.lf#LF4-imec0.lf#LF5-imec0.lf#LF196-imec0.lf#LF197'
 'imec0.lf#LF6-imec0.lf#LF7-imec0.lf#LF198-imec0.lf#LF199'
 'imec0.lf#LF8-imec0.lf#LF9-imec0.lf#LF200-imec0.lf#LF201'
 'imec0.lf#LF10-imec0.lf#LF11-imec0.lf#LF202-imec0.lf#LF203'
 'imec0.lf#LF12-imec0.lf#LF13-imec0.lf#LF204-imec0.lf#LF205'
 'imec0.lf#LF14-imec0.lf#LF15-imec0.lf#LF206-imec0.lf#LF207'
 'imec0.lf#LF16-imec0.lf#LF17-imec0.lf#LF208-imec0.lf#LF209'
 'imec0.lf#LF18-imec0.lf#LF19-imec0.lf#LF210-imec0.lf#LF211'
 'imec0.lf#LF20-imec0.lf#LF21-imec0.lf#LF212-imec0.lf#LF213'
 'imec0.lf#LF22-imec0.lf#LF23-imec0.lf#LF214-imec0.lf#LF215'
 'imec0.lf#LF24-imec0.lf#LF25-imec0.lf#LF216-imec0.lf#LF217'
 'imec0.lf#LF26-imec0.lf#LF27-imec0.lf#LF218-imec0.lf#LF219'
 'imec0.lf#LF28-imec0.lf#LF29-imec0.lf#LF220-imec0.lf#LF221'
 'imec0.lf#LF30-imec0.lf#LF31-imec0.lf#LF222-imec0.lf#LF223'
 'imec0.lf#LF32-imec0.lf#LF33-imec0.lf#LF224-imec0.lf#LF225'
 'imec0.lf#LF34-imec0.lf#LF35-imec0.lf#LF226-imec0.lf#LF227'
 'imec0.lf#LF36-imec0.lf#LF37-imec0.lf#LF228-imec0.lf#LF229'
 'imec0.lf#LF38-imec0.lf#LF39-imec0.lf#LF230-imec0.lf#LF231'
 'imec0.lf#LF40-imec0.lf#LF41-imec0.lf#LF232-imec0.lf#LF233'
 'imec0.lf#LF42-imec0.lf#LF43-imec0.lf#LF234-imec0.lf#LF235'
 'imec0.lf#LF44-imec0.lf#LF45-imec0.lf#LF236-imec0.lf#LF237'
 'imec0.lf#LF46-imec0.lf#LF47-imec0.lf#LF238-imec0.lf#LF239'
 'imec0.lf#LF96-imec0.lf#LF97-imec0.lf#LF288-imec0.lf#LF289'
 'imec0.lf#LF98-imec0.lf#LF99-imec0.lf#LF290-imec0.lf#LF291']`

alejoe91, 2025-01-18:
Hi,

That's because of the average across direction, which combines multiple channels at the same depth before averaging. So the new channelmidnreflect which channels have been averaged.

taningh86, 2025-01-19:
Thanks @alejoe91. That makes sense and my assumptions were along the same line. One thing I am not understanding is the output I am getting seems to be invariant of channel or depth. In the two figures below I have used different channel groups but the output is the same. is something going wrong here?

[image]

[image]

For the first image I used `channel_ids=['imec0.lf#LF42-imec0.lf#LF43-imec0.lf#LF234-imec0.lf#LF235',
 'imec0.lf#LF44-imec0.lf#LF45-imec0.lf#LF236-imec0.lf#LF237']` and for the second image I use `channel_ids=['imec0.lf#LF48-imec0.lf#LF49', 'imec0.lf#LF50-imec0.lf#LF51']`. You can see that the two are in two extremes of locations of the probe but I am not sure why they have the same output. I would expect some difference in the two if not a complete contrast. Any ideas why this could be and if something is not correct here?

taningh86, 2025-01-19:
here's the full code I used
`lfprec2 = si.bandpass_filter(
    raw_rec,
    freq_min=10,
    freq_max=15,

    margin_ms=100.,
    filter_order=3,
    dtype="float32",
    add_reflect_padding=True,
)
lfprec2 = si.phase_shift(lfprec2)
lfprec2 = si.resample(lfprec2, resample_rate=150, margin_ms=100)

lfprec2 = si.directional_derivative(lfprec2, order=2, edge_order=1)
lfprec2 = si.average_across_direction(lfprec2)

print(lfprec2)`

`si.plot_traces(lfprec2, backend="matplotlib", mode="map", channel_ids=[ 'imec0.lf#LF52-imec0.lf#LF53', 'imec0.lf#LF382-imec0.lf#LF383'],clim=(-0.01, 0.01), time_range=(5,7))`

zm711, 2025-01-19:
Just ping this again tomorrow since it might fall off our radar otherwise :)

taningh86, 2025-01-20:
Will do. Sorry to bother you guys on a Sunday. My bad!!

taningh86, 2025-01-20:
Hi. Just pinging you guys. Please take a look when you guys get around this.

alejoe91, 2025-01-20:
@cwindolf can you take a look at this one?

cwindolf, 2025-01-20:
Sure, hi @taningh86 ! I'm a bit confused as well -- the `channel_ids=` lists that you have passed into plot_traces both have length 2, but there looks to be more than 2 channels of stuff going on in the images that got generated. I'm not sure yet what is going on there, but to help with figuring it out, could you post a couple more images and data?

 - Would you do a `si.plot_probe_map(raw_rec)` so that we can see the original probe geometry?
 - `si.plot_traces(raw_rec, time_range=(5, 7))` so that we can see the raw data on the full probe for that time range?
 - `si.plot_traces(lfprec2, time_range=(5, 7))` so that we can see the full processed data on that time range?
 - Lastly, I'm trying to understand if there could be something up with the channel IDs logic. To help me figure that out, could you paste the result of `lfprec2._main_ids.tolist()`?

Thanks!

taningh86, 2025-01-20:
Thnaks @cwindolf 
here's the probe map. For some reason the zoom function is not appearing on the output. I remember it was not like this. I am not sure how helpful it is for you. is there a way to get 

[image]

Here's the image for `time_range= (5,7)` for all channels

[image]

Here is the trace for lfprec2 for `time_range=(5,7)

[image]

And, here's the list of lfp channels in lfprec2
`['imec0.lf#LF48-imec0.lf#LF49',
 'imec0.lf#LF50-imec0.lf#LF51',
 'imec0.lf#LF52-imec0.lf#LF53',
 'imec0.lf#LF54-imec0.lf#LF55',
 'imec0.lf#LF56-imec0.lf#LF57',
 'imec0.lf#LF58-imec0.lf#LF59',
 'imec0.lf#LF60-imec0.lf#LF61',
 'imec0.lf#LF62-imec0.lf#LF63',
 'imec0.lf#LF64-imec0.lf#LF65',
 'imec0.lf#LF66-imec0.lf#LF67',
 'imec0.lf#LF68-imec0.lf#LF69',
 'imec0.lf#LF70-imec0.lf#LF71',
 'imec0.lf#LF72-imec0.lf#LF73',
 'imec0.lf#LF74-imec0.lf#LF75',
 'imec0.lf#LF76-imec0.lf#LF77',
 'imec0.lf#LF78-imec0.lf#LF79',
 'imec0.lf#LF80-imec0.lf#LF81',
 'imec0.lf#LF82-imec0.lf#LF83',
 'imec0.lf#LF84-imec0.lf#LF85',
 'imec0.lf#LF86-imec0.lf#LF87',
 'imec0.lf#LF88-imec0.lf#LF89',
 'imec0.lf#LF90-imec0.lf#LF91',
 'imec0.lf#LF92-imec0.lf#LF93',
 'imec0.lf#LF94-imec0.lf#LF95',
 'imec0.lf#LF144-imec0.lf#LF145',
 'imec0.lf#LF146-imec0.lf#LF147',
 'imec0.lf#LF148-imec0.lf#LF149',
 'imec0.lf#LF150-imec0.lf#LF151',
 'imec0.lf#LF152-imec0.lf#LF153',
 'imec0.lf#LF154-imec0.lf#LF155',
 'imec0.lf#LF156-imec0.lf#LF157',
 'imec0.lf#LF158-imec0.lf#LF159',
 'imec0.lf#LF160-imec0.lf#LF161',
 'imec0.lf#LF162-imec0.lf#LF163',
 'imec0.lf#LF164-imec0.lf#LF165',
 'imec0.lf#LF166-imec0.lf#LF167',
 'imec0.lf#LF168-imec0.lf#LF169',
 'imec0.lf#LF170-imec0.lf#LF171',
 'imec0.lf#LF172-imec0.lf#LF173',
 'imec0.lf#LF174-imec0.lf#LF175',
 'imec0.lf#LF176-imec0.lf#LF177',
 'imec0.lf#LF178-imec0.lf#LF179',
 'imec0.lf#LF180-imec0.lf#LF181',
 'imec0.lf#LF182-imec0.lf#LF183',
 'imec0.lf#LF184-imec0.lf#LF185',
 'imec0.lf#LF186-imec0.lf#LF187',
 'imec0.lf#LF188-imec0.lf#LF189',
 'imec0.lf#LF190-imec0.lf#LF191',
 'imec0.lf#LF240-imec0.lf#LF241',
 'imec0.lf#LF242-imec0.lf#LF243',
 'imec0.lf#LF244-imec0.lf#LF245',
 'imec0.lf#LF246-imec0.lf#LF247',
 'imec0.lf#LF248-imec0.lf#LF249',
 'imec0.lf#LF250-imec0.lf#LF251',
 'imec0.lf#LF252-imec0.lf#LF253',
 'imec0.lf#LF254-imec0.lf#LF255',
[... 64 lines cut ...]
 'imec0.lf#LF4-imec0.lf#LF5-imec0.lf#LF196-imec0.lf#LF197',
 'imec0.lf#LF6-imec0.lf#LF7-imec0.lf#LF198-imec0.lf#LF199',
 'imec0.lf#LF8-imec0.lf#LF9-imec0.lf#LF200-imec0.lf#LF201',
 'imec0.lf#LF10-imec0.lf#LF11-imec0.lf#LF202-imec0.lf#LF203',
 'imec0.lf#LF12-imec0.lf#LF13-imec0.lf#LF204-imec0.lf#LF205',
 'imec0.lf#LF14-imec0.lf#LF15-imec0.lf#LF206-imec0.lf#LF207',
 'imec0.lf#LF16-imec0.lf#LF17-imec0.lf#LF208-imec0.lf#LF209',
 'imec0.lf#LF18-imec0.lf#LF19-imec0.lf#LF210-imec0.lf#LF211',
 'imec0.lf#LF20-imec0.lf#LF21-imec0.lf#LF212-imec0.lf#LF213',
 'imec0.lf#LF22-imec0.lf#LF23-imec0.lf#LF214-imec0.lf#LF215',
 'imec0.lf#LF24-imec0.lf#LF25-imec0.lf#LF216-imec0.lf#LF217',
 'imec0.lf#LF26-imec0.lf#LF27-imec0.lf#LF218-imec0.lf#LF219',
 'imec0.lf#LF28-imec0.lf#LF29-imec0.lf#LF220-imec0.lf#LF221',
 'imec0.lf#LF30-imec0.lf#LF31-imec0.lf#LF222-imec0.lf#LF223',
 'imec0.lf#LF32-imec0.lf#LF33-imec0.lf#LF224-imec0.lf#LF225',
 'imec0.lf#LF34-imec0.lf#LF35-imec0.lf#LF226-imec0.lf#LF227',
 'imec0.lf#LF36-imec0.lf#LF37-imec0.lf#LF228-imec0.lf#LF229',
 'imec0.lf#LF38-imec0.lf#LF39-imec0.lf#LF230-imec0.lf#LF231',
 'imec0.lf#LF40-imec0.lf#LF41-imec0.lf#LF232-imec0.lf#LF233',
 'imec0.lf#LF42-imec0.lf#LF43-imec0.lf#LF234-imec0.lf#LF235',
 'imec0.lf#LF44-imec0.lf#LF45-imec0.lf#LF236-imec0.lf#LF237',
 'imec0.lf#LF46-imec0.lf#LF47-imec0.lf#LF238-imec0.lf#LF239',
 'imec0.lf#LF96-imec0.lf#LF97-imec0.lf#LF288-imec0.lf#LF289',
 'imec0.lf#LF98-imec0.lf#LF99-imec0.lf#LF290-imec0.lf#LF291']
`

taningh86, 2025-01-20:
I might be a bit slow in responding back. I would say few hours as I am doing some surgeries. Thanks!

cwindolf, 2025-01-20:
Thanks, that's great.

First of all, just a high level suggestion. Since you have two well-separated "layers" that you're recording from, I'd suggest to process both of those groups separately. The step `si.directional_derivative()` will attempt to take the second derivative along the probe, which will lead to strange results across the ~3mm gap there, which could make it challenging to proceed to whatever downstream steps you're planning to carry out. You can use `.channel_slice()` on the raw recording to split it up before processing. It would likely also make sense to separate things by shank, but I'm less sure about that -- it depends on how consistent the LFP is over shanks, but in principle this could make a big difference in your analysis and conclusions.

Second, I just wanted to double check something. In the plots of your message above, the time domain goes from 0-2 instead of 5-7 as described in your code, so there is a mismatch. I'd appreciate if we could just triple check the outputs, just to make sure that the images come from the exact code given. In particular, could you display:
 - `si.plot_traces(lfprec2, backend="matplotlib", mode="map", channel_ids=['imec0.lf#LF42-imec0.lf#LF43-imec0.lf#LF234-imec0.lf#LF235', 'imec0.lf#LF44-imec0.lf#LF45-imec0.lf#LF236-imec0.lf#LF237'], time_range=(5,7))`
 - `si.plot_traces(lfprec2, backend="matplotlib", mode="map", channel_ids=['imec0.lf#LF48-imec0.lf#LF49', 'imec0.lf#LF50-imec0.lf#LF51'], time_range=(5,7))`

Sorry about those, just want to double check! Finally, could you try the below plots of the first and last 20 channels, to check whether there is something weird going on when plotting only 2 channels?
 - `si.plot_traces(lfprec2, backend="matplotlib", mode="map", channel_ids=lfprec.channel_ids[:20], time_range=(5,7))`
 - `si.plot_traces(lfprec2, backend="matplotlib", mode="map", channel_ids=lfprec.channel_ids[-20:], time_range=(5,7))`

My apologies for all the requests!

taningh86, 2025-01-20:
Thanks for the detailed message @cwindolf . No worries at all and below are what you asked for. I am not seing any difference between them.
For `si.plot_traces(lfprec2, backend="matplotlib", mode="map", channel_ids=['imec0.lf#LF42-imec0.lf#LF43-imec0.lf#LF234-imec0.lf#LF235', 'imec0.lf#LF44-imec0.lf#LF45-imec0.lf#LF236-imec0.lf#LF237'], time_range=(5,7))` below is the output
[image]

for `si.plot_traces(lfprec2, backend="matplotlib", mode="map", channel_ids=['imec0.lf#LF48-imec0.lf#LF49', 'imec0.lf#LF50-imec0.lf#LF51'], time_range=(5,7))` below is the output

[image]

For,` si.plot_traces(lfprec2, backend="matplotlib", mode="map", channel_ids=lfprec.channel_ids[:20], time_range=(5,7))`
[image]

For, `si.plot_traces(lfprec2, backend="matplotlib", mode="map", channel_ids=lfprec.channel_ids[-20:], time_range=(5,7))` below is the output

[image]

Also, I will get back to you with more questions about averaging through the length of the shank and across shanks. I have always thought about that and I believe the same logic applies to AP's as well. I believe many people are doing that and that should be the way IMO. I will get back with more questions after I am done with my surgeries. Thanks a lot again for troubleshooting this.

cwindolf, 2025-01-20:
Thanks @taningh86 ! Turns out this was just a straight up bug. I've submitted a PR to fix: https://github.com/SpikeInterface/spikeinterface/pull/3631

closed 2025-01-20 by alejoe91

taningh86, 2025-01-21:
@cwindolf Thank a lot for submitting the PR to Fix. Sorry for the ate response.