SpikeInterface/spikeinterface #3634: Is Spikeinterface a suitable choice for macro-micro electrodes?

gbShaw, 2025-01-21:
Hi, I found that Spikeinterface is a comprehensive tool for spike analysis and processing, especially for neuropixels. 
However, I'm not sure if many of its excellent features are suitable for macro-micro electrodes (which have flexible microfilaments without spatial relationship information between the microfilaments). I know that custom electrodes can be used with Spikeinterface (even though their spatial information is meaningless for macro-micro electrodes), but is this appropriate?

Thank you in advance

[image]

Antho2422, 2025-01-21:
Hi @gbShaw and @yger ! 

Yes, in our lab we were using SpyKing Circus 1 for many years and more recently switched to SC2 with Spikeinterface and we are pretty happy about it. Pierre worked a lot on it to make it suit macro-micro !

Of course results are not as accurate compared to what you could expect with neuropixels but it turns out that if your signal to noise ratio is sufficient you can extract pretty good units accurately

The only thing is that you could pick up the same unit twice on 2 microwires for example if they are close from each other in space (which happens very rarely imo). So you will need to spend a bit of time on the curation ! 

Let me know if you have more questions !

Cheers,
Anthony

zm711, 2025-01-24:
I will close this for now. But feel free to reach out with more questions as they come up @gbShaw. Thanks for the great answer @Antho2422 !

closed 2025-01-24 by zm711

gbShaw, 2025-01-25:
Thank you, @zm711 and @yger, for your responses. I'm excited to use the spike interface in human cells. Thanks again!