SpikeInterface/spikeinterface #3637: "hard" merge of units in sorting analyzer fails on extensions computations

Antho2422, 2025-01-22:
Hi all, 

I am encountering an error during the extension computation (amplitude_scalings + spike_amplitudes + spike_locations) when applying curation (with merges) on an analyzer. 

This error happens only when merging mode is set to `"hard"` in the call of `apply_curation()`. I'm not very familiar with the internal spikeinterface code for computing extensions so I'm struggling to debug this and I would love a bit of help. Note that when applying the curation to the sorting object, it works fine. 

You'll find here the analyzer, sorting and recording object together with the curation_dict.json file that I'm using so you can reproduce (hopefully) this behavior : https://instituteicm-my.sharepoint.com/:f:/g/personal/anthony_pinto_icm-institute_org/EhU1nQWpaNBFtVovmQaDUeEB5RJnkYReYfWu4ChMQ088OA?e=Aw9XSe

Here is also the code that you need to be able to reproduce : 

```python
import spikeinterface as si
import json
from spikeinterface.curation import apply_curation

raw_sorting = si.load_extractor(r'path_to_sorting')
raw_analyzer = si.load_sorting_analyzer(r'path_to_analyzer')
recording = si.load_extractor(r'path_to_recording')
raw_analyzer._recording = recording
with open(r'path_to_curation_dict', 'r') as f:
    curation_dict = json.load(f)

print(recording), print(raw_sorting), print(raw_analyzer), print(curation_dict)

curated_analyzer = apply_curation(raw_analyzer, curation_dict, new_id_strategy='take_first', merging_mode='hard')
``` 

Help would be really appreciated ! Thank you !

Cheers, 
Anthony

alejoe91, 2025-01-22:
Hi @Antho2422 

What's the error? :)

Antho2422, 2025-01-22:
@alejoe91 oops I completly forgot this point ! 

Here it is :  
```python
--> [370](file:///C:/Users/anthony.pinto/Documents/poetryenvs/priam-5spsvreu-py3.9/lib/site-packages/spikeinterface/core/node_pipeline.py:370)     waveforms = traces[peaks["sample_index"][:, None] + np.arange(-self.nbefore, self.nafter)]
    [371](file:///C:/Users/anthony.pinto/Documents/poetryenvs/priam-5spsvreu-py3.9/lib/site-packages/spikeinterface/core/node_pipeline.py:371)     return waveforms

IndexError: index 33028 is out of bounds for axis 0 with size 33028
```

alejoe91, 2025-01-22:
I cannot reproduce the issue. Can you make sure you are using the main branch?

[image]

Antho2422, 2025-01-22:
@alejoe91 I just tested with the last update of spikeinterface on branch master an unfortunately I have the same issue....

alejoe91, 2025-01-22:
This was running the code you provided using the main branch, so I'm not able to reproduce it

Antho2422, 2025-01-22:
Weird, I just updated my environment and a collegue was able to reproduce the issue aswell... I will create a new environment from scratch and test again :)

alejoe91, 2025-01-22:
Which OS + Python version are you using? Could you reproduce on the data you shared or on another dataset?

Antho2422, 2025-01-22:
I'm on windows and using Python3.9. I only tried on the data I shared today but I remember having the same issue on other data (not 100% sure tho). 

Here is the same issue when creating a new venv in python3.9 then installing spikeinterface directly from the git. 

[image]

alejoe91, 2025-01-22:
Can you try to run it with 1 job? Might be Windows related, but it's strange!

Antho2422, 2025-01-22:
I already ran it with one job only same issue. I think it is not Python related aswell because I just tried with Python3.11 and I have the same issue. Another collegue managed to pass the line that was causing problem (**On windows**) (and got another issue). So maybe this has something to do with a dependency ? I will investigate and compare environments.

alejoe91, 2025-01-22:
Can you try to install from source and from main, not 0.101.2?

Antho2422, 2025-01-22:
@alejoe91 I did, I was getting the error when using directly source code from main.

Antho2422, 2025-01-22:
I think the issue is not OS dependent. @alejoe91 are you sure you were using the main branch here https://github.com/SpikeInterface/spikeinterface/issues/3637#issuecomment-2607041926 ? I'm a bit intrigued by the warning (I don't have it on main)

alejoe91, 2025-01-22:
Yep, reinstalled from source to be sure

[image]

[image]

And added a print statement for you ;)

[image]

Are you sure you are using the main branch??

Antho2422, 2025-01-22:
The only reason I can imagine right now would be some package that as a different version in my env, here is the output of pip freeze 

[non_working_env.txt](https://github.com/user-attachments/files/18508775/non_working_env.txt)

Antho2422, 2025-01-22:
Same problem...

alejoe91, 2025-01-22:
let's see if we can reproduce it on windows!

samuelgarcia, 2025-01-22:
I think, there are spikes too near the border and we are not safe enought after the merging in mode "hard".
But this is strange.

Antho2422, 2025-01-22:
That is my guess as well, when chunking the recording, some peaks might be near the border of the chunk, when adding n_samples the index of the peaks goes over the samples of the chunk

But the question is why is it working in @alejoe91 setup ?

samuelgarcia, 2025-01-22:
I think this depend on the numpy version installed that behave strictly or not.
we need to fix anyway

Antho2422, 2025-01-22:
@alejoe91 what numpy version do you have at the moment ? I'm on `1.26.4`

chrishalcrow, 2025-01-23:
Hello, I managed to reproduce the issue! I haven't solved it yet but I've found something worth looking at. If I run

```
raw_analyzer.compute_several_extensions({"spike_amplitudes":{},"spike_locations":{},"amplitude_scalings":{}})
```

it's fine. But the `all_nodes` list (a variable used in `run_node_pipeline` in `sortinganalyzer.py`) is in a different order than when the merging takes place. For the `compute_several_extensions`, all_nodes is equal to 

```
all_nodes = [<spikeinterface.core.node_pipeline.SpikeRetriever object at 0x1766bb0e0>, 
<spikeinterface.postprocessing.spike_amplitudes.SpikeAmplitudeNode object at 0x313befc20>, 
<spikeinterface.core.node_pipeline.SpikeRetriever object at 0x175b3fec0>, 
<spikeinterface.core.node_pipeline.ExtractDenseWaveforms object at 0x175b3e0c0>, 
<spikeinterface.sortingcomponents.peak_localization.LocalizeCenterOfMass object at 0x175b3daf0>, 
<spikeinterface.core.node_pipeline.SpikeRetriever object at 0x175b3fe30>, 
<spikeinterface.postprocessing.amplitude_scalings.AmplitudeScalingNode object at 0x175b3ea50>]
```

while during the merging, all_nodes is equal to:

```
all_nodes = [<spikeinterface.core.node_pipeline.SpikeRetriever object at 0x1766ef950>, 
<spikeinterface.postprocessing.amplitude_scalings.AmplitudeScalingNode object at 0x175cf2f00>, 
<spikeinterface.core.node_pipeline.SpikeRetriever object at 0x175cf32c0>, 
<spikeinterface.postprocessing.spike_amplitudes.SpikeAmplitudeNode object at 0x175cf2f30>, 
<spikeinterface.core.node_pipeline.SpikeRetriever object at 0x177eacbf0>, 
<spikeinterface.core.node_pipeline.ExtractDenseWaveforms object at 0x1760ea5a0>, 
<spikeinterface.sortingcomponents.peak_localization.LocalizeCenterOfMass object at 0x176025670>]
```
 
If I change the order:
```
raw_analyzer.compute_several_extensions({"amplitude_scalings":{},"spike_amplitudes":{},"spike_locations":{}})
```
I get the familiar error...

```
...
File [~/fromgit/spikeinterface/src/spikeinterface/core/node_pipeline.py:370](http://localhost:65144/lab/tree/~/fromgit/spikeinterface/src/spikeinterface/core/node_pipeline.py#line=369), in ExtractDenseWaveforms.compute(self, traces, peaks)
    369 def compute(self, traces, peaks):
--> 370     waveforms = traces[peaks["sample_index"][:, None] + np.arange(-self.nbefore, self.nafter)]
    371     return waveforms

IndexError: index 33028 is out of bounds for axis 0 with size 33028
```

So I suspect it's to do with the order of computing the extensions. @samuelgarcia is there a reason that `amplitude_scalings` should be computed after `spike_amplitudes`? Or maybe the nodes need to be reordered during `compute_several_extensions`?

Antho2422, 2025-01-23:
Maybe it's related with the fact that list ordering can be handled different from one OS to another ? (it could explain why it works on @alejoe91 computer) 

I don't know !

samuelgarcia, 2025-01-24:
@chrishalcrow spike_ampitudes and amplitude_scaling should be independant and the order should not count.

But what you point is really important.
In short :  there is mechanim to make the first special (the first spike reteiver) in the nodepipeline.
My guess is that the others spikeretreiver behave differently and the border are not handle correctly. I will also try to have a look.

chrishalcrow, 2025-01-27:
Hey @Antho2422 . Maybe while we try and figure this out, you could only compute `amplitude_scalings` at the very end. I guess you don't need these to help figure out if you want to merge or not. So I suggest

Load your sorting analyzer
Delete the amplitude scalings extension
Do your merging
Compute amplitude scalings (if you really need them!)

And we'll let you know when this is fixed!

Antho2422, 2025-01-27:
Thank you @samuelgarcia @chrishalcrow !

I will do this workaround for the moment :)

Please let me know when this is fixed !

Cheers

alejoe91, 2025-01-27:
Ok guys I found a lead. Turns out that this only fails if `amplitude_scalings` is first in the list. I think this is because the first node is used throughout the pipeline to estimate some indices, and the `amplitude_scaling` spike retriever is trickier because it also retrieves spikes within the margin (which are used for proper fitting at the borders).

Hopefully will find qa fix soon!

alejoe91, 2025-01-27:
Should be fixed here: #3642

Antho2422, 2025-01-27:
Awesome ! Will test asap ! Thank you guys !

closed 2025-01-31 by alejoe91