SpikeInterface/spikeinterface #3647: global_job_kwargs issue in running Kilosort 2.5 in docker environment: invalid parameter max_threads_per_process

Djoels, 2025-01-28:
In my experiment pipeline, I only set job_kwargs as such:

```
global_job_kwargs = dict(n_jobs=.7)
si.set_global_job_kwargs(**global_job_kwargs)
```

When running the KS2.5 sorter in docker mode as such:
```
params={
            "nblocks": 0,
            "do_correction": False,
            "delete_recording_dat": True,
            "minFR": 0.01,
            "projection_threshold": [8, 4],
            "detect_threshold": 4,
            "preclust_threshold": 8,
            "freq_min":  300,
            "lam": 0,
            "whiteningRange": 32
}
sorting = ss.run_sorter(sorter_name=sorter, 
            recording=recording,
            folder=output_folder,
            docker_image=docker_image,
            verbose=_verbose,
            raise_error=True,
            extra_requirements=["numpy==1.26.4"],
            installation_mode="github",
            delete_container_files=True,
            **params
        )
```

I get the following error:
```
Traceback (most recent call last):
  File "/hybridgt_20240816/in_container_sorter_script.py", line 23, in <module>
    sorting = run_sorter_local(
  File "/root/.local/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py", line 257, in run_sorter_local
    SorterClass.set_params_to_folder(recording, folder, sorter_params, verbose)
  File "/root/.local/lib/python3.9/site-packages/spikeinterface/sorters/basesorter.py", line 184, in set_params_to_folder
    raise ValueError(error_msg)
ValueError: Invalid parameters: ['max_threads_per_process'] 
Valid parameters are: dict_keys(['detect_threshold', 'projection_threshold', 'preclust_threshold', 
'whiteningRange', 'momentum', 'car', 'minFR', 'minfr_goodchannels', 'nblocks', 'sig', 'freq_min', 
'sigmaMask', 'lam', 'nPCs', 'ntbuff', 'nfilt_factor', 'NT', 'AUCsplit', 'do_correction', 'wave_length', 
'keep_good_only', 'skip_kilosort_preprocessing', 'scaleproc', 'save_rez_to_mat', 
'delete_tmp_files', 'delete_recording_dat', 'pool_engine', 'n_jobs', 'chunk_duration', 
'progress_bar', 'mp_context', 'max_threads_per_worker'])
```

The job_kwargs most similar is max_threads_per_worker so potentially it's a version mismatch issue?

alejoe91, 2025-01-28:
Hi @Djoels 

Yes. We refactored the job kwargs in main and since this still has to be released it's causing you an error.
Can you try the `installation_mode="dev"` instead? This will force the same version that you have in your host machine.

Djoels, 2025-01-28:
I've adapted to `dev` mode and get this as output:

```
{'n_jobs': 0.7, 'chunk_duration': '1s', 'progress_bar': True, 'mp_context': None, 'max_threads_per_process': 1}
```

Will report on how this runs. 
But having trouble with the setup: the spikeinterface_folder_source, does it have to be set to the parent directory of the github repo, or to the src directory, when using dev mode?

alejoe91, 2025-01-28:
If you use dev mode the code should automatically find the correct folder, so you don't have to specify the Spikeinterface folder source

Djoels, 2025-01-28:
even from the correct conda environment, it apparently doesn't find spikeinterface?

```
Traceback (most recent call last):
  File "hpoptuna.py", line 313, in main
    scores = _run_sorter(
             ^^^^^^^^^^^^
  File "hpoptuna.py", line 85, in _run_sorter
    sorting = ss.run_sorter(sorter_name=sorter,
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/anaconda/envs/boc_minimal/lib/python3.12/site-packages/spikeinterface/sorters/runsorter.py", line 193, in run_sorter
    return run_sorter_container(
           ^^^^^^^^^^^^^^^^^^^^^
  File "/anaconda/envs/boc_minimal/lib/python3.12/site-packages/spikeinterface/sorters/runsorter.py", line 648, in run_sorter_container
    raise SpikeSortingError(f"Spike sorting in {mode} failed with the following error:\n{run_sorter_output}")
spikeinterface.sorters.utils.misc.SpikeSortingError: Spike sorting in docker failed with the following error:
Traceback (most recent call last):
  File "/hybridgt_20240816/in_container_sorter_script.py", line 4, in <module>
    from spikeinterface import load
ModuleNotFoundError: No module named 'spikeinterface'
```

alejoe91, 2025-01-28:
I see. Can you try to install from source but in editable mode with `pip install -e .`

Djoels, 2025-01-28:
aha, that made it work! 
didn't know you had to install this way to use dev mode. thanks!

I will close then, given that the original issue will automagically be resolved?

closed 2025-01-28 by Djoels