SpikeInterface/spikeinterface #3650: Spike locations from Kilosort4 output vs through SpikeInterface's SortingAnalyzer -> si.widgets.plot_spike_locations are wildly different

samarasinghelab, 2025-01-29:
Our workflow involves reading in a Maxwell MEA h5 file, basic preprocessing steps (CAR/highpass through kilosort4), followed by spike sorting. I've extracted the spike locations from the kilosort4 output files and create a dataframe and can plot it (with the convex hull of spikes) seen in the figure below. 

Directly imported 
[image]

Overall looks somewhat reasonable. However, as a sanity check I subsequently create a SortingAnalyzer object from the sorting object and run it through the spikeinterface pipeline eventually to widgets.plot_spike_locations and the results are very different. Sometimes the results have all spike locations overlapping, other times they look more spread out but still not consistent with the kilosort results. For an example of the former, I took 2 early spikes from the data, the unit/timing of these spikes is correct in both the kilosort output folder and my saved spikeinterface sorting_analyzer folder. However, the locations (xpos, ypos) are very different. These results are for center_of_mass arguments (although I've tried grid_convolution as well).

Spike locations from Kilosort4 folder:
       cluster       time       xpos         ypos
103    0.00245  1412.917236  1069.924438
207    0.00415  16.870857  1485.881592  1714.498535

Spike locations from SortingAnalyzer folder:
       cluster       time       xpos         ypos
103    0.00245  1673.93564356  578.01980198
207    0.00415  1670.025  581.35

I wanted to see if there was a way SpikeInterface could automatically extract the kilosort4 spike locations or if there was a workaround (I guess I could manually overwrite the SortingAnalyzer file (.../sorting_analyzer/extensions/spike_locations/spike_locations.npy) with results from Kilosort but this seems unnecessarily cumbersome).

Thanks so much for any assistance with this.

Code:
    ```
# Load in maxwell file
rec = si.extractors.read_maxwell(input_file)
rec = si.preprocessing.unsigned_to_signed(recording=rec,bit_depth=10)

# Update probe info (prior version electrode size 5.5 x 9.3)
probe0 = rec.get_probe()
probe0.set_contacts(positions=probe0.contact_positions, shapes=probe0.contact_shapes, shape_params=np.full(len(probe0.contact_shape_params),{'width': 8.75, 'height': 12.5}))
rec = rec.set_probe(probe0)

rec = rec.time_slice(start_time=0.0, end_time=300.0)
fs = rec.get_sampling_frequency()
tdur = rec.get_duration()
lsb = sc_mode(rec.get_property('gain_to_uV'))[0]
    
# save recording
rec_dir = output_dir
rec.save_to_folder(rec_dir)
with open(rec_dir / 'tdur.pkl', 'wb') as f:
    pickle.dump(tdur, f)
save_probe_map(rec,rec_dir) 

# Run kilosort with optimal settings, dminx 18-32
spk_dir = output_dir/'spikes'
spk_rec = rec
srt_dir = spk_dir/'sorting'
spk_srt = si.sorters.run_sorter(sorter_name='kilosort4', recording=spk_rec, folder=srt_dir, \
    batch_size=60000, nblocks=0, Th_universal=9, Th_learned=8, nt=int(0.002*fs+1), dminx=17.5, x_centers=14, highpass_cutoff=300, do_CAR=True)

# Extract data from kilosort files and create dataframe, then save spike location plot
srt_dir = srt_dir/'sorter_output'
spk_clu = np.load(srt_dir/'spike_clusters.npy')
spk_tim = np.load(srt_dir/'spike_times.npy') / fs # in seconds
spk_amp = np.load(srt_dir/'amplitudes.npy')
spk_pos =  np.load(srt_dir/'spike_positions.npy')
assert len(spk_clu) == len(spk_tim) == len(spk_amp) == len(spk_pos)
spk_dfr = pd.DataFrame(data={'cluster': spk_clu, 'time': spk_tim, 'amplitude': spk_amp, 'xpos': spk_pos[:,0], 'ypos': spk_pos[:,1]}, index=None)
spk_dfr = spk_dfr.sort_values(by=['cluster','time'], ascending=[True,True])   
spk_dfr.reset_index(drop=True,inplace=True)
spk_dfr.to_csv(spk_dir/'spike_data.csv', index=False)
save_spike_map(spk_dfr,spk_rec,spk_dir)

# Use the spikeinterface method to extract spike locations from sorting object/recording and plot
spk_est_sparsity = si.estimate_sparsity(sorting=spk_srt, recording=spk_rec, ms_before=1.0, ms_after=2.5)
spk_srt_anl = si.core.SortingAnalyzer.create(sorting=spk_srt, recording=spk_rec, format='binary_folder', folder=spk_dir/'sorting_analyzer', sparsity=spk_est_sparsity)
spk_srt_anl.compute({'random_spikes':{},'noise_levels':{},'waveforms':{'ms_before':1.0,'ms_after':2.5},'templates':{'operators':['average','std',]},'spike_locations':{'method':'center_of_mass ','ms_before':0.5,'ms_after':0.5},'unit_locations':{'method':'center_of_mass '}},save=True)
si.widgets.plot_spike_locations(spk_srt_anl)
```

chrishalcrow, 2025-01-29:
Hi @samarasinghelab , my guess is as follows:

Kilosort takes in your recording and applies some preprocessing (common median reference, bandpass filter and whitening) to it before sorting it. However, your sorting_analyzer is using a non-preprocessed recording. One solution: before making the sorting_analyzer, apply some preprocessing. It won't be identical to what kilosort does but it should be roughly consistent. Maybe something like:

``` python
from spikeinterface.preprocessing import bandpass_filter, common_reference
pp_rec = bandpass_filter(common_reference(spk_rec))
```

then use `pp_rec` in your `SortingAnalyzer.create`.

A little note: the `create_sorting_analyzer` function is nice, and will deal with sparsity for you. Unless you've got a good reason for changing the sparsity, I'd advise using

``` python
spk_srt_anl  = si.create_sorting_analyzer(recording=pp_rec, sorting= spk_srt, format='binary_folder', folder=spk_dir/'sorting_analyzer')
```

samuelgarcia, 2025-01-30:
Note that we have 3 methods for spike/unit localization:
  * center of mass : fast but inacurate on border
  * monopolar triangultaion : slower but more accurate can give some strange result sometimes when the spatial shape is not "smooth"
  * grid convolution (quite similar to what kilosort is doing) : somhow a in between solution.

You should try to play with this 3 ones to get your own idea.

samarasinghelab, 2025-02-01:
Thank you so much @chrishalcrow for your explanation. I'll give this a try.

closed 2025-02-01 by samarasinghelab