SpikeInterface/spikeinterface #3653: PCA quality metrics crash when using "concatenated" mode

b-grimaud, 2025-01-29:
Computing PCA quality metrics crashes if PCs were calculated with `mode='concatenated'` : 

```
analyzer.compute("principal_components",
                 mode='concatenated',
                 n_components=5,
                 dtype='float32')
```

Then : 

```
analyzer.compute("quality_metrics",
                 metric_names=[
                  'd_prime',
                  'silhouette',
                  'nearest_neighbor',
                 ],
                 delete_existing_metrics=True
                 )
```
I checked with each of these metrics individually and they all return the same error.

Here's the full trace : 

```
Traceback (most recent call last):
  File "/home/user/Documents/ephy/pca_crash.py", line 50, in <module>
    analyzer.compute("quality_metrics",
  File "/home/user/Documents/GitHub/spikeinterface/src/spikeinterface/core/sortinganalyzer.py", line 1324, in compute
    return self.compute_one_extension(extension_name=input, save=save, verbose=verbose, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/Documents/GitHub/spikeinterface/src/spikeinterface/core/sortinganalyzer.py", line 1403, in compute_one_extension
    extension_instance.run(save=save, verbose=verbose, **job_kwargs)
  File "/home/user/Documents/GitHub/spikeinterface/src/spikeinterface/core/sortinganalyzer.py", line 2171, in run
    self._run(**kwargs)
  File "/home/user/Documents/GitHub/spikeinterface/src/spikeinterface/qualitymetrics/quality_metric_calculator.py", line 261, in _run
    computed_metrics = self._compute_metrics(
                       ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/Documents/GitHub/spikeinterface/src/spikeinterface/qualitymetrics/quality_metric_calculator.py", line 224, in _compute_metrics
    pc_metrics = compute_pc_metrics(
                 ^^^^^^^^^^^^^^^^^^^
  File "/home/user/Documents/GitHub/spikeinterface/src/spikeinterface/qualitymetrics/pca_metrics.py", line 144, in compute_pc_metrics
    dense_projections, spike_unit_indices = pca_ext.get_some_projections(channel_ids=None, unit_ids=unit_ids)
                                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/Documents/GitHub/spikeinterface/src/spikeinterface/postprocessing/principal_component.py", line 263, in get_some_projections
    some_projections = all_projections[selected_inds, :, :][:, :, channel_indices]
                       ~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^
IndexError: too many indices for array: array is 2-dimensional, but 3 were indexed
```

Looks like the [_transform_waveforms](https://github.com/SpikeInterface/spikeinterface/blob/d71cabafdba69bf8282149d82f53bbfecff67696/src/spikeinterface/postprocessing/principal_component.py#L517) function returns a three dimensional array if the mode is `"by_channel_local"` or `"by_channel_global"`, and a two dimensional array if the mode is `"concatenated"`.

Is this expected behavior ? If so, maybe checking for the mode and skipping PCA quality metrics if the mode is `"concatenated"` would help.

alejoe91, 2025-01-30:
Hi @b-grimaud 

Definitely a bug! We have to admit that the `concatenated` mode is largley under-tested...would you have time to try to work on a fix?

chrishalcrow, 2025-02-05:
Hey @b-grimaud , I think I fixed it! Are you able to test out PR #3670 ? Thanks!

PR #3670 (2025-02-05): Allow for concatenated mode in `get_some_projections` and `compute_pc_metrics`
Should fix #3653

Fixes bug where `get_some_projections` and `compute_pc_metrics` couldn't extract pcas when they were computed in concatenated mode. Fixing this allows pca quality metrics to be computed too.

Added a test which used to fail, and now passes.

Harder to test if this definitely works. Here's some evidence: it's code which computes the quality metrics in `concatenated` mode (with 20 components) and in non-concatenated mode (with `5x4=20` components), then computes the metrics. We wouldn't expect the re [... cut]
closed 2025-02-05 by alejoe91