SpikeInterface/spikeinterface #3655: remove_artifact seems to not zero at index

mik-schutte, 2025-01-29:
Recently started using spikeinterface and want to remove my stimulation actifact from the recordings.  From your tutorials this seems to be done with a single line of code. However, I cannot get it to function. Can someone please provide me with some insight as to what I might be doing incorrect?

```
import spikeinterface.full as si
import matplotlib.pyplot as plt

filtFolder = 'D:/nPixelData/SNA-137704/SNA-137704_mstim_20112024_S1_1_g0/catgt_SNA-137704_mstim_20112024_S1_1_g0'
filt_rec = si.read_spikeglx(folder_path=filtFolder, stream_id='imec0.ap')
mStim_TTLs = [37.668]
blanked_rec = si.remove_artifacts(recording=filt_rec, list_triggers=mStim_TTLs, ms_after=200)

fig, axs = plt.subplots(ncols=2)
si.plot_traces(filt_rec, backend='matplotlib', ax=axs[0], clim=(-50, 50), time_range=(mStim_TTLs[0]-0.2, mStim_TTLs[0]+0.4), channel_ids=filt_rec.channel_ids[:10])
si.plot_traces(blanked_rec, backend='matplotlib', ax=axs[1], clim=(-50, 50), time_range=(mStim_TTLs[0]-0.2, mStim_TTLs[0]+0.4), channel_ids=blanked_rec.channel_ids[:10])
[axs[i].set_title(title) for i, title in zip([0,1],['Non-blanked', 'Blanked'])]
plt.show()
```

[image]

alejoe91, 2025-01-29:
Hi,

list_triggers is in samples, whilee time range is in seconds! That's why you are not seeing any difference.

mik-schutte, 2025-01-30:
@alejoe91 thanks for the fast reply!
If anyone stumbles on the same issue in the future all I had to do was:
```
mStim_TTLs = [37.668]
mStim_sample_idx = [int(mStim_TTL * filt_rec.sampling_frequency) for mStim_TTL in mStim_TTLs]
blanked_rec = si.remove_artifacts(recording=filt_rec, list_triggers=mStim_sample_idx, ms_after=200)
```