SpikeInterface/spikeinterface #3656: Navigating SpikeInterface Documentation

JacobKelley101, 2025-01-29:
Hello,

I am a relative new-user for all things spike sorting and python. For context, I needed a tutorial and multiple attempts to just install anaconda, jupyter notebook, and SpikeInterface, so my troubleshooting skills are very limited/naive. I have a couple of questions regarding navigating the SpikeInterface documentation, best-practices for pipeline development, and simpler smaller questions that I have collected over the past weeks.

1.) I am aware of certain SI functions that can detect and remove bad channels from recording objects before they are used for sorting and post-processing. My lab only uses tetrodes, however, so removing entire channels is not an option. We would rather detect instances of high-amplitude noise ourselves and replace it with low-amplitude noise so that we don't lose 25% of our data because someone bumped into our floating table 3hrs into a recording. Are there SI functions that do this, or is this something we would have to perform on our raw data ourselves? If we need to do it ourselves, are there any recommendations the developers would have for a detection-threshold, and how many adjacent indices we should replace?

2.) When calculating extensions for the sorting analyzer such as waveforms, templates, principal components, etc, is stated in the ReadTheDocs page that "each extension can retrieve some data." The example used is the get_waveforms_one_unit function (or is it called an attribute? I am not sure of the terminology here). I cannot find an exhaustive list of these functions for each analyzer extension, so trying to figure out what I need to calculate myself is challenging. How would the developers suggest I assess which extensions can do what, without bouncing around the various example pages on the ReadTheDocs page?

3.) It appears that the sorter I am using (MountainSort5) is picking up the hyper polarization that precedes an action potential as the center of the waveform (see picture). This is causing, especially units with large amplitudes, to be split into multiple clusters. Some are centered, others are not. I am imagining that rasing my detection threshold could fix this, but I fear I will miss out on low-amplitude units in future analyses if I do that. Are there any SpikeInterface functionalities that fix this problem? 

Thank you very much for all your help. I have always appreciated the responses I get from SI developers.

[image]

zm711, 2025-01-30:
> 1.) I am aware of certain SI functions that can detect and remove bad channels from recording objects before they are used for sorting and post-processing. My lab only uses tetrodes, however, so [...]

Something like bumping the table could be handled by giving the explicit timestamp and removing with [remove_artifacts](https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/preprocessing/remove_artifacts.py). If you have other types of noise then it likely needs to be filtered or referenced in some way. Most spike sorters as part of their pipelines filter, reference (+/- whiten) data. So at the preprocessing level you would need to look up your sorter and see if you should do additional preprocessing. MS5 for example uses the SI machinery to filter data and whiten (I need to check if it references too). But if you have explicit times of large noise then `remove_artifacts` can do that. It is still slightly manual since it doesn't autodetect the noise, instead it requires you to say where the noise is. Does that make sense?

> 2.) When calculating extensions for the sorting analyzer such as waveforms, templates, principal components, etc, is stated in the ReadTheDocs page that "each extension can retrieve some data." The [...]

this really depends what you want to do. For my purposes I typically don't get that granular. I just use the `sorting_analyzer.compute()` function to calculate all my extensions. So I think it would be helpful to know what you want to do. Each extension basically has `data` which can be grabbed with the `get_data()` function. Some extensions have extra functionality because it is useful. But I think you're right that we don't document that super well. Currently reading the source code is the best way to know if something has extra functionality. Maybe it is worth us having something better. You could see if it is being written in the [api docs](https://spikeinterface.readthedocs.io/en/stable/api.html), but that is pretty dense. 

> 3.) It appears that the sorter I am using (MountainSort5) is picking up the hyper polarization that precedes an action potential as the center of the waveform (see picture). This is causing, [...]

This is the type of question that is more for the spike sorters rather than spikeinterface. We have curation machinery for merging units that you think have been split incorrectly (either computationally or manually), but if you're hoping to prevent this splitting you would need to tune the spike sorting parameters. And that is a question that should go to the creator of the sorter. In this case the dev of MS5 Jeremy. 

Does all of this make sense?

JacobKelley101, 2025-01-30:
Hi @zm711 

Thank you very much for your thoughtful responses. For my first question, I do not have manual time stamps for removing my artifacts unfortunately, and I have lots of data I need to process. It sounds like I just need to do this part myself in MATLAB or Python, autodetecting egregious threshold crossing and replacing that point + the surrounding area with white noise. Would you agree? 

For the extensions, I appreciate at least knowing there is not a better way to do what I have set out to do. I am working my way through it, using commands like print(waveform_extension.__dict__) to discern when extensions have extra retrieval functions. Then I just search for them in the ReadTheDocs search bar, and they normally have examples. I will try your recommendation of perusing the source code for further information.

For the sorter question, I will reach out to the MS5 developers. For what it is worth, I did try resorting with a higher threshold and it did recenter my waveforms, but I lost a lower amplitude unit as I suspected I would. Our use case is tricky, as we are only expecting to get 2-4 usable units for every batch of data, so losing one means much more than it would to others I think. 

I had another question regarding some SI quality metrics. Our lab previously used IgorPro, developed by Christophe Pouzat (referenced in many places throughout SI documentation), to spike-sort our data. Now that we are transitioning to MS5/SI, we want to recreate some of the quality metrics used in IgorPro in Spike Interface. The problem is that many of the SI metrics are calculated differently. One parameter we would really like to have is the standard deviation of noise after all the detected spikes have been removed from the recording. I see in the Standard Deviation (SD) ratio quality metric that this parameter is calculated, but I do not know how to extract it by itself. Is there any way to do this? If not, do you have any recommendation on how I can calculate this parameter inside the SI ecosystem myself? We want to use it to calculate an isolation-distance metric (called "The Projection Test" in the Pouzat paper), and compare our results to the isolation distance metric calculated in a different way by SI. 

Thank you again for all the advice, it is deeply appreciated.

JacobKelley101, 2025-01-30:
@magland 

Do you have any advice regarding my centering problem with MS5? I can provide more details if it would help.

zm711, 2025-01-30:
> Thank you very much for your thoughtful responses. For my first question, I do not have manual time stamps for removing my artifacts unfortunately, and I have lots of data I need to process. It [...]

You could come up with some sort of detection threshold and then return the indices and use remove_artifact to fix it. Alessio has a tutorial [here](https://github.com/SpikeInterface/SpikeInterface-Training-Edinburgh-May24). I forget the name that has it, but in the tutorial he demonstrates detecting opto-artifacts with a "homemade" script and then feeding those times into our spikeinterface functions. Might be less work for you. So check it out and maybe you can use that code to build yours!

> For the extensions, I appreciate at least knowing there is not a better way to do what I have set out to do. I am working my way through it, using commands like print(waveform_extension.dict) to [...]

Yep that works. I'll suggest we think about how/if we want to document these!

> I see in the Standard Deviation (SD) ratio quality metric that this parameter is calculated, but I do not know how to extract it by itself. Is there any way to do this? If not, do you have any recommendation on how I can calculate this parameter inside the SI ecosystem myself?

If the metric exists and you don't want to calculate everything then when you do `analyzer.compute('quality_metrics', metric_names = ['sd_ratio'])` then only that will be calculated. 

I'm not sure about the other part of your question. I understand that we probably calculate things differently but are you saying you want to calculate the sd ratio differently than how we have done it? You definitely could make your own quality metrics, but then you would have to access some of our lower-level functions which although totally possible might be a little confusing for a beginner. 

Is your concern that you want to be able to compare "old" qc-ed data and your newer "si" qc'ed data? 

As far as the Projection Test-- I'm not familiar with that one. But you can also access the PCs and make your own metric that way. But it will be lower level functions. So the best advice there is look at our source code for how we normally do PC metrics and then you can build your own metric using our current metrics as templates!

JacobKelley101, 2025-01-30:
@zm711 

What I mean is that the SD_ratio quality metric is calculated by dividing the std_dev of the peak spike amplitude by the std_dev of the background noise (see picture). What I want is just the std_dev of the background noise (the denominator of the SI quality metric) so I can manually calculate some other parameters from IgorPro. Essentially yes, I do want to compare old qc-ed data and SI qc'ed data. We are trying to establish new inclusion criteria for adding units to our datasets, and the SI quality metrics available to us are not directly comparable to what we used before.

This particular parameter, the std_dev of noise, would be particularly useful for our lab to have, as it is the basis for 2/4 of the quality metrics described in the Pouzat paper. Comparing signal-to-noise to quantitatively describe unit quality/isolation is the core subject of the Pouzat paper referenced in the SD_Ratio quality metric documentation, and was explicitly designed to curate tetrode data collected from the same model organism (the locust) that our lab uses. Since SI systematically does not load all of the data into memory unless it has to, manually calculating this parameter inside the SI ecosystem is challenging. But, I see that it is being calculated in this quality metric, so I thought it must be generated and accessible in some way, at some point. Any thoughts/suggestions?

Thank you for your continued support Zach

[image]

zm711, 2025-01-30:
Howdy about our noise levels
https://github.com/SpikeInterface/spikeinterface/blob/cbe54712b1041597d86cc99e365d8a495fec62dd/src/spikeinterface/core/analyzer_extension_core.py#L683-L705

If you wanted the std of the noise levels then you could play around with these values?

JacobKelley101, 2025-01-30:
This metric was the closest I had seen to what I wanted, but is not exactly the same. The key point of the std_dev metric calculated as the denominator of the SD_Ratio metric is that it clips-out all the detected waveforms before it is calculated. This noise level metric is calculated using the Median Absolute Deviation (MAD) formula, which to the best of my knowledge does not clip-out detected spikes with its implementation in SI. Would you have any idea if the MAD std_dev would be analogous to the std_dev calculated as the denominator of the SD_Ratio function?

zm711, 2025-01-30:
For the SD_ratio the implementation strategy was discussed here https://github.com/SpikeInterface/spikeinterface/pull/2146, where there is a quick explanation of why there was divergence from the paper. 

this does not cut out waveforms to my knowledge, but @samuelgarcia can comment more on the calculation of noise_levels. You could always try to use this with the waveforms and only worry about the waveforms if you find that it is not agreeing with your previous results. Or re-run some old data through SI and try to see if there is a clear correlation between old metrics and our metrics.

JacobKelley101, 2025-01-30:
Thank you for the suggestion, I will wait for @samuelgarcia to respond. One alternative I had thought of was just doing this calculation myself.  I know I can load the traces into memory (which you don't recommend, I know) using the get_traces function from the extractors module. I also obviously have all of the spiking events, the only thing I was unsure of was how I can assess which units correspond to which tetrodes, so when I go to clip-out my waveforms I don't clip-out data I shouldn't. Rather, units calculated from my first tetrode only clip-out data from those 4 channels, etc, etc. Any thoughts on if/how I should do this? I am very thankful for your continued correspondence.

zm711, 2025-01-30:
I think that depends on how your tetrode data is set up. You can use `unit_locations` to know where each unit is and then delete out those waveforms for your analysis. The issue with `spike_locations` are that they are likely just a subsample of where your spikes appear so unit information is probably best to then choose which units go where and thus which spikes go where.

magland, 2025-01-30:
> [@magland](https://github.com/magland)
> 
> Do you have any advice regarding my centering problem with MS5? I can provide more details if it would help.

My only thought would be to automatically detect such a situation (by using something like the cross-correlograms) and then exclude the cluster with the smaller amplitude.

JacobKelley101, 2025-01-30:
@zm711

I don't need to clip-out spikes on a single-channel basis, just on a single-tetrode basis. I was more curious if units were ever assigned a group-label depending on the tetrode they came from, or if that information only exists if I use the run_sorter_by_property and run it on each group separately. I stopped doing this because it runs the sorter twice, ~doubling the amount of time it takes for me to sort my data, but I don't know if there is any other way for me to identify which units came from which tetrodes. Is your best recommendation the unit_locations capability?

@magland 

The cluster with smaller amplitude is normally still good data, often it comprises a large amount of data that is obviously associated with another. My fear merging these units, however, is that the spike timing will be misaligned, possibly artificially lowering the overall unit quality despite the unit being easily isolated. Have you ever seen this kind of problem before? How did others go about fixing it? Or is this not an issue you normally see. 

Thank you both for your responses

magland, 2025-01-30:
> The cluster with smaller amplitude is normally still good data, often it comprises a large amount of data that is obviously associated with another. My fear merging these units, however, is that the [...]

@JacobKelley101 I have seen this before, and I think it is a fairly common issue. Unfortunately I don't have a good solution.

JacobKelley101, 2025-01-30:
@magland 

Thank you for your honesty! I will work with my PI on a potential solution if required.

zm711, 2025-01-30:
> I don't need to clip-out spikes on a single-channel basis, just on a single-tetrode basis.

the unit localization would give you a location around a tetrode so you could choose to do a per channel or a per tetrode. 

>  was more curious if units were ever assigned a group-label depending on the tetrode they came from

many sorters use some sort of sparsity/distance metric (including MS5 if I'm not mistaken). So if your tetrodes are close enough then yes you could get cross-tetrode units. If the tetrodes are far enough away from each other than they should be isolated even if using `run_sorter` so that comes back to my earlier question of how your tetrodes are set up. 

> but I don't know if there is any other way for me to identify which units came from which tetrodes. Is your best recommendation the unit_locations capability?

Yeah I think so. In this case you'll get the location and should be able to figure out which tetrode a unit came from or if the unit was between tetrodes this should help figure that out.

JacobKelley101, 2025-01-30:
My tetrodes are intentionally setup to be outside the MS5 neighborhood radius, so I do not get any cross-tetrode units. I can tell from plotting the mean-waveforms which tetrode they came from based on the colors used, but I cannot find any index that states which unit corresponds to which channels. For example, units from tetrode1 get plotted in red, blue, orange, and green. Units from tetrode2 get plotted in purple, pink, brown, and grey. I don't understand how that information is getting passed through, all I am doing is the following: 

ext_wf = analyzer.get_extension("waveforms")
for unit_id in analyzer.unit_ids:
    wfs = ext_wf.get_waveforms_one_unit(unit_id)
    print(unit_id, ":", wfs.shape)

ext_templates = analyzer.get_extension("templates")

av_templates = ext_templates.get_data(operator="average")
print(av_templates.shape)

median_templates = ext_templates.get_data(operator="median")
print(median_templates.shape)

for unit_index, unit_id in enumerate(analyzer.unit_ids):
    fig, ax = plt.subplots()
    template = av_templates[unit_index]
    ax.plot(template)
    ax.set_title(f"{unit_id}")

If I had a way of tracking those units which were plotted in RGBO and PPBG I could use that info to clip-out waveforms from their respective traces. I just don't know how I am to extract that information, though it clearly exists in some capacity.

samuelgarcia, 2025-01-31:
Hi @JacobKelley101 
I  understand correctly you want a way to estimate the noise with some random chunk that do not contains spike for better accuray. Is it this ?
We do not have this in spike interface because for high channel count this is almost impossible statistically.

I used to have this kind of code somewhere in tridesclous and it was usefull for tetrode data (where finding noise without spikes is more easy).

The general approach is to used the MAD instead of STD so having spikes or not in chunks should affect too much the results.

I just read the our implementation `compute_sd_ratio()` done @DradeAW. It is using the STD.
Maybe you could make a PR for having the optionally MAD there.

Another possibility which is more tedious would be implement a spike-free random chunk retriever in spikeinterface but then it would hihly depend on a spike detection steps and there are many method for this (some method are able to find spikes in noise when multichannel for instance).

DradeAW, 2025-01-31:
I would like to point out that using STD rather than MAD in the `sd_ratio` in intentional!

When computing the STD of a spike, this spike sometimes collides with other spikes, so (in this case) you do not want to use the MAD

Maybe for other use cases it could be useful to use the MAD, it depends on what you want.
I just wanted to point out the logic behind using the STD there :)

JacobKelley101, 2025-01-31:
@samuelgarcia
Thank you very much for your advice! At this point I think I have a pretty good idea of what is/isn't available to me, and I will have a discussion with my PI regarding next steps. 

@DradeAW 
Thank you very much for your input as well. I understand the logic of using STD when calculating the deviation of spike amplitudes. What I am concerned with is calculating the STD of background noise to use as a basis for custom quality metrics, and comparison to old qc'ed data. In the SD_Ratio QC Metric, the denominator does the exact calculation I desire, but I cannot access the denominator. Any thoughts on how I can access that information, or recreate the calculation myself?

JacobKelley101, 2025-01-31:
@samuelgarcia @zm711 @DradeAW 

I've actually just taken a look at the spike interface API, trying to figure out if I could use some of the code in the SD_Ratio calculation and I think I have found a discrepancy in what is written in the ReadTheDocs description and the code. The noise levels used to calculate the SD_Ratio are taken from the get_noise_levels() function, which uses the MAD, and from what I can tell the spikes are not removed from the trace anywhere. Can someone help me understand how this metric is being calculated, and if I have made a mistake? I apologize for belaboring this point, but the more I look at it the more confused I become. The directory for the function is 
(see picture) with the SD_Ratio defined on line 1434

[image]

DradeAW, 2025-01-31:
The noise level uses `method="std"`
https://github.com/SpikeInterface/spikeinterface/blob/12a12768a82b51ddea7ed4a93ffba4747f518cea/src/spikeinterface/qualitymetrics/misc_metrics.py#L1499

The spike itself isn't removed because this is computationally too expensive,
This is however estimated here:
https://github.com/SpikeInterface/spikeinterface/blob/12a12768a82b51ddea7ed4a93ffba4747f518cea/src/spikeinterface/qualitymetrics/misc_metrics.py#L1504

JacobKelley101, 2025-01-31:
@DradeAW 

I see, thank you for clarifying. As a last request (I promise), could you briefly describe how this spike-removal estimation works? I read the documentation but can't fully wrap my head around how it is removing the impact of spikes.

DradeAW, 2025-02-01:
It's a bit complex and I don't have it on the top of my head right now,

But it is explained in our manuscript that we should publish to biorxiv very soon (end of February / March)

JacobKelley101, 2025-02-03:
@DradeAW 

Thank you very much, we will look for the manuscript on biorxiv. I will close this issue. Thank you so much for everyone who offered their expertise and advice! Being able to communicate with the SI Devs is such a blessing. 

Jacob