SpikeInterface/spikeinterface #3667: Expanding spike interfaces functions - extracting baseprocessing traces using list ranges [EMG]

jesusdpa1, 2025-02-03:
Hi!

I have been working on the following functions to allow for emg activity (onset/offset) detection and additional features using spike interface. 

- moving averager and low pass filter
- tkeo
- logic detector [rise_thru, fall_thru, rise_fall_thru]

now I am working on being able to extract based on the onset-offset detection the contractile period to be use for measurement calculation [time of contraction, "peak amplitude"] etc. 

Most of this was possible using the baseprocessor class but I am unable to apply a function within the detectorlogic recording that allows to both extract the indices of the activity (i.e np.where()) and extract a list of vector array for further use in functions like PCA for classification. I saw that there is a module that kind of do this but for spike sorting and I was wondering what would be a good approach to expand an create one for emg

```py
def get_activity_vector(traces: np.array):
    indices_onset = np.where(traces == 1)[0]
    indices_offset = np.where(traces == -1)[0]
    return np.array(list(zip(indices_onset, indices_offset)))

# %%
activity_vector = get_activity_vector(threshold_test)
# %%

def slice_recording(traces: np.array, fs, activity_vector, margin_ms=100.0):
    margin = int(margin_ms * fs / 1000.0)
    emg_activity = []
    for onset, offset in activity_vector:
        contraction_slice = traces[np.arange(onset - margin, offset + margin), :]
        emg_activity.append(contraction_slice)
    return emg_activity
```

```py
"""
module to compute logic over a time series

Rise thru
Fall thru
Peak,
Valley,
etc...
"""

import numpy as np
from spikeinterface.core import get_chunk_with_margin
from spikeinterface.core.core_tools import define_function_from_class
from spikeinterface.preprocessing.basepreprocessor import (
    BasePreprocessor,
    BasePreprocessorSegment,
)

def fix_dtype(recording, dtype):
    if dtype is None:
        dtype = recording.get_dtype()
    dtype = np.dtype(dtype)

    # if uint --> force int
    if dtype.kind == "u":
        dtype = np.dtype(dtype.str.replace("u", "i"))

    return dtype

class DetectorLogicRecording(BasePreprocessor):
    """
    A preprocessor that applies a logic detector to the recording to identify features.

    Parameters
    ----------
    recording : Recording
        The recording extractor to be processed
    threshold_method: str,
    threhold_multiplier: float,
    detector_method: str, "rise_thru" | "fall_thru" | "rise_fall_thru" | "stay_above_thru" | "stay_below_thru", default: "rise_fall_thru"
    Type of detector
    margin_ms : float, default: 5.0
        Margin in ms on border to avoid border effect
    dtype : dtype or None, default: None
        The dtype of the returned traces. If None, the dtype of the parent recording is used
    add_reflect_padding : Bool, default False
        If True, uses a left and right margin during calculation.
    """

    def __init__(
        self,
        recording,
        threshold_method,
        threshold_multiplier,
        detector_method="rise_fall_thru",
        margin_ms=5.0,
        add_reflect_padding=False,
        dtype=None,
    ):
        fs = recording.get_sampling_frequency()

        dtype = fix_dtype(recording, dtype)

        BasePreprocessor.__init__(self, recording, dtype=dtype)
        self.annotate(is_detector=True)

        if "offset_to_uV" in self.get_property_keys():
            self.set_channel_offsets(0)

        margin = int(margin_ms * fs / 1000.0)
        for parent_segment in recording._recording_segments:
            self.add_recording_segment(
                DetectorLogicRecordingSegment(
                    parent_segment,
                    threshold_method,
                    threshold_multiplier,
                    detector_method,
                    margin,
                    dtype,
                    add_reflect_padding=add_reflect_padding,
                )
            )

        self._kwargs = dict(
            recording=recording,
            threshold_method=threshold_method,
            threshold_multiplier=threshold_multiplier,
            detector_method=detector_method,
            margin_ms=margin_ms,
            add_reflect_padding=add_reflect_padding,
            dtype=dtype.str,
        )

class DetectorLogicRecordingSegment(BasePreprocessorSegment):
    def __init__(
        self,
        parent_recording_segment,
        threshold_method,
        threshold_multiplier,
        detector_method,
        margin,
        dtype,
        add_reflect_padding=False,
    ):
        BasePreprocessorSegment.__init__(self, parent_recording_segment)
        self.threshold_method = threshold_method
        self.threshold_multiplier = threshold_multiplier
        self.detector_method = detector_method
        self.margin = margin
        self.add_reflect_padding = add_reflect_padding
        self.dtype = dtype

    def get_traces(self, start_frame, end_frame, channel_indices):
        traces_chunk, left_margin, right_margin = get_chunk_with_margin(
            self.parent_recording_segment,
            start_frame,
            end_frame,
            channel_indices,
            self.margin,
            add_reflect_padding=self.add_reflect_padding,
        )

        # Apply logic
        traces_dtype = traces_chunk.dtype
        # if uint --> force int
        if traces_dtype.kind == "u":
            traces_chunk = traces_chunk.astype("float32")

        threshold_traces = self._apply_threshold(traces_chunk)

        detector_traces = self._apply_detector(threshold_traces)

        if right_margin > 0:
            detector_traces = detector_traces[left_margin:-right_margin, :]
        else:
            detector_traces = detector_traces[left_margin:, :]

        if np.issubdtype(self.dtype, np.integer):
            detector_traces = detector_traces.round()

        return detector_traces.astype(self.dtype)

    def rise_thru(self, traces):
        diff_traces = np.diff(traces.astype(int), axis=0, prepend=False)
        return diff_traces == 1

    def fall_thru(self, traces):
        below_mean = ~traces
        diff_traces = np.diff(below_mean.astype(int), axis=0, prepend=False)
        return diff_traces.astype(int) == 1

    def rise_fall_thru(self, threshold_traces: np.ndarray) -> np.ndarray:
        rise_thru_traces = self.rise_thru(threshold_traces).astype(int)
        fall_thru_traces = -self.fall_thru(threshold_traces).astype(int)
        return rise_thru_traces + fall_thru_traces

    def stay_above_thru(self, threshold_traces: np.ndarray) -> np.ndarray:
        return threshold_traces.astype(int)

    def stay_below_thru(self, threshold_traces: np.ndarray) -> np.ndarray:
        return ~threshold_traces.astype(int)

    def _apply_threshold(self, traces: np.ndarray) -> np.ndarray:
        if self.threshold_method == "mean":
            return (
                traces
                >= np.mean(traces, axis=0, keepdims=True) * self.threshold_multiplier
            )
        if self.threshold_method == "median":
            return (
                traces
                >= np.median(traces, axis=0, keepdims=True) * self.threshold_multiplier
            )
        if self.threshold_method == "std":
            return (
                traces
                >= np.std(traces, axis=0, keepdims=True) * self.threshold_multiplier
            )
        else:
            raise ValueError(f"Unsupported threshold method: {self.threshold_method}")

    def _apply_detector(self, threshold_traces: np.ndarray) -> np.ndarray:
        if self.detector_method == "rise_thru":
            return self.rise_thru(threshold_traces)
        elif self.detector_method == "fall_thru":
            return self.fall_thru(threshold_traces)
        elif self.detector_method == "rise_fall_thru":
            return self.fall_rise_thru(threshold_traces)
        elif self.detector_method == "stay_above_thru":
            return self.stay_above_thru(threshold_traces)
        elif self.detector_method == "stay_below_thru":
            return self.stay_below_thru(threshold_traces)
        # elif self.detector_method == "peak_detection":
        #     return self.peak_detection(threshold_traces)
        else:
            raise ValueError(f"Unsupported detector method: {self.detector_method}")

detector_logic = define_function_from_class(
    source_class=DetectorLogicRecording, name="detector_logic"
)

```

yger, 2025-02-04:
I don't think your approach is the one I would have followed. In spikeinterface, if you want to apply an operation to all your chunks, on the fly, you should make of the recording_node pipeline that has been made by @samuelgarcia for the sorting components. All the PeakDetector, for example, are using such a syntax and you can implement your own node that will perform an operation on a chunk, when it will be loaded. A year ago, for example, in a slighly different context and to remove artefacts, I did a prototype that would detect onsets/offsets of an enveloppe of the signal

```
class DetectThresholdCrossing(PeakDetectorWrapper):
    
    name = "threshold_crossings"
    preferred_mp_context = None
    
    @classmethod
    def check_params(
        cls,
        recording,
        detect_threshold=5,
        noise_levels=None,
        random_chunk_kwargs={},
    ):
        if noise_levels is None:
            noise_levels = get_noise_levels(recording, return_scaled=False, **random_chunk_kwargs)
        abs_thresholds = noise_levels * detect_threshold
        return (abs_thresholds, )

    @classmethod
    def get_method_margin(cls, *args):
        return 0

    @classmethod
    def detect_peaks(cls, traces, abs_thresholds):
        z =  (traces - abs_thresholds).mean(1)
        threshold_mask = np.diff((z > 0) != 0)
        indices,  = np.where(threshold_mask)
        peak_sample_ind = np.zeros(0, dtype=np.int64)
        peak_chan_ind = np.zeros(0, dtype=np.int64)
        onsets = indices[:-1]
        offsets = indices[1:]
        peak_sample_ind = np.concatenate((peak_sample_ind, onsets))
        peak_chan_ind = np.concatenate((peak_chan_ind, np.zeros(len(onsets), dtype=np.int64)))
        peak_sample_ind = np.concatenate((peak_sample_ind, offsets))
        peak_chan_ind = np.concatenate((peak_chan_ind, np.ones(len(offsets), dtype=np.int64)))
        return peak_sample_ind, peak_chan_ind
```

Once this node is created, you could use it in the following manner, on your enveloppe (low-pass filtered signal):

``
def detect_artefacts(enveloppe, detect_threshold=5, **job_kwargs):

    from spikeinterface.core.node_pipeline import (
        run_node_pipeline,
    )

    node0 = DetectThresholdCrossing(enveloppe, **{'detect_threshold' : detect_threshold})
    
    times = run_node_pipeline(
        enveloppe,
        [node0],
        job_kwargs,
        job_name="detecting threshold crossings",
        )

    mask = times["channel_index"] == 0
    onsets = times[mask]

    mask = times["channel_index"] == 1
    offsets = times[mask]
    
    periods = []
    
    if onsets['sample_index'][0] > offsets['sample_index'][0]:
        periods += [(0, offsets['sample_index'][0])]
        offsets = offsets[1:]
    
    for i in range(len(onsets)):
        periods += [(onsets['sample_index'][i], offsets['sample_index'][i])]

    if len(onsets) > len(offsets):
        periods += [(onsets['sample_index'][0], enveloppe.get_num_samples())]            
    
    return periods
```

What this function does, in theory, it that it returns you the list [t_start, t_stop] of threshold crossings. Note that this was only a sketchy prototype, maybe it has to be refine, but I hope you get the spirit. The node pipeline is powerful because you can chain several node if you have any. They can each use the peaks/data from previous nodes, and the traces (or traces filtered by previous nodes). Let me know if questions

yger, 2025-02-04:
This is currently not properly documented, because a paper on the sortingcomponents in on the way, but documentation should catch up soon

jesusdpa1, 2025-02-04:
Thanks @yger! currently trying to test this, I went through the code base and found the following spikeinterface.core.node_pipeline

```py
class PipelineNode:
    def __init__(
        self,
        recording: BaseRecording,
        return_output: bool | tuple[bool] = True,
        parents: Optional[list[Type["PipelineNode"]]] = None,
    ):
        """
        This is a generic object that will make some computation on peaks given a buffer of traces.
        Typically used for exctrating features (amplitudes, localization, ...)

        A Node can optionally connect to other nodes with the parents and receive inputs from them.

        Parameters
        ----------
        recording : BaseRecording
            The recording object.
        return_output : bool or tuple[bool], default: True
            Whether or not the output of the node is returned by the pipeline.
            When a Node have several toutputs then this can be a tuple of bool
        parents : Optional[list[PipelineNode]], default: None
            Pass parents nodes to perform a previous computation.
        """

        self.recording = recording
        self.return_output = return_output
        if isinstance(parents, str):
            # only one parents is allowed
            parents = [parents]
        self.parents = parents

        self._kwargs = dict()

    def get_trace_margin(self):
        # can optionaly be overwritten
        return 0

    def get_dtype(self):
        raise NotImplementedError

    def compute(self, traces, start_frame, end_frame, segment_index, max_margin, *args):
        raise NotImplementedError
```

I am assuming the idea is to branch from here and implement the logic functions? something like 

```py

from __future__ import annotations

import struct
from pathlib import Path
from typing import Optional, Type

import numpy as np
from spikeinterface.core import (
    BaseRecording,
    get_chunk_with_margin,
)
from spikeinterface.core.job_tools import (
    ChunkRecordingExecutor,
    _shared_job_kwargs_doc,
    fix_job_kwargs,
)

base_emg_dtype = [
    ("sample_index", "int64"),
    ("channel_index", "int64"),
    ("amplitude", "float64"),
    ("segment_index", "int64"),
]

contraction_dtype = base_emg_dtype + [
    ("contraction_index", "int64"),
]

class PipelineNode:
    def __init__(
        self,
        recording: BaseRecording,
        return_output: bool | tuple[bool] = True,
        parents: Optional[list[Type["PipelineNode"]]] = None,
    ):
        """
        This is a generic object that will make some computation on peaks given a buffer of traces.
        Typically used for exctrating features (amplitudes, localization, ...)

        A Node can optionally connect to other nodes with the parents and receive inputs from them.

        Parameters
        ----------
        recording : BaseRecording
            The recording object.
        return_output : bool or tuple[bool], default: True
            Whether or not the output of the node is returned by the pipeline.
            When a Node have several toutputs then this can be a tuple of bool
        parents : Optional[list[PipelineNode]], default: None
            Pass parents nodes to perform a previous computation.
        """

        self.recording = recording
        self.return_output = return_output
        if isinstance(parents, str):
            # only one parents is allowed
            parents = [parents]
        self.parents = parents

        self._kwargs = dict()

    def get_trace_margin(self):
        # can optionaly be overwritten
        return 0

    def get_dtype(self):
        raise NotImplementedError

    def compute(self, traces, start_frame, end_frame, segment_index, max_margin, *args):
        raise NotImplementedError

```

if that is true my question would be regarding to the functions at the end of the module, specifically `run_node_pipeline()` since this one seems to be specifically written for spike extraction, would this be another one that would need to be rewritten?

yger, 2025-02-04:
Just to be clear, I am not saying you should do that the way I'm doing, just that we need to think about it. While you are right that currently these nodes are mostly tailored for spike sorting, there are no conceptual reasons they should stay like that. The good point is that there is a whole parallelized machinery to process everyghing quickly, so if your function can be applied to chunks, then it is good to try to use it. To detect onset/offsets, i.e. threshold crossings, I think this is doable, and as shown in my previous example, I made a custom function (internally using the run_node_pipeline) to gather everything and return what I wanted. Up to you to decide what you want to do exactly, and we can discuss about that. In the node framework, a worker is only aware of the local information of his chunk, so you'll have to handle border effects and concatenate all results at the end if you want to have a global view over all your recording. However, if you just want to chain operations, i.e. launch something when traces are entering a particular regime (i.e. after a threshold crossing, for example for PSTH, or whatever), then you can think about chaining several nodes. There are several nodes types PeakDetecor (that are detecting times), and WaveformNode (that are performing operations on the traces). We can discuss more if needed

yger, 2025-02-04:
Are you working from main (latest git version?) ? Because in recent version, this extra argument allows you, if you want, to explicitly specify a list of time slices that should be processed

jesusdpa1, 2025-02-04:
Currently running "spikeinterface>=0.101.2", and makes sense,

I think certain steps still would make sense to have them implemented like the filter and tkeo transformations but I agree that the logic block could be updated to use the approach you are suggesting since that one is mostly to do onset/offset detection and extracting the waveforms to do post processing analysis (PCA classification, burst comparisons) and would benefit from the parallel processing. 

Also I think I should have started with testing directly the code you provided 😅 I think I got confuse when I saw the name "peak".  

Testing the code you provided, just to see if I understand, when the values are return because this is done in parallel, all the data is brought back in 1d array? Therefore, to get back for example, the index for each burst for each channel I would need to re-arange the array using the channel index as the grouping factor and the order of the sequence as the indicators of which are onsets and which are offsets? 

[image]

Lastly, when aiming to use for example the PCA options would you have any recommendations on how to handle differences in sizes of the bursts. i.e does spikeinterface appends values to the start and end of the extracted waveform to make them all the same size
```py
burst01 [100, 200]
burst02 [200, 250]
....
```

yger, 2025-02-05:
I should have said that indeed, the draft code I gave you was made for detecting onsets/offsets common to all electrodes. But there is no reason why you would not be able to extend it on a per channel basis, indeed returning also the peak_channel_indices. I'll try to make the code during the day if I have time, to share it with you. 

Regarding PCA, this is a bit more tricky. SpikeInterface can indeed extract a given temporal period after a "peak", i.e. an event that has been detected. In our case, this is a waveform, but it could be your burst. This is a WaveformNode that would need to be plugged in after the peak detection. However, these waveforms have the same duration. So if you have heterogeneous onsets/offsets, I think you will have to do it externally, outside your pipeline (or assume a max value). PCA, in any way, will need samples of the same duration to be processed. So not sure how you'll deal with heterogeneous durations. Currently, SpikeInterface is not filling anything, because when there is a spike, we extract always the same temporal window around it. Note that if you have pre-extracted events and snippets, you can precompute the PCA model, and that is the purpose of the TemporalPCANode. Given a model, each time an event is detected, internally the waveforms is extracted, projected given the model, and the nodes returns the PCA components. This compress a lot the information that has to be stored. This is what is used for sorting pipelines.

I'll try to draft your onset/offset detector, I think it could be useful as a node in SI in general

jesusdpa1, 2025-02-05:
Thanks @yger!! I'll test this later this week,

closed 2025-02-06 by jesusdpa1