SpikeInterface/spikeinterface #3673: How attached probes are handled for `concatenate_recordings`

JoeZiminski, 2025-02-06:
Just a question as I'm curious, I'm loving the `concatenate_recordings` function it works so nicely. I have some recordings each with a probe attached, and it handles the probe neatly (the resultant concatenated recording has the probe) and I was wondering how this is handled under the hood. But in the [relevant module](https://github.com/SpikeInterface/spikeinterface/blob/b22fc3774108dea1a7647a0f794acda43cb6af92/src/spikeinterface/core/segmentutils.py#L89) I could not find any handling of the probe. How does this work?

alejoe91, 2025-02-06:
Hey Joe! 

Under the hood, the concatenate recordings uses the append recordings, which, after relevant checks, copies the metadata (including probe info) from the [first recording](https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/core/segmentutils.py#L64)

JoeZiminski, 2025-02-06:
Cheers @alejoe91! that's great

closed 2025-02-06 by JoeZiminski