SpikeInterface/spikeinterface #3675: How to easily extract spike times related to a recording chunk.

Djoels, 2025-02-06:
Just a question: how can one easily create selection of a time fragment along a recording.
For the recording you can make a call to select_channels and time_slice, to get a chunk of the recording.
However when trying to extract the affiliated spike times to this time_slice, I find I'm having to play with the sampling frequency (given that I don't have a time vector).

Am I missing some elegant way of doing this?

zm711, 2025-02-06:
I always just do this myself by converting to samples from time (as you said using the sampling_frequency), but maybe @alejoe91 or @samuelgarcia know of a machinery in spikeinterface to easily do this (in which case I might switch myself :) )

chrishalcrow, 2025-02-07:
Hello! There's a few ways to look at spike trains: `to_spike_vector`, `get_unit_spike_train` and I recently found `spike_vector_to_spike_train`(!). Of these, the `get_unit_spike_train` method has a `return_times` parameter. So if you have a sorting object called `sort` you can do something like

``` python
unit_id = 5
unit_spike_train = sort.get_unit_spike_train(unit_id = unit_id, return_times=True)

t_start = 1
t_end = 4
interval_train = unit_spike_train[(1 < unit_spike_train) & (unit_spike_train < 4)]
```

But if you're working with the `to_spike_vector` output, you'll need to deal with the sampling_frequency yourself.

alejoe91, 2025-02-07:
Another option is to use the `sorting.frame_slice()` to get a new sorting object. I think we could add a `time_slice`, which automatically does the conversion/checks the registered recording for timestamps. 

@chrishalcrow can you give it a try?

chrishalcrow, 2025-02-10:
Hey @Djoels , thanks to your comment we've just added a new `time_slice` method for the sorting object, so you can do things like

```
reduced_spike_vector = sorting.time_slice(start_time=60, end_time=120).to_spike_vector()
```

The `reduced_spike_vector`  will still give you the spike times using sample indices, so you'll need to convert at the end, but hopefully this will make life easier!

Djoels, 2025-02-10:
Wow, thank you very much for this!