SpikeInterface/spikeinterface #3679: Assert that at least one unit is included when creating a sorting analyzer ?

b-grimaud, 2025-02-10:
I'm iterating over clustering parameters to try to find the best settings for my data.

To that end, I have a pipeline that goes : Clustering (HS2) -> Sorting Extractor -> Remove units with <10 spikes -> Sorting Analyzer -> Quality Metrics

Under some clustering conditions, this can result on sorting objects with no units left. This then results in a crash when computing the first extension, in this case random spikes :

zm711, 2025-02-11:
I think this makes sense, but I guess we could also just say this is the user's responsibility to check their sorting themselves before using it. I think this is a @alejoe91 and @samuelgarcia level decision.

alejoe91, 2025-02-13:
@b-grimaud we discussed about this and we think that a sorting analyzer with no units is still a valid analyzer so we should deal with it.

@chrishalcrow will help making sure this is properly supported!

zm711, 2025-02-14:
Just for the example we discussed so there is a record if we need to go back and search:

What if you do a multishank recording and 3/4 shanks had good units, but one shank did not (maybe you missed your target region or you damaged the tissue leading to low spike counts). You probably wouldn't want your work flow to error out as you try to work with the 4 analyzers potentially remerging the data at the end for final saving of files to disk.

closed 2026-09-03 by alejoe91