SpikeInterface/spikeinterface #3688: channel locations missing from zarr saved files

kshtjkumar, 2025-02-12:
Hi am saving the .rhs files in zarr format as:
```

recording = se.read_intan(file_path, stream_name='RHS2000 amplifier channel', use_names_as_ids=True)
recording_bp = spre.bandpass_filter(recording, freq_min=300, freq_max=9000)
rec_cmr = spre.common_reference(recording_bp, operator="median", reference="global")  # Rereferencing the data

probe = generate_multi_columns_probe(num_columns=3,
                       num_contact_per_column=[1, 2, 1],
                       xpitch=75, ypitch=150, y_shift_per_column=[0, -60, 0],
                       contact_shapes='square', contact_shape_params={'width' : 20 , 'height' :20 })

channel_indices = np.arange(4)
probe.set_device_channel_indices(channel_indices)
rec_cmr.set_probe(probe, in_place = True)
rec_cmr.get_channel_locations()

bad_channel_ids, channel_labels = spre.detect_bad_channels(rec_cmr, method='coherence+psd')
print('bad_channel_ids', bad_channel_ids)
print('channel_labels', channel_labels)

recording_good_channels_f = rec_cmr.remove_channels(bad_channel_ids)
recording_good_channels = spre.common_reference(recording_good_channels_f, reference='global', operator='median')

base_name = Path(file_name).stem
zarr_path = zarr_folder / f"{base_name}.zarr"

compressor = numcodecs.Blosc(cname="zstd", clevel=9, shuffle=numcodecs.Blosc.BITSHUFFLE)
recording_good_channels.save(format="zarr", folder=zarr_path, compressor=compressor, **job_kwargs)
```

and when i read the zarr file using:

`se.read_zarr`,  the channel locations are missing!

alejoe91, 2025-02-13:
@kshtjkumar can you make sure you're using the latest spikeinterface version?

kshtjkumar, 2025-02-13:
well I just upgraded! but i was wondering do i have to reconvert those to zarr files again ? or the new version's load analyser function will take care of the error ?