SpikeInterface/spikeinterface #3695: Add electrode ID for each unit to NWB Units table from MDA sorting

rly, 2025-02-15:
I am using spikeinterface and NeuroConv to convert a firings.mda sorting file from an older version of MountainSort to an NWB file. Currently this creates an NWB Units table with spike times and unit labels from MountainSort, but the table lacks information about which channel a unit was maximally detected on, which is available in the MDA file. NWB has an "electrodes" column in the Units table to store that.

`MdaSortingSegment._max_channels` (https://github.com/SpikeInterface/spikeinterface/blob/dd0c6d33f98f071362fa06abb631a7629b97de58/src/spikeinterface/extractors/mdaextractors.py) stores channel/electrode IDs for each spike event from the MDA file but those are not used anywhere.

In the MDA files that I have, for each unit, every spike assigned to that unit has the same max channel index. Is that always the case, @magland?

If so, I propose making `MdaSortingSegment._max_channels` public, as in drop the leading underscore in the variable name, so that downstream tools, e.g., NeuroConv, can use that information, e.g., when writing the sorting to an NWB file.

Does that sound reasonable? I am not an expert in spikeinterface and neuroconv. If there is a better approach, please let me know. Thanks!

Reference for firings.mda file format: https://mountainsort.readthedocs.io/en/latest/first_sort.html#format-of-the-firings-mda

rly, 2025-02-15:
Also, if I have only one `firings.mda` file for a session, and I am not concatenating across sessions, should there always be only one `MdaSortingSegment` per `MdaSortingExtractor`?

alejoe91, 2025-02-15:
Hi @rly 

Second question first: there is always a Segment. That's the class in charge of retrieving data. Most Extractors are in fact mono-segment, but this organization allows us also to seamlessly deal with multi-segments.

First question:
Currently, there is no way in SI to attach spike vectors to spike trains. Since max channels are defined for each spike and might change over time, we are not attaching it to the object. 
If you need a "mean max channel", the easiest option is to get the max channel with the largest count for each unit an use the `set_property`. Properties are automatically saved as NWB unit columns by neuroconv. 
I wouldn't save them to the `electrodes` column though by default. We normally use that column when we also save waflceform means and stds to point to the electrodes table.

magland, 2025-02-17:
> In the MDA files that I have, for each unit, every spike assigned to that unit has the same max channel index. Is that always the case, @magland?

Yes, that is always the case. @rly

alejoe91, 2025-02-17:
Ok, then we can just add max channel as a property :)

PR #3701 (2025-02-24): Set max_channel property of MdaSortingExtractor
Fix #3695
closed 2025-02-24 by alejoe91