SpikeInterface/spikeinterface #3698: AttributeError: 'NoneType' object has no attribute 'split' in spikeinterface 0.97.0

ZoeChen96, 2025-02-17:
Hi, I am using spikeinterface to do spike sorting based on dataset PAIRED_KAMPFF. And I got the following errors (while ignored because of exceptions). Do you have any clues on it? The threadpoolctl is with version  2.2.0

```
AttributeError: 'NoneType' object has no attribute 'split'
    self.version = self.get_version()
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/threadpoolctl.py", line 646, in get_version
    config = get_config().split()
AttributeError: 'NoneType' object has no attribute 'split'
Exception ignored on calling ctypes callback function: <function _ThreadpoolInfo._find_modules_with_dl_iterate_phdr.<locals>.match_module_callback at 0x15193c0e8c10>
Traceback (most recent call last):
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/threadpoolctl.py", line 400, in match_module_callback
    self._make_module_from_path(filepath)
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/threadpoolctl.py", line 515, in _make_module_from_path
    module = module_class(filepath, prefix, user_api, internal_api)
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/threadpoolctl.py", line 606, in __init__
    self.version = self.get_version()
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/threadpoolctl.py", line 646, in get_version
    config = get_config().split()
AttributeError: 'NoneType' object has no attribute 'split'
```

zm711, 2025-02-17:
Hey @ZoeChen96, we are onto version 0.102.0 so it is likely because you aren't synced up between all the packages ( I vaguely remember this being a problem ages ago). I think the issue here is just that there have been so many bug fixes and package updates between 97 and 102 that it would be hard to diagnose the exact problem Our rec would be to try to update. We had a big API switch over in 101, so you could try to update to 100 to see if that fixes your issues and if not then I would recommend updating to 102.

alejoe91, 2025-02-17:
Can you also paste the entire traceback?

ZoeChen96, 2025-02-17:
> Can you also paste the entire traceback?

Loaded 1946 spikes.
Fitting dimensionality reduction using all spikes...
...projecting...
...done
Clustering...
Clustering 1946 spikes...
number of seeds: 21
seeds/job: 22
using 1 cpus
[Parallel(n_jobs=1)]: Using backend SequentialBackend with 1 concurrent workers.
[Parallel(n_jobs=1)]: Done   1 out of   1 | elapsed:    0.1s remaining:    0.0s
[Parallel(n_jobs=1)]: Done   1 out of   1 | elapsed:    0.1s finished
/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/herdingspikes/clustering/mean_shift_.py:242: DeprecationWarning: `np.bool` is a deprecated alias for the builtin `bool`. To silence this warning, use `bool` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.bool_` here.
Deprecated in NumPy 1.20; for more details and guidance: https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations
  unique = np.ones(len(sorted_centers), dtype=np.bool)
Exception ignored on calling ctypes callback function: <function _ThreadpoolInfo._find_modules_with_dl_iterate_phdr.<locals>.match_module_callback at 0x1527a317a040>
Traceback (most recent call last):
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/threadpoolctl.py", line 400, in match_module_callback
    self._make_module_from_path(filepath)
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/threadpoolctl.py", line 515, in _make_module_from_path
    module = module_class(filepath, prefix, user_api, internal_api)
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/threadpoolctl.py", line 606, in __init__
    self.version = self.get_version()
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/threadpoolctl.py", line 646, in get_version
    config = get_config().split()
AttributeError: 'NoneType' object has no attribute 'split'
Exception ignored on calling ctypes callback function: <function _ThreadpoolInfo._find_modules_with_dl_iterate_phdr.<locals>.match_module_callback at 0x1527a317a040>
Traceback (most recent call last):
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/threadpoolctl.py", line 400, in match_module_callback
    self._make_module_from_path(filepath)
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/threadpoolctl.py", line 515, in _make_module_from_path
    module = module_class(filepath, prefix, user_api, internal_api)
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/threadpoolctl.py", line 606, in __init__
    self.version = self.get_version()
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/threadpoolctl.py", line 646, in get_version
    config = get_config().split()
AttributeError: 'NoneType' object has no attribute 'split'
Exception ignored on calling ctypes callback function: <function _ThreadpoolInfo._find_modules_with_dl_iterate_phdr.<locals>.match_module_callback at 0x1527a317a040>
Traceback (most recent call last):
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/threadpoolctl.py", line 400, in match_module_callback
    self._make_module_from_path(filepath)
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/threadpoolctl.py", line 515, in _make_module_from_path
    module = module_class(filepath, prefix, user_api, internal_api)
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/threadpoolctl.py", line 606, in __init__
    self.version = self.get_version()
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/threadpoolctl.py", line 646, in get_version
    config = get_config().split()
AttributeError: 'NoneType' object has no attribute 'split'
Exception ignored on calling ctypes callback function: <function _ThreadpoolInfo._find_modules_with_dl_iterate_phdr.<locals>.match_module_callback at 0x1527a317a040>
Traceback (most recent call last):
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/threadpoolctl.py", line 400, in match_module_callback
    self._make_module_from_path(filepath)
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/threadpoolctl.py", line 515, in _make_module_from_path
    module = module_class(filepath, prefix, user_api, internal_api)
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/threadpoolctl.py", line 606, in __init__
    self.version = self.get_version()
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/threadpoolctl.py", line 646, in get_version
    config = get_config().split()
AttributeError: 'NoneType' object has no attribute 'split'
Exception ignored on calling ctypes callback function: <function _ThreadpoolInfo._find_modules_with_dl_iterate_phdr.<locals>.match_module_callback at 0x1527a317a040>
Traceback (most recent call last):
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/threadpoolctl.py", line 400, in match_module_callback
    self._make_module_from_path(filepath)
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/threadpoolctl.py", line 515, in _make_module_from_path
[... cut ...]
Deprecated in NumPy 1.20; for more details and guidance: https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations
  labels = np.zeros(n_samples, dtype=np.int)
Number of estimated units: 5
no filter, set =  kampff , Hungrian 0.0 accuracy =  0.087248322147651 ,best match 0.1 accuracy =  0.0
no filter, set =  kampff , best match 0.1 summary(well/bad/fp/overmerged/redundant) =  [0]
-----------HS, finish--------
> /imec/other/phd4yzhu/users/chen14/lcmacaw/cfg/data/journal_rev_lib/scripts/kampff_plot.py(109)sorter_loop_filt()

ZoeChen96, 2025-03-04:
OK，thanks a lot!

closed 2025-03-04 by ZoeChen96