SpikeInterface/spikeinterface #3699: Zarr `load_sorting_analyzer` trying to use non-existent sorting folder

chrishalcrow, 2025-02-17:
Made a sorting analyzer; and moved it. It now tries to find the sorting output folder, but can't and errors. So

``` python
import spikeinterface.full as si
sa = si.load_sorting_analyzer("/Users/chris/zarr_sorting_analyzer.zarr", load_extensions=False)
sa.select_units(unit_ids=[3,4])
```

gives the error:

```
---------------------------------------------------------------------------
FileNotFoundError                         Traceback (most recent call last)
Cell In[1], line 3
      1 import spikeinterface.full as si
      2 sa = si.load_sorting_analyzer("/Users/chris/ks_3.zarr", load_extensions=False)
----> 3 sa.select_units(unit_ids=[3,4])

File ~/Work/Edinburgh/Spike/si_dev/spikeinterface_dev/spikeinterface/src/spikeinterface/core/sortinganalyzer.py:1057, in SortingAnalyzer.select_units(self, unit_ids, format, folder)
   1055 if format == "zarr":
   1056     folder = clean_zarr_folder_name(folder)
-> 1057 return self._save_or_select_or_merge(format=format, folder=folder, unit_ids=unit_ids)

File ~/Work/Edinburgh/Spike/si_dev/spikeinterface_dev/spikeinterface/src/spikeinterface/core/sortinganalyzer.py:911, in SortingAnalyzer._save_or_select_or_merge(self, format, folder, unit_ids, merge_unit_groups, censor_ms, merging_mode, sparsity_overlap, verbose, new_unit_ids, backend_options, **job_kwargs)
    908     sparsity = None
    910 # Note that the sorting is a copy we need to go back to the orginal sorting (if available)
--> 911 sorting_provenance = self.get_sorting_provenance()
    912 if sorting_provenance is None:
    913     # if the original sorting object is not available anymore (kilosort folder deleted, ....), take the copy
    914     sorting_provenance = self.sorting

File ~/Work/Edinburgh/Spike/si_dev/spikeinterface_dev/spikeinterface/src/spikeinterface/core/sortinganalyzer.py:1261, in SortingAnalyzer.get_sorting_provenance(self)
   1259 if "sorting_provenance" in zarr_root.keys():
   1260     sort_dict = zarr_root["sorting_provenance"][0]
-> 1261     sorting_provenance = load(sort_dict, base_folder=self.folder)
   1262 else:
   1263     sorting_provenance = None

File ~/Work/Edinburgh/Spike/si_dev/spikeinterface_dev/spikeinterface/src/spikeinterface/core/loading.py:73, in load(file_or_folder_or_dict, **kwargs)
     71     if object_type is None:
     72         raise ValueError(_error_msg.format(file_path="provided dictionary"))
---> 73     return _load_object_from_dict(d, object_type, base_folder=base_folder)
     75 is_local = not is_path_remote(file_or_folder_or_dict)
     76 if is_local:

File ~/Work/Edinburgh/Spike/si_dev/spikeinterface_dev/spikeinterface/src/spikeinterface/core/loading.py:175, in _load_object_from_dict(d, object_type, base_folder)
    173 def _load_object_from_dict(d, object_type, base_folder=None):
    174     if object_type in ("Recording", "Sorting", "Recording|Sorting"):
--> 175         return BaseExtractor.from_dict(d, base_folder=base_folder)
    177     elif object_type == "Templates":
    178         from spikeinterface.core import Templates

File ~/Work/Edinburgh/Spike/si_dev/spikeinterface_dev/spikeinterface/src/spikeinterface/core/base.py:534, in BaseExtractor.from_dict(dictionary, base_folder)
    532     assert base_folder is not None, "When  relative_paths=True, need to provide base_folder"
    533     dictionary = make_paths_absolute(dictionary, base_folder)
--> 534 extractor = _load_extractor_from_dict(dictionary)
    535 folder_metadata = dictionary.get("folder_metadata", None)
    536 if folder_metadata is not None:

File ~/Work/Edinburgh/Spike/si_dev/spikeinterface_dev/spikeinterface/src/spikeinterface/core/base.py:1075, in _load_extractor_from_dict(dic)
   1073     new_kwargs[name] = {k: transform_dict_to_extractor(v) for k, v in value.items()}
   1074 elif isinstance(value, list):
-> 1075     new_kwargs[name] = [transform_dict_to_extractor(e) for e in value]
   1076 else:
   1077     new_kwargs[name] = value

File ~/Work/Edinburgh/Spike/si_dev/spikeinterface_dev/spikeinterface/src/spikeinterface/core/base.py:1068, in _load_extractor_from_dict.<locals>.<lambda>(x)
   1066 # Create new kwargs to avoid modifying the original dict["kwargs"]
   1067 new_kwargs = dict()
-> 1068 transform_dict_to_extractor = lambda x: _load_extractor_from_dict(x) if is_dict_extractor(x) else x
   1069 for name, value in dic["kwargs"].items():
   1070     if is_dict_extractor(value):

File ~/Work/Edinburgh/Spike/si_dev/spikeinterface_dev/spikeinterface/src/spikeinterface/core/base.py:1090, in _load_extractor_from_dict(dic)
   1084     warnings.warn(
   1085         f"Versions are not the same. This might lead to compatibility errors. "
   1086         f"Using {class_name.split('.')[0]}=={dic['version']} is recommended"
   1087     )
   1089 # Initialize the extractor
-> 1090 extractor = extractor_class(**new_kwargs)
   1092 extractor._annotations.update(dic["annotations"])
   1093 for k, v in dic["properties"].items():

File ~/Work/Edinburgh/Spike/si_dev/spikeinterface_dev/spikeinterface/src/spikeinterface/extractors/phykilosortextractors.py:265, in KiloSortSortingExtractor.__init__(self, folder_path, keep_good_only, remove_empty_units)
    264 def __init__(self, folder_path: Path | str, keep_good_only: bool = False, remove_empty_units: bool = True):
--> 265     BasePhyKilosortSortingExtractor.__init__(
    266         self,
    267         folder_path,
    268         exclude_cluster_groups=None,
    269         keep_good_only=keep_good_only,
    270         remove_empty_units=remove_empty_units,
    271     )
    273     self._kwargs = {"folder_path": str(Path(folder_path).absolute()), "keep_good_only": keep_good_only}

File ~/Work/Edinburgh/Spike/si_dev/spikeinterface_dev/spikeinterface/src/spikeinterface/extractors/phykilosortextractors.py:47, in BasePhyKilosortSortingExtractor.__init__(self, folder_path, exclude_cluster_groups, keep_good_only, remove_empty_units, load_all_cluster_properties)
     44     raise ImportError(self.installation_mesg)
     46 phy_folder = Path(folder_path)
---> 47 spike_times = np.load(phy_folder / "spike_times.npy").astype(int)
     49 if (phy_folder / "spike_clusters.npy").is_file():
     50     spike_clusters = np.load(phy_folder / "spike_clusters.npy")

File ~/.my_venvs/intro_to_si/lib/python3.12/site-packages/numpy/lib/npyio.py:427, in load(file, mmap_mode, allow_pickle, fix_imports, encoding, max_header_size)
    425     own_fid = False
    426 else:
--> 427     fid = stack.enter_context(open(os_fspath(file), "rb"))
    428     own_fid = True
    430 # Code to distinguish from NumPy binary files and pickles.

FileNotFoundError: [Errno 2] No such file or directory: '../kilosort4_sort_3/0/sorter_output/spike_times.npy'
```

Replacing https://github.com/SpikeInterface/spikeinterface/blob/dd0c6d33f98f071362fa06abb631a7629b97de58/src/spikeinterface/core/sortinganalyzer.py#L1191
with a try/except works, but maybe there's a clever-er thing to do @alejoe91 ?? I'll do a PR if my simple fix is ok!

chrishalcrow, 2025-02-17:
Also, does this mean I'm the only one using zarr'd sorting analyzers??? They're really really small, everyone!

alejoe91, 2025-02-17:
I use them all the time! Probably a problem of the select units!

The problem seems to be that the `select_units`..I'll take a look :)

alejoe91, 2025-02-17:
yeah, we should have try-except as in the binary folder :) this is totally acceptable since the analyzer makes is own copy of the sorting, so a user could easily delete one. If you're ok, I'll just update #3694 to fix it

closed 2025-02-17 by chrishalcrow