SpikeInterface/spikeinterface #3703: Question about PAIRED_KAMPFF datasets accuracy on various sorters

ZoeChen96, 2025-02-17:
Dear,
I am trying to use different sorters for the dataset PAIRED_KAMPFF. However, all 5 sorters I used give a lot more spikes than expected. The code is shown here, with the raster plot from the most accurate sorter (spykingcircus 2), which is also only 20% accuracy. I also plot the raster plots of ground_truth and timeseries. Could you please help a little bit about where to start debugging? Thank you very much!

```
import numpy as np
import pandas as pd
import os
import matplotlib.pyplot as plt
import spikeinterface.extractors as se 
from spikeinterface.preprocessing import bandpass_filter
from probeinterface import Probe
import spikeinterface.comparison as sc
##############################file loading###########################
file_folder =  "a"
file_name = "sub-paired-kampff_ses-paired-kampff-2014-11-25-Pair-3-0_ecephys.nwb"

print(file_name)
recording,sorting_GT = se.read_nwb(file_path=os.path.join( file_folder, file_name), load_recording=True,load_sorting=True)
recording = recording.frame_slice(start_frame=0,end_frame = 60*recording.get_sampling_frequency())  
sorting_GT = sorting_GT.frame_slice(start_frame=0,end_frame = 60*recording.get_sampling_frequency())  
recording = bandpass_filter(recording)

probe_folder = "a"
probe_name = "probe_sub-paired-kampff_ses-paired-kampff-2014-11-25-Pair-3-0_ecephys.csv"

probe_df = pd.read_csv(os.path.join(probe_folder,probe_name))
xycoords = np.vstack( (probe_df['x'], probe_df['y'])).T #shape = 32,2
probe = Probe(ndim=2, si_units='um')
probe.set_contacts(positions=xycoords, shapes='circle', shape_params={'radius': 5})
channel_indices = np.arange(32)
probe.set_device_channel_indices(channel_indices)
recording.set_probe(probe)
recording.set_channel_locations(probe.contact_positions)

#############################sorting#######################################
from pathlib import Path
import spikeinterface.sorters as ss
import sys
from spikeinterface.preprocessing import common_reference, zscore
base_folder = Path("./test")
original_stdout = sys.stdout # Save a reference to the original standard output
sorter_params = dict(n_jobs=-1, chunk_duration="1s", progress_bar=True)
ss.IronClustSorter.set_ironclust_path('/imec/other/macaw/chen14/spikeinterface/Official_Tutorial_SI_0.96_Oct22/ironclust')
ss.Kilosort3Sorter.set_kilosort3_path('/imec/other/macaw/chen14/spikeinterface/Official_Tutorial_SI_0.96_Oct22/Kilosort')
cmp_method = ("raw_count", "by_unit", "pooled_with_average")

def sorter_loop_nofilt(srtr,sorter,recording,sorting_GT,set_name): 
    acc_list1 = []
    acc_list2 = []
    well_detected = []
    filename = 'test.txt'
    #recording = recording.frame_slice(start_frame=0*fs, end_frame=60*fs)
    path = base_folder /  ("14449576_sorters_test/results_nofilt"+srtr+set_name )
    if srtr == 'IC': 
        sorting_sorter = ss.run_sorter(sorter, recording, output_folder=base_folder /  ("14449576_sorters_test/results_nofilt"+srtr+set_name), delete_output_folder = True, filter = False, verbose=False,**sorter_params)
    elif srtr == 'KS': 
        sorting_sorter = ss.run_sorter(sorter, recording, output_folder=base_folder /  ("14449576_sorters_test/results_nofilt"+srtr+set_name), delete_output_folder = True, freq_min = 40,verbose=True,**sorter_params)
    elif srtr == 'pyKS' :
        sorting_sorter = ss.run_sorter(sorter, recording, output_folder=base_folder /  ("14449576_sorters_test/results_nofilt"+srtr+set_name), fshigh = 40,verbose=True,**sorter_params)
    elif srtr == 'HS':
        sorting_sorter = ss.run_sorter(sorter, recording, output_folder=base_folder /  ("14449576_sorters_test/results_nofilt"+srtr+set_name),  filter = False, verbose=False)
    elif srtr == 'TDC': 
        recording = common_reference(recording)
        recording = zscore(recording, dtype='float32')
        sorting_sorter = ss.run_sorter(sorter, recording, output_folder=base_folder /  ("14449576_sorters_test/results_nofilt"+srtr+set_name), delete_output_folder = True,  apply_preprocessing = False, verbose=False)
    elif srtr == 'TDC0':
        sorting_sorter = ss.run_sorter(sorter, recording, output_folder=base_folder /  ("14449576_sorters_test/results_nofilt"+srtr+set_name ),  freq_min = 40, freq_max = None, verbose=False)

    elif srtr == 'SC': 
        recording = common_reference(recording)
        sorting_sorter = ss.run_sorter(sorter, recording, output_folder=base_folder /  ("14449576_sorters_test/results_nofilt"+srtr+set_name), delete_output_folder = True,  apply_preprocessing = False, verbose=False)
    file_path = base_folder/("sorting_results/results_nofilt"+srtr+set_name)
    se.NpzSortingExtractor.write_sorting(sorting_sorter, file_path)
    comp1 = sc.compare_sorter_to_ground_truth(sorting_GT, sorting_sorter,match_score = 0.0,exhaustive_gt=False)
    tmp_accuracy1 = comp1.get_performance(method = cmp_method[2])['accuracy']
    acc_list1.append(tmp_accuracy1)
    comp2 = sc.compare_sorter_to_ground_truth(sorting_GT, sorting_sorter,chance_score = 0.1,match_mode = 'best', exhaustive_gt=False)
    tmp_accuracy2 = comp2.get_performance(method = cmp_method[2])['accuracy']
    acc_list2.append(tmp_accuracy2)
    tmp_well_units = [len(comp2.get_well_detected_units())]
    well_detected.append(tmp_well_units)
    print('no filter, set = ',set_name,', Hungrian 0.0 accuracy = ', tmp_accuracy1, ',best match 0.1 accuracy = ', tmp_accuracy2)
    print('no filter, set = ',set_name,', best match 0.1 summary(well/bad/fp/overmerged/redundant) = ', tmp_well_units)
    cmd = "rm -rf mearec_sorters_test/results_nofilt"+srtr+set_name
    os.system(cmd)
    print('-----------'+srtr+', finish--------')

    with open(filename, "a") as f:    
        sys.stdout = f # Change the standard output to the file we created.   
        print('------------------Results for '+ sorter+'---------------')  

        print('no filter, set = ',file_name,',Hungrian 0.0 accuracy = ', acc_list1,',best match 0.1 accuracy = ', acc_list2 )

        print('no filter, set = ',file_name,', best match 0.1 summary(well/bad/fp/overmerged/redundant) = ', well_detected)
    sys.stdout = original_stdout
    breakpoint()
    return

#sorter_loop_nofilt('HS'  ,'herdingspikes' ,recording,sorting_GT,'kampff') #too many false positives ac= 8%
sorter_loop_nofilt('IC'  ,'ironclust'     ,recording,sorting_GT,'kampff') 
#sorter_loop_nofilt('SC'  ,'spykingcircus2',recording,sorting_GT,'kampff') #acc = 20%
#sorter_loop_nofilt('TDC0','tridesclous'   ,recording,sorting_GT,'kampff') #acc = 2%, too many false positives
sorter_loop_nofilt('KS'  ,'kilosort3'     ,recording,sorting_GT,'kampff') #acc=12.6%
```
The plot is here: 

[image]

ZoeChen96, 2025-02-17:
PS: I download the datasets from [here](https://dandiarchive.org/dandiset/000618/draft/files?location=sub-paired-kampff&page=1), and the probe information is [here](https://neurosift.app/?p=/nwb&url=https://api.dandiarchive.org/api/assets/dfa4d7a6-ae91-4291-afb0-3212d52522e8/download/&dandisetId=000618&dandisetVersion=draft&tab=neurodata-item:/general/extracellular_ephys/electrodes%7CDynamicTable)

yger, 2025-02-17:
Thanks a lot for your interest. I do not know exactly this GT dataset, but several considerations:
- this is in vivo with 32 channels, am I right? This means that there are drifts, and drifts with 32 channels are not amazingly handled by any methods at the moment.
- You have several brain states in these recordings, like ripples, thus again, the stationary assumption made by all these sorters are challenged. Maybe some recordings of the datasets are easier to sort than other, so I guess you should not draw conclusion only on one recording.
- The fact that you have more spikes, what do you means? The sorters will find more cells, which is normal, given that you are only comparing to the one cell  that has been patched. This is a non exhaustive GT, you have the real activity for only one neuron, and thus you need to find, in the raster plots found by the sorter, the one that matches the most to your GT spikes. Ideally, there should be one cell. But because you might have (uncorrected) drifts/brain states, maybe there are oversplits, and the GT cell has been splitted in two by the sorter. I don't know. You can check by looking at accuracies of other cells also.

Otherwise, I can talk more about SC2, the one I know the most regarding parameters

Hope this helps.

ZoeChen96, 2025-02-18:
Hi @yger ,thanks for your reply! I assume this dataset will have good results, as it is reported on [spikeforest](https://spikeforest.flatironinstitute.org/studyresults/paired_kampff), where we can see  Spiking Circus has the accuracy of 96%, and most sorters have the accuracy >90%, that's why I think it's weird. 

> this is in vivo with 32 channels, am I right? This means that there are drifts, and drifts with 32 channels are not amazingly handled by any methods at the moment.

The picture I attached (left) shows the filtered recording. Can we see drifting issue from there?

> You have several brain states in these recordings, like ripples, thus again, the stationary assumption made by all these sorters are challenged. Maybe some recordings of the datasets are easier to sort than other, so I guess you should not draw conclusion only on one recording.

Exactly, so I tried this dataset first as it seems to be a easy one.

> The fact that you have more spikes, what do you means? The sorters will find more cells, which is normal, given that you are only comparing to the one cell that has been patched. This is a non [...]

If you see the attached picture (middle and right), they are the raster plots of SC2 sorter and ground truth. I understand that I need to compare one neuron to one cluster, but all 15 sorted neurons from SC2 (middle) look having a lot more events than the ground truth (right). And this occurs the same to all other sorters (Kilosort, Tridesclous etc), that's why I assume there could be a problem in the dataset.

BTW, as for SC2, do you have any suggestions on the parameter given this result? I am now using the default parameters. I tried to change the 'detection_threshold' to be 2-10, but the accuracy does not become higher. 

```
 _default_params = {
        'general' : {'ms_before' : 2, 'ms_after' : 2, 'local_radius_um' : 100},
        'waveforms' : {'max_spikes_per_unit' : 200, 'overwrite' : True},
        'filtering' : {'dtype' : 'float32'},
        'detection' : {'peak_sign': 'neg', 'detect_threshold': 5},
        'selection' : {'n_peaks_per_channel' : 5000, 'min_n_peaks' : 20000},
        'localization' : {},
        'clustering': {},
        'matching':  {},
        'registration' : {},
        'apply_preprocessing': True,
        'shared_memory' : False,
        'job_kwargs' : {}
    }
```

yger, 2025-02-18:
Ah, then my bad, somethings is weird. If you have a simple dataset, with a proper SNR for the GT neurons (have you checked that), then you should get it. Especially if SC1 had it, SC2 should have it also. Your params seems legit to me, and changing the detection threshold will not change a lot. However, from the default params you are showing, I think you do not have the latest version (please update spikeinterface, double checking first the version you are using). Because I think it might have improved in latest versions.

ZoeChen96, 2025-03-04:
Hi @yger , thanks for your help! I upgraded the version, and tried to run the SC2 again. But I encountered the following question:

```
 UserWarning: `n_jobs` is not set so parallel processing is disabled! To speed up computations, it is recommended to set n_jobs either globally (with the `spikeinterface.set_global_job_kwargs()` function) or locally (with the `n_jobs` argument). Use `spikeinterface.set_global_job_kwargs?` for more information about job_kwargs.
  warnings.warn(
detect and localize: 100%|###############################################################################################################################################| 495/495 [00:33<00:00, 14.98it/s]
envs/spkInterface/lib/python3.12/site-packages/spikeinterface/core/baserecordingsnippets.py:269: UserWarning: There is no Probe attached to this recording. Creating a dummy one with contact positions
  warn("There is no Probe attached to this recording. Creating a dummy one with contact positions")
Error running sorter spkSC: Spike sorting error trace:
Traceback (most recent call last):
envs/spkInterface/lib/python3.12/site-packages/spikeinterface/sorters/basesorter.py", line 261, in run_from_folder
    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)
  File "anaconda43/envs/spkInterface/lib/python3.12/site-packages/spikeinterface/sorters/internal/spyking_circus2.py", line 146, in _run_from_folder
    recording_f = correct_motion(recording_f, **params["motion_correction"])
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "anaconda43/envs/spkInterface/lib/python3.12/site-packages/spikeinterface/preprocessing/motion.py", line 436, in correct_motion
    motion = estimate_motion(recording, peaks, peak_locations, progress_bar=progress_bar, **estimate_motion_kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "anaconda43/envs/spkInterface/lib/python3.12/site-packages/spikeinterface/sortingcomponents/motion/motion_estimation.py", line 111, in estimate_motion
    motion = method_class.run(
             ^^^^^^^^^^^^^^^^^
  File "anaconda43/envs/spkInterface/lib/python3.12/site-packages/spikeinterface/sortingcomponents/motion/decentralized.py", line 147, in run
    non_rigid_windows, non_rigid_window_centers = get_spatial_windows(
                                                  ^^^^^^^^^^^^^^^^^^^^
  File "anaconda43/envs/spkInterface/lib/python3.12/site-packages/spikeinterface/sortingcomponents/motion/motion_utils.py", line 312, in get_spatial_windows
    raise Exception(
Exception: get_spatial_windows(): win_step_um=200.0/win_scale_um=300.0/win_margin_um=-150.0 are too large for the probe size (depth range=99.0). You can try to reduce them or use rigid motion.
```

Could you please guide me how to do this? BTW, I  think I set the probe already, by the code below:
```
probe_df = pd.read_csv(os.path.join(probe_folder,probe_name))
xycoords = np.vstack( (probe_df['x'], probe_df['y'])).T #shape = 32,2
probe = Probe(ndim=2, si_units='um')
probe.set_contacts(positions=xycoords, shapes='circle', shape_params={'radius': 5})
channel_indices = np.arange(32)
probe.set_device_channel_indices(channel_indices)
recording.set_probe(probe)
recording.set_channel_locations(probe.contact_positions)
```

yger, 2025-03-04:
Just add apply_motion_correction=False to the run_sorter command, it should avoid motion correction (otherwise, for a 32 channel probe, you need to adjust parameters as requested)

yger, 2025-03-04:
Only in latest main branch I think job_kwargs is properly set, and it should work... Otherwise set it globally si.set_global_job_kwargs(n_jobs=-1) before launching run_sorter and it should work

ZoeChen96, 2025-03-04:
> Just add apply_motion_correction=False to the run_sorter command, it should avoid motion correction (otherwise, for a 32 channel probe, you need to adjust parameters as requested)

Hi, thanks for the quick response!
Can I also try the preset = rigid_fast? do you think it makes more sense?

ZoeChen96, 2025-03-04:
BTW， how much memory/CPUs would you recommend for a 10min-32channel dataset (floatpoint data) running in SC2?  As I found other sorters like kilosort4, tridesclous can finish the tasks in ~1-2min, while SC2 take around 20-30min. (in spikeinterface 0.97.0)

yger, 2025-03-04:
Please upgrade to latest release or to main, because it should be much faster, as fast as KS in fact or faster. I'm puzzled by your numbers... Maybe check parallelism?

ZoeChen96, 2025-03-04:
> Please upgrade to latest release or to main, because it should be much faster, as fast as KS in fact or faster. I'm puzzled by your numbers... Maybe check parallelism?

Apart from 0.97.0, I have another environment with spikeinterface Version: 0.101.1. Is it possible to stay in this environment? As I don't want to take risk to break the existing dependencies.

So I am using 0.101.1 SI to run SC2. I found out that if I check htop, indeed a lot of CPUs are idle. But by setting the n_jobs =-1 globally as you suggested, the warning is gone. Do you have any other ideas to check or set the parallelism correctly?