SpikeInterface/spikeinterface #3704: group_mode="by_shank" does not propagate to .mat probe file generated for Kilosort4

nikhilchandra, 2025-02-18:
I understand that Kilosort4 has added direct support for the processing of multi-shank data. My reading of the Kilosort4 documentation indicates that  proper processing of a multi-shank probe requires electrodes belonging to each shank to have a correctly assigned value for the "kcoords" parameter (see this page: https://kilosort.readthedocs.io/en/latest/tutorials/make_probe.html).

When I run Kilosort4 from its own GUI (no SpikeInterface involved) using a multi-shank probe map (.mat format), the output file channel_shanks.npy correctly assigns each channel to its proper shank_id. However, when I run Kilosort4 through SpikeInterface's run_sorter command, channel_shanks.npy is uniformly populated by 0's. I think this means that shankIDs are not getting correctly assigned to the kcoords parameter in the .mat probe map generated in kilosortbase.py.

For reference, here is the script I am using to run Kilosort4.

`def BIN_1024ch_SI_4S1024_try2():
    recording = si.read_binary(bin1024_file, 
                               sampling_frequency=40_000, 
                               dtype=np.int16, 
                               num_channels=1024, 
                               time_axis=0, 
                               gain_to_uV=0.18317874024609845,
                               offset_to_uV=0, 
                               is_filtered=False)
    
    # recording = si.bandpass_filter(recording)
    # recording = si.common_reference(recording)
    
    prb1024 = get_probe(manufacturer="plexon", probe_name="4S1024")
    prb1024.set_device_channel_indices(channel_indices=prb1024.contact_ids)
    recording = recording.set_probe(prb1024, group_mode="by_shank")
    sorting = si.run_sorter(
        sorter_name = "kilosort4",
        recording = recording,
        verbose = True,
        folder = output_folder / dataset_name,
        remove_existing_folder = False,
        save_preprocessed_copy = True,
    )`

Notice that when I assign the probe to the recording object, I set group_mode to "by_shank".  Further, if you look at the probeinterface probe definition [here](https://github.com/SpikeInterface/probeinterface_library/blob/main/plexon/4S1024/4S1024.json), you will see that it includes shank_ids starting on line 19,570.

Is there a way to fix this?

alejoe91, 2025-02-18:
@nikhilchandra we don't propagate kcoords and actually never have @zm711 because we have a SpikeInterface way to handle multi-shanks that's sorter agnostic. 

I agree though that we could add an option to do it.

zm711, 2025-02-18:
That makes sense from our perspective. Thanks Alessio. My only concern with giving the option is whether users will know when it matters or not and get confused thinking that by saying to write it that it is actually doing something (when I'm pretty sure for some versions the kcoords stop being used).

alejoe91, 2025-02-18:
@nikhilchandra you can use `run_sorter_by_property` to use the spikeinterface grouping

nikhilchandra, 2025-02-18:
@alejoe91 does this mean that SpikeInterface does not support Kilosort4's direct multi-shank processing?

Relevant, please check this out: [https://github.com/MouseLand/Kilosort/issues/848](https://github.com/MouseLand/Kilosort/issues/848)

zm711, 2025-02-18:
Our wrapper should allow you to do it if you want. Our argument is always that by doing the per shank in spikeinterface you're better isolating the shanks. This increase accuracy and isolation but requires a pretty big time commitment which we try to help with through our job engines. That being said the end-user can empirically test and if KS4's multi-shank processing works for them or not ( I believe settings like `dminx` help tune the shank isolation). The idea that KS4 is doing (I think) is: you treat the data as one big mass of data and only exclude templates if the distances don't make sense (so if two shanks are close together they could technically have shared templates). Then the `kcoords` is just to keep track of the shank information but not for isolation purposes. So it is not true shank isolation but rather a distance based isolation. Our technique is to have the user make a decision about whether or not they want to isolate the shanks. 

I know you're pretty good about optimizing so I would recommend try a smaller dataset with their multishank at once vs our shank by shank and see if there is a big enough accuracy difference (if there is one) to warrant uses our`run_sorter_by_property` or whether you want to just do `run_sorter` instead.

nikhilchandra, 2025-02-18:
Okay, thanks @zm711 It'll take me some time to parse all of this and figure out how to proceed.

nikhilchandra, 2025-03-24:
I've been diving deeper into Kilosort4's code and came across this function. I might be off base, but wouldn't the highlighted line get affected by incorrectly assigned shank_ids?

```
def template_centers(ops):
    shank_idx = ops['kcoords']
    xc = ops['xc']
    yc = ops['yc']
    dmin = ops['settings']['dmin']
    if dmin is None:
        # Try to determine a good value automatically based on contact positions.
        y_uniq = np.unique(yc)
        if y_uniq.size == 1:
            dmin = 1
        else:
            dmin = np.median(np.diff(np.unique(y_uniq)))
    ops['dmin'] = dmin
    ops['dminx'] = dminx = ops['settings']['dminx']

    # Iteratively determine template placement for each shank separately.
    yup = np.array([])
    xup = np.array([])
    **for i in np.unique(shank_idx):
        xc_i = xc[shank_idx == i]
        yc_i = yc[shank_idx == i]
        xmin, xmax, ymin, ymax = xc_i.min(), xc_i.max(), yc_i.min(), yc_i.max()

        yup = np.concatenate([yup, np.arange(ymin, ymax+.00001, dmin/2)])
        nx = np.round((xmax - xmin) / (dminx/2)) + 1
        xup = np.concatenate([xup, np.linspace(xmin, xmax, int(nx))])**

    ops['yup'] = np.unique(yup)
    ops['xup'] = np.unique(xup)

    return ops
```

nikhilchandra, 2025-03-25:
Understood, my apologies. I'm still learning :)

Is this the kind of link you mean?

https://github.com/MouseLand/Kilosort/blob/47014f335f3df6bd647307a79ab4736f330befb1/kilosort/spikedetect.py

(go to line 96)

zm711, 2025-03-25:
No worries. So actually if you highlight the lines three dots will appear and give you the option to copy a permalink to those lines.

like this https://github.com/MouseLand/Kilosort/blob/47014f335f3df6bd647307a79ab4736f330befb1/kilosort/spikedetect.py#L96-L126

nikhilchandra, 2025-03-31:
Hi, any update on this? @zm711 @alejoe91

zm711, 2025-03-31:
We are meeting Thursday. I'll bring it back up then.

zm711, 2025-04-03:
@nikhilchandra, we all agreed that it would be appropriate to change how we write the probe file. We will try to use kilosort's own probe saving function of our probe representation. So we should get to this soon.

alejoe91, 2025-04-08:
@nikhilchandra 

So right now we are writing a prb file [here](https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/sorters/external/kilosort4.py#L141-L145), but we shold instead use the KS4 io functions: https://kilosort.readthedocs.io/en/latest/tutorials/make_probe.html

Do you have time to work on it and make a PR?

PR #3852 (2025-04-17): Updated KilosortBase's channel map generation to support json probe maps with kcoord information.
Fixes #3704
closed 2025-04-17 by alejoe91