SpikeInterface/spikeinterface #3708: zarrextractor

yger, 2025-02-19:
Hi all

I have a bug with my benchmarks, triggered by I guess new zarrextractors

`
[~/github/spikeinterface/src/spikeinterface/core/zarrextractors.py:83](http://localhost:8888/home/pierre/github/spikeinterface/src/spikeinterface/core/zarrextractors.py#line=82), in super_zarr_open(folder_path, mode, storage_options)
     81         except Exception as e:
     82             pass
---> 83 if root is None:
     84     raise ValueError(f"Cannot open {folder_path} in mode {mode} with storage_options {storage_options}")
     85 return root
`

Indeed, looks like L83 can be reach in core/zarrextractor.py without having the root variable defined. Is it normal?

yger, 2025-02-19:
My bad, looks like if I am regenerating everything this is working. However, maybe worth checking L83 and the fact that root can be non defined? Not sure of why i had that

closed 2025-02-19 by yger

alejoe91, 2025-02-19:
Already fixed here :)
https://github.com/SpikeInterface/spikeinterface/pull/3686