SpikeInterface/spikeinterface #3720: [Question]: Usage of Graph Algorithms and Any Slowdowns?

nv-rliu, 2025-02-24:
Hi there,

I'm interested in understanding if `spikeinterface` depends on any graph algorithms from its usage of NetworkX? If so,

- What algorithms are used for what purpose?
- What graph sizes are they being used with?
- Have users experienced any slowdowns or issues with algorithms provided by NetworkX? (Speed, algorithm availability, etc)

Furthermore, would users be interested in accelerated nx algorithms via a GPU backend? This would involve zero code change.

Any insight into this topic would be greatly appreciated! Thank you.

zm711, 2025-02-24:
I think the main place using NetworkX is in [comparison](https://github.com/SpikeInterface/spikeinterface/tree/main/src/spikeinterface/comparison), but correct me if I'm wrong @alejoe91 . I think there was discussion of refactoring that a bit, but I think it has been pretty low-down on the too list at the moment. I haven't really worked on that submodule, but the main idea there is to make a graph of the units that different sorters find to try to find the agreement units that all sorters/user specified number sorters are able to find. 

I honestly don't know how many users we have that use this submodule currently. As far as speed ups I haven't checked our test suite to see if our comparison tests are super slow but if they are I would be interested improving those, but I think the final request will be filtered through @alejoe91 and @samuelgarcia.

yger, 2025-02-24:
Actually, there could be an interest also with graph based clustering methods such as Louvain with millions of points (the spikes). This is what we are currently trying to do with @samuelgarcia.

zm711, 2025-03-10:
#3714 is a draft of adding some graph stuff if you're willing to read the source code other @samuelgarcia or @yger can also explain more here. :)