SpikeInterface/spikeinterface #3727: Wrong gain inference for Plexon2 files

FrancescoNegri, 2025-03-03:
When the `NeoBaseRecordingExtractor` class set the `gain_to_uV` property, it starts from the gains retrieved by `Neo` and applies an `additional_gain` according to the signal unit such that the signal is automatically scaled to microvolts (uV).
This is done in a hardcoded way by checking the `units` array values against a predefined set of alternatives`["V", "Volt", "mV", "uV"]`.

Unluckily, Plexon2 files encode the unit as `"Volts"` and this is not recognized by `NeoBaseRecordingExtractor`, hence a standard `additional_gain` of `1` is applied, leading to a wrong `gain_to_uV` value.

The situation is illustrated in the attached image, that is the `NeoBaseRecordingExtractor` class during debug.

A quick fix would be to add a line checking for `"Volts"`:
`additional_gain[units == "Volts"] = 1e6`.

However, this workaround might still fail in future, whenever a unit is not recognized.
Additionally, the `has_non_standard_units` property of `NeoBaseRecordingExtractor` is correctly set to `True`, but no warning or error is raised.

[image]

alejoe91, 2025-03-03:
Thanks @FrancescoNegri 

Would you minf making a PR? Also adding a warning for non standard units sounds fair!

zm711, 2025-03-03:
Is this worth also mentioning on Neo since Plexon has been working with us over there to ensure their DLL plays well with Neo. We could ask for clarity from them on their policies for units moving forward or whether they would refuse to offer any such guarantees.

h-mayorquin, 2025-03-04:
OK, this should be fixed by #3728. Thanks a lot @FrancescoNegri!

closed 2025-03-04 by h-mayorquin