SpikeInterface/spikeinterface #3729: Importing spikeinterface.full takes too long

aka2333, 2025-03-03:
I'm trying to import spikeinterface (v0.102.1, installed via conda) in the vscode interactive window (base: python 3.10.14) using `import spikeinterface.full as si` , but it takes super long to run (literally hours) and still not finishing. I tried to import submodules separately, and the same issue also happened to `import spikeinterface.sorters as ss` and `import spikeinterface.curation`, but not to the other submodules. I'm wondering what can be done to troubleshoot.

aka2333, 2025-03-04:
> That is wild. We actually run tests on each PR to ensure that on all OSes import time is less than 1.5 minutes for full and submodules. could you try doing it from a terminal and see if it still takes over two minutes on a terminal. Also what OS are you using?

It didn't work on a terminal either. I also tried re-installed to a new virtual env (python 3.11 as recommended in the installation_tip) via the full_spikeinterface_environment_windows.yml but still having the same problem. The OS is win10.

However I tried this morning to install on another Win11 device following the same steps and ran `import spikeinterface.full as si` with vscode interactive window and it worked very smoothly (just a few secs to finish running). So it seems to be some problem specific to my previous device...

zm711, 2025-03-04:
That's definitely going to be tough if this ends up being a station specific issue.... Could you try

```
conda create test_si_pip python=3.11
pip install spikeinterface
```

and then try loading in a terminal. I'm wondering if there is some package confusion that happened with our full install and so let's try a super skinny install so we can test that first!

aka2333, 2025-03-10:
> That's definitely going to be tough if this ends up being a station specific issue.... Could you try
> 
> ```
> conda create test_si_pip python=3.11
> conda activate test_si_pip
> pip install spikeinterface
> ```
> 
> and then try loading in a terminal. I'm wondering if there is some package confusion that happened with our full install and so let's try a super skinny install so we can test that first!

Thanks for following up and sorry for the late reply since I was on leave last week.
I tried with this method on the initial device and yes now it can be loaded properly in the terminal.

[image]

aka2333, 2025-03-13:
> Okay, so maybe there is some specific package that is messing your install up. Do you want to try to use this minimal install in vscode and see if that is sped up now. That should help us diagnose if this is a vscode issue on your station or whether it was one of the packages are install did.

Yep it also worked in vscode.

[image]

zm711, 2025-03-13:
Okay cool that's great. So my recommendation would try this next

```
conda create si_env python=3.11
conda activate si_env
pip install spikeinterface[full,widgets]
```

This will test a full install with pip that allows for plotting too. If this install works and is not slow then you're set for all your needs with spikeinterface itself. If that has problems then we would need to figure out what package is slowing the install.

aka2333, 2025-03-14:
> pip install spikeinterface[full,widgets]

Hummm, I tried with the new codes and it gets stuck again with both interactive window and terminal :( 

[image]

[image]

zm711, 2025-03-14:
Okay, sorry this is taking so long! We just have to test one thing at a time. Now you can break this up and try both

```
conda create si_env_full python=3.11
conda activate si_env_full 
pip install spikeinterface[full]
```

and then try 
```
conda create si_env_widgets python=3.11
conda activate si_env_widgets
pip install spikeinterface[widgets]
```

Once we know which set of packages are the problem for you then we can try to ensure the correct packages are installed without downloading the slow package.

aka2333, 2025-03-14:
> pip install spikeinterface[widgets]

No worries. I actually should thank you for helping troubleshoot 😇

I accidentally figured out that the importing issue was relevant to the admin access. Previously I was not running the code as admin, and it's seems working with only the widget version but got stuck with the full one.
[image]

But when I run vscode or cmd as admin, both version worked properly.

[image]

I also tried on the inital environment where I first encountered the problem, and it also worked with admin access.

zm711, 2025-03-14:
No worries. Happy to help. And this is super valuable info that I can share with people in the future. I'll change your title and close to make it easier for people to figure out :) 

Feel free to open more issues if they come up!