SpikeInterface/spikeinterface #3731: Identifying Which Units come from Which Tetrode

JacobKelley101, 2025-03-03:
Hello,

My lab has been developing some custom visualization and QC pipelines, and it has become necessary for us to compare units that wwere detected on the same tetrode against each other, but not against units detected on other tetrodes. I am having difficulty finding where this data is stored, but I figured it is such a simple thing that it must be somewhere. Can you help me figure out the best way to access this data?

Thank you!

Jacob

zm711, 2025-03-03:
Hey @JacobKelley101,

That information would be stored in the `unit_locations` extension of the `sorting_analyzer`. How did you sort your tetrodes? Did you do it as one giant probe with the tetrodes spaced in the x direction or did you sort each tetrode separately? That will slightly change out to interpret this info so it might be helpful for you to post your sorting script so we know how you're processing the tetrodes so we know where that specific info is.

chrishalcrow, 2025-03-04:
Another idea: if you've computed the templates, you can use 
```
import spikeinterface as si
template_extremum_channel = get_template_extremum_channel(sorting_analyzer)
```
to get the channel which each unit waveform has a maximum amplitude on.

alejoe91, 2025-03-04:
@JacobKelley101 did you run spike sorting all together or split by group (with the `run_sorter_by_property`)?

In the latter case, you can access the tetrode group with: `sorting.get_property("group")`

zm711, 2025-03-07:
> but the sorting took ~2x as long because MS5 would run twice. On a separate note, would you expect that to be the case (run time ~doubling for two tetrodes, ~tripling for three, etc), or is does that seem extreme to you

That makes sense to me. If one tetrode takes 1 minute then rerunning the exact same algorithm in sequential order should take ~2 minutes. You could try one of our other engines to try to parallelize, to see if you can reduce the time down to less than 2x/3x. They might actually work. Our docs for this unfortunately need some work but see
https://github.com/SpikeInterface/spikeinterface/blob/aa3d561c930569f1bc777a4a55abea04f8d62619/src/spikeinterface/sorters/launcher.py#L244-L248
and here:
https://github.com/SpikeInterface/spikeinterface/blob/aa3d561c930569f1bc777a4a55abea04f8d62619/src/spikeinterface/sorters/launcher.py#L40-L61

As you can see the default is to loop through your jobs which means each job should take 1x time. You could try joblib (or dask/slurm if you have the infrastructure) to see if they help with speed.

So your solution is to make 1 giant probe with artificial space between. and only running the algorithm once which may lead to a very small slowdown or no slowdown This used to be Kilosort's (unofficial) recommendation for dealing with multishank experiments. There are two disadvantages 1) you might mess up the distance exclusion and get impossible cross-talk between tetrodes 2) our location information is no longer the "true" location because you have added artificial distances. I think both of these issues can be ameliorated (the first with using appropriate distance exclusion and the second with you doing post-hoc corrections on the data). 

That being said our goal with the `by_property` is that it guarantees tetrode isolation at the cost of speed. Everyone has to make their tradeoffs and should test our different engines to see if they can get to the ideal mix of isolation + reduced slowdown. But I agree in the case of 60 tetrodes the 60x time to perfectly isolate the tetrodes may not be worth it. I would really recommend trying joblib (and then slurm if you an HPC) to see if you get any benefit from that.

 > If the template extremum falls in the range of 0-3 it belongs to the first tetrode, if it is 4-7 then it is in the second.

yeah you could definitely do something like that as long as you number the tetrodes in that way.

> but I am sure there is something I am missing

Not all sorters necessarily have distance exclusion criteria. In that case doing multishank probe without isolation could lead to cross shank units that shouldn't be biologically possible. Ages ago when I was doing something similar with KS2 I would routinely see cross shank units.

I think I've hit most of your points but let us know if anything is confusing.

tl;dr
sequential run_sorter_by_property would be expected to take 2x for two probes because it is sequential
play with other engines to combine isolation with multiprocessing to see if you can get a speedup (probably not quite down to 1x of running multishank)
extremum_channel should give quick readout of which tetrode if you continue with the "multishank" single probe strategy

JacobKelley101, 2025-03-07:
@zm711 Thank you so much for the response. It is comforting to know that my previous struggles with running the sorter by property were not down to my own technical ineptitude. I think I have a good idea of what strategies are available to me know, so I will close the comment. As always, it is such a joy to have developers that respond as quickly and thoroughly as you all do. Enjoy your weekend!

closed 2025-03-07 by JacobKelley101

zm711, 2025-03-07:
Thanks for your kind words. One other thing I should've mentioned but got lost in diving deep on `run_sorter_by_property` is that the spikeinterface framework lets you do preprocessing in isolation too. So if your tetrodes are far enough away the artifacts and noise they experience might actually be different so preprocessing everything together may not always be the right option (sometimes it could work fine so this isn't a hard rule). So our machinery lets you preprocess in isolation and then sort in isolation. We think that in general this should improve results, but always with an accuracy-time tradeoff (that will likely be different for each user). We are currently (through Chris's efforts) working on making the preprocessing `by_property` more user friendly. But that would be another reason why you might sacrifice speed -- because you build all of your preprocessing and sorting based on this isolated tetrode analysis.

JacobKelley101, 2025-03-07:
That makes a lot of sense. I can understand why holistically processing every tetrode in isolation would logically improve accuracy/reliability, while also just being a good book-keeping practice. My main issue with doing things this way is that, with the data our lab collects, we often have to try a few different detection thresholds before things sort well. As a result speed is much more important to me at the moment. My goal is migrating our labs processing to spikeInterface is to speed this process up, add flexibility to the data we can input, and improve reliability. As I refine the process (and the choices for detection thresholds becomes clearer), I think I will eventually sort and preprocess everything in isolation based on user-defined channel groups. I understand your vision, I think.