SpikeInterface/spikeinterface #3734: Compare multiple sorters

paul-aparicio, 2025-03-04:
Hi, I have been trying to see if I could speed up the compare_multiple_sorters function (it has been taking a very long time to complete). I have tried changing the n_jobs parameter, but it seems to not have any effect? In every case I see only 1 CPU as active. I am running si version 0.102.0, and using the results from 4 sorters. Thanks for any help or suggestions!

paul-aparicio, 2025-03-05:
I am testing this, everything seems to be working fine without error, but it hasn't changed the time for me much (it's still running 12+ hours so far). My computer only shows 1 CPU running.

alejoe91, 2025-03-06:
Mmm maybe the n_jobs parameter isn't working as expected. I'll reopen the issue and check the implementation

zm711, 2025-03-07:
What OS are you using? And are you doing `thread` or `process`?

paul-aparicio, 2025-03-25:
Hello. I have tried multiple times to install the pr 3737. I seem to get similar results when using agreement_method='count' or agreement_method='distance'. The first three steps are very fast (less than few minutes), and the 'count' method does seem to use multiple CPUs. Then 'step 4: extract agreement form graph' typically takes 12+ hours and only seems to use 1 CPU (well at least at the times I looked in on it). Here is a screen shot of the code I used. Is this the correct behavior, or maybe I haven't installed the pr correctly? I am using Ubuntu 20.04 and I am getting the number of CPUs from the system monitor app. Thank you!

[image]

closed 2025-08-22 by paul-aparicio

paul-aparicio, 2025-08-27:
If it's helpful, I tried this again with version 0.103.0 and multicomparison step 1 was clearly using multiple processors and worked quickly. Steps 2 & 3 seemed to complete quickly as well. Steps 4 took > 10 hours though, and seemed to only use 1 CPU (mainly, as far as I could tell).

PR #3737 (2025-03-05): Propagate `agreemennt_method` to `compare_multiple_sorters`
Fixes #3734

@paul-aparicio 

@h-mayorquin developed a faster method for sorting comparison here https://github.com/SpikeInterface/spikeinterface/pull/2192, but that was not propagated to the `compare_multple_sorters`

With this PR, you can do `compare_multiple_sorters(..., agreement_method="distance")` and the speed should be much higher. Can you give it a try?