SpikeInterface/spikeinterface #3735: OpenEphys neo extractor - get_streams call not complete

remi-pr, 2025-03-04:
Hi everyone,

Just tumbled upon a little bug when trying to open one stream from a openephys recording using the corresponding extractor.
Here is a function call that might better explain the issue and that you may use the reproduce the error:

`d = se.OpenEphysBinaryRecordingExtractor(filepath, load_sync_channel=False,
                                         stream_name='Record Node 101#NI-DAQmx-103.PXIe-6341',
                                         experiment_names='experiment2')
`

This fails because there is no sync channel (since I only load a NI stream) and advise me to put `load_sync_channel` to `False`. Which I did. Diving a bit deeper to understand to root of the issue I found that the extractor in SI calls the [get_streams](https://github.com/SpikeInterface/spikeinterface/blob/e1535daac114ad293cb0acdb555f1a96d797f038/src/spikeinterface/extractors/neoextractors/openephys.py#L167) method but with two positional arguments (`folder_path` and `experiment_names`) not with  `load_sync`. This in turns lead to a call to `neo_kwargs = cls.map_to_neo_kwargs(*args, **kwargs)` to produce wrong `neo_kwargs` (`load_sync` ends up being `experiment_names`). 

Anyway, changing the line 167 to `stream_names, stream_ids = self.get_streams(folder_path, load_sync_channel, experiment_names)` is enough to apparently fix the issue. But because this is a rather complicated piece of code I would rather have someone more knowledgeable than me (@samuelgarcia ?) have a look before I even dare to offer a PR.

Let me know what you want me to clarify (I realize that this reads quite confused, sorry...)

alejoe91, 2025-03-04:
@remi-pr thanks for spotting this! Your proposed solution sounds very reasonable! Please dare making a PR :)

h-mayorquin, 2025-03-04:
Should we separate the sync channels as their own stream? @chrishalcrow

zm711, 2025-03-07:
With the fix on Neo can this be closed or do you want to discuss the sync more @h-mayorquin ?