SpikeInterface/spikeinterface #3741: Question about herdingspikes (on raw data)

ZoeChen96, 2025-03-05:
Hi, I am trying to test different filter settings for herdingspikes. I tried different cut-off frequencies, which works well, but if I try to use the unfiltered data, it raises an error. Could you please have a look at the error message? Does it mean that Herdingspikes does not work with unfiltered data (with LFP), or can I change some settings to make it work (although the accuracy could be low)?

```
{
    "sorter_name": "herdingspikes",
    "sorter_version": "0.3.7",
    "datetime": "2025-03-03T21:50:05.440519",
    "runtime_trace": [],
    "error": true,
    "error_trace": "Traceback (most recent call last):\n  File \"/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/spikeinterface/sorters/basesorter.py\", line 226, in run_from_folder\n    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)\n  File \"/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/spikeinterface/sorters/herdingspikes/herdingspikes.py\", line 226, in _run_from_folder\n    uids = C.spikes.cl.unique()\n  File \"/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/pandas/core/generic.py\", line 5902, in __getattr__\n    return object.__getattribute__(self, name)\nAttributeError: 'DataFrame' object has no attribute 'cl'\n",
    "run_time": null
}
```

zm711, 2025-03-10:
@mhhennig did you want to respond here?

zm711, 2025-04-04:
Maybe @chrishalcrow did you want to comment? Or grab someone from the group to comment?

chrishalcrow, 2025-08-22:
Hello, sorry sorry.

I only know the newer versions of herdingspikes - it changed a bit at version v0.4.3 (https://github.com/mhhennig/HS2) I'd recommend installing the newest version and having another go - the algorithm should work ok on unfiltered data.