SpikeInterface/spikeinterface #3755: Kilosort 4 log

atombysx, 2025-03-09:
Since kilosort 4 is still developing and they recommend sending them the kilosort4.log file to troubleshoot, is it possible to produce it with spikeinterface run_sorter()?

Spikeinterface_log.json is useful but it contains much less information, e.g.:

{
    "sorter_name": "kilosort4",
    "sorter_version": "4.0.27",
    "datetime": "2025-03-08T17:03:09.698068",
    "runtime_trace": [],
    "error": true,
    "error_trace": [
        "Traceback (most recent call last):",
        "  File \"/home/saleem_lab/si_lab/spikeinterface/src/spikeinterface/sorters/basesorter.py\", line 262, in run_from_folder",
        "    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)",
        "  File \"/home/saleem_lab/si_lab/spikeinterface/src/spikeinterface/sorters/external/kilosort4.py\", line 366, in _run_from_folder",
        "    st, tF, _, _ = detect_spikes(",
        "                   ^^^^^^^^^^^^^^",
        "  File \"/home/saleem_lab/anaconda3/envs/si_env/lib/python3.11/site-packages/kilosort/run_kilosort.py\", line 673, in detect_spikes",
        "    clu, Wall = clustering_qr.run(",
        "                ^^^^^^^^^^^^^^^^^^",
        "  File \"/home/saleem_lab/anaconda3/envs/si_env/lib/python3.11/site-packages/kilosort/clustering_qr.py\", line 493, in run",
        "    Xd, igood, ichan = get_data_cpu(",
        "    ^^^^^^^^^^^^^^^^",
        "  ValueError: too many values to unpack (expected 3)"
    ],
    "run_time": null
}

zm711, 2025-03-09:
We should run everything when we call the sorter. How is the log generated? Is it an argument you enter when running? That would be easy to add.

atombysx, 2025-03-10:
From Kilosort4 API, it seems that run_kilosort() has a new parameter:
verbose_log (bool; default=False.) – If True, include additional debug-level logging statements for some steps. This provides more detail for debugging, but may impact performance.

But I see spikeinterface directly uses APIs detect_spikes() and cluster_spikes() which have:
verbose (bool; False.) – If True, include additional debug-level logging statements.

zm711, 2025-03-10:
That would be easy to add. @alejoe91 do we want to just add a verbose here or treat this as a KS-kwarg?

alejoe91, 2025-03-10:
Yes we should!

zm711, 2025-03-18:
This should have been fixed by #3768. Let us know if this didn't help @atombysx. (You would need to install from main to use this :))

closed 2025-03-18 by zm711

atombysx, 2025-03-22:
Thanks a lot! I've tested it with inputing verbose = True.

But the new kilosort version also fixed my bug so I can't verify if it will generate a log for me as all the new kilosort 4 processes I've run were successful...

Will update if it happens.