SpikeInterface/spikeinterface #3763: Kilosort3 error:

ZoeChen96, 2025-03-11:
Hi,
I am trying to run kilosort3. The GPU is configured as:
```
(si96_228) bash-4.4$ nvidia-smi
Tue Mar 11 21:54:52 2025       
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 550.127.05             Driver Version: 550.127.05     CUDA Version: 12.4     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA L40S                    On  |   00000000:21:00.0 Off |                    0 |
| N/A   31C    P8             32W /  350W |       1MiB /  46068MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
|   1  NVIDIA L40S                    On  |   00000000:E1:00.0 Off |                    0 |
| N/A   32C    P8             33W /  350W |       1MiB /  46068MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|  No running processes found                                                             |
+-----------------------------------------------------------------------------------------+
(si96_228) bash-4.4$ nvcc --version
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2019 NVIDIA Corporation
Built on Wed_Apr_24_19:10:27_PDT_2019
Cuda compilation tools, release 10.1, V10.1.168
```

And the error message from Kilosort3 is:

```
                            < M A T L A B (R) >

                  Copyright 1984-2019 The MathWorks, Inc.

              R2019a Update 1 (9.6.0.1099231) 64-bit (glnxa64)

                               April 12, 2019

 

To get started, type doc.

For product information, visit www.mathworks.com.

 

Time   0s. Computing whitening matrix.. 

matlab

signal_toolbox

Getting channel whitening matrix... 

Warning: The CUDA driver must recompile the GPU libraries because your device

is more recent than the libraries. Recompiling can take several minutes. 

> In get_whitening_matrix (line 26)

  In preprocessDataSub (line 62)

  In kilosort3_master (line 20) 

Channel-whitening matrix computed. 

Time 382s. Loading raw data and applying filters... 

----------------------------------------Error using *

Call to sgemm in CUBLAS failed with error status: CUBLAS_STATUS_EXECUTION_FAILED.
Error running kilosort3
Traceback (most recent call last):
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/spikeinterface/sorters/basesorter.py", line 226, in run_from_folder
    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)
  File "/imec/other/macaw/chen14/anaconda43/envs/si96_228/lib/python3.9/site-packages/spikeinterface/sorters/kilosortbase.py", line 178, in _run_from_folder
    raise Exception(f'{cls.sorter_name} returned a non-zero exit code')
Exception: kilosort3 returned a non-zero exit code
```

Could you tell me the reason about it?

alejoe91, 2025-03-12:
Hi @ZoeChen96 

Unfortunately these GPU errors are hard to decipher. I would open an issue on the Kilosort page or try to run it in docker.

zm711, 2025-03-12:
The people still using KS in my lab are finding it nearly impossible to get the older versions of KS to compile the GPU components (combo of Matlab version, GPU architecture changes, getting a compiler that works well with CUDA on our stations). So I agree that Docker is your best bet rather than trying to get it working locally at this point. Sorry! But I feel your pain!

zm711, 2025-03-12:
@alejoe91 is the expert on MEArec so he can comment on how much variability there could be with generating the data. I haven't actually used MEArec myself since I work only on recordings that I've obtained myself. 

As far as KS. There was a huge discussion on one of their threads with concerns for non-deterministic results. they tried to fix it in later versions, but it is possible that there is still some stochasticity in the algorithm. I think especially with giant recordings and lots of units it is very hard to always get the exact same results. So I don't think this is unexpected. I just don't have an estimate for the degree of variability ( again maybe Alessio knows this better or one of the team members who works on creating sorters).

ZoeChen96, 2025-03-12:
> [@alejoe91](https://github.com/alejoe91) is the expert on MEArec so he can comment on how much variability there could be with generating the data. I haven't actually used MEArec myself since I work [...]

@zm711  Hi, Zach! Thanks a lot for the insight. Could you please explain why the kilosort result is not deterministic?

alejoe91, 2025-03-12:
@ZoeChen96 we don't know, but it's just not! Again, you should ask the KS developers, but KS3 has been discontinued, so it is what it is :)

This said, fluctuations (especially in the noisy units) is expected, regardles of MEArec

zm711, 2025-03-13:
I think we've solved this to the limit that we can. Feel free to open more issues or if you need help getting docker to work with spikeinterface!

closed 2025-03-13 by zm711

ZoeChen96, 2025-03-13:
Thank you very much, Zach and Alessio! With the help of a colleague, I solved the problem. It's because of the compatibility : `https://ww2.mathworks.cn/help/parallel-computing/run-mex-functions-containing-cuda-code.html#mw_20acaa78-994d-4695-ab4b-bca1cfc3dbac `, it was not compatible between CUDA and MATLAB version before. Thank you again!