SpikeInterface/spikeinterface #3764: Make all imports absolute

alejoe91, 2025-03-12:
As we discussed, we should make all imports absolute throughout the source code

chrishalcrow, 2025-03-12:
Hello, turns out this is super easy with ruff. `PEP8` and the default `ruff` and `flake-8` rules allow for simple (same folder) relative imports, so that

``` python
# This is considered ok
from .benchmark_base import Benchmark, BenchmarkStudy
# rather than
from spikeinterface.benchmark.benchmark_base import Benchmark, BenchmarkStudy

# This is considered confusing
from ..preprocessing.astype import astype
# and would be changed to
from spikeinterface.preprocessing.astype import astype
```

Note: we have 1279 same-level relative import and only 194 deeper-than-one-level.

Should we keep the "simple" same-level relative imports? Opinions? @samuelgarcia @h-mayorquin @zm711 @JoeZiminski @yger ?

alejoe91, 2025-03-12:
Mmm, not sure here. I would vote for all absolute, but I agree that same level imports are less confusing!

zm711, 2025-03-12:
I agree with Alessio, but my only thought is if we can convince Sam to allow us to use ruff (see my PR #3756 for how easy linting is to add and how quick it is), then I would use whatever default is convenient so that it is all done in the pre-commit/commit action. In that case I would sacrifice the full consistency in order to have the automaticity. :)

chrishalcrow, 2025-03-12:
The Ruff pre-commit can handle both options with ease; the default is to allow for same-level imports.

zm711, 2025-03-12:
Okay that sounds like the best then to help new developers find their way around the library so I'm good with that. Thanks for playing around with it. I'm good with going all in on absolute then!

h-mayorquin, 2025-03-12:
I am fine with same-level relative imports. I am +1 on whatever gets this through, both options are better than what we have for me.

chrishalcrow, 2025-03-13:
Done in #3766

closed 2025-03-13 by chrishalcrow