SpikeInterface/spikeinterface #3775: Error reading maxwell data (version 20160704)

michaelhadler, 2025-03-14:
Hi everyone,

I want to use SpikeInterface on my MaxOne data (recorded using the "Network" recording assay in MaxLab, version 20160704). For starters, I want to create a bin and probe file to give it a go in Kilosort 4. However, I can't get the extractor to work. (For convenience with my other analysis pipelines, I'm calling python from MATLAB.)

% Import required Python modules
pathlib = py.importlib.import_module("pathlib");
np = py.importlib.import_module("numpy");
spikeinterface = py.importlib.import_module("spikeinterface.extractors");

% Load the NWB data using SpikeInterface
recording = spikeinterface.read_maxwell(filepath);     %%filepath refers to the .h5 file of the original MaxOne recording

The last command gives the following error:

    Error using [maxwellrawio>_parse_header](matlab:matlab.internal.language.introspective.errorDocCallback('maxwellrawio>_parse_header'))
    Python Error: ValueError: could not assign tuple of length 2 to structure with 3 fields.

        Error in [baserawio>parse_header] [line 211]
        
        Error in [neobaseextractor>get_neo_io_reader] [line 66]
        
        Error in [neobaseextractor>__init__] [line 27]
        
        Error in [neobaseextractor>__init__] [line 188]
        
        Error in [maxwell>__init__] [line 62]

My impression is that there is a discrepancy between the actual .h5 file structure and the expectations in _init_. The file is unchanged from the initial recording, so I'm not quite sure what's amiss. Any ideas?

Thanks, and best,

Michael

zm711, 2025-03-14:
Howdy @michaelhadler!

Happy to help. Could you open the issue on https://github.com/NeuralEnsemble/python-neo/issues. That is the project that we use for reading the files. There is some overlap between the projects, but just to keep problems in the correct place. it would also be helpful if you could share a test file with us so we can see what the problem is locally. That can be done publicly on the Neo repo or you can contact us privately if the data can't be shared.

alejoe91, 2025-03-15:
Could you also run it in Python directly and paste the error traceback? Thanks!

michaelhadler, 2025-03-17:
Dear Alessio,

the traceback in py is as such:

"Error loading data: could not assign tuple of length 2 to structure with 3 fields.

(kilosort) D:\HD-MEA\250116_Mouse-Ca12\Slice_6\Recording\002_Baseline2>python run_kilosort.py
Error loading data: could not assign tuple of length 2 to structure with 3 fields.
Traceback (most recent call last):
  File "D:\HD-MEA\250116_Mouse-Ca12\Slice_6\Recording\002_Baseline2\run_kilosort.py", line 24, in <module>
    recording = read_maxwell(file_path)
  File "C:\Users\analyzer\anaconda3\envs\kilosort\lib\site-packages\spikeinterface\extractors\neoextractors\maxwell.py", line 62, in __init__
    NeoBaseRecordingExtractor.__init__(
  File "C:\Users\analyzer\anaconda3\envs\kilosort\lib\site-packages\spikeinterface\extractors\neoextractors\neobaseextractor.py", line 188, in __init__
    _NeoBaseExtractor.__init__(self, block_index, **neo_kwargs)
  File "C:\Users\analyzer\anaconda3\envs\kilosort\lib\site-packages\spikeinterface\extractors\neoextractors\neobaseextractor.py", line 27, in __init__
    self.neo_reader = self.get_neo_io_reader(self.NeoRawIOClass, **neo_kwargs)
  File "C:\Users\analyzer\anaconda3\envs\kilosort\lib\site-packages\spikeinterface\extractors\neoextractors\neobaseextractor.py", line 66, in get_neo_io_reader
    neo_reader.parse_header()
  File "C:\Users\analyzer\anaconda3\envs\kilosort\lib\site-packages\neo\rawio\baserawio.py", line 211, in parse_header
    self._parse_header()
  File "C:\Users\analyzer\anaconda3\envs\kilosort\lib\site-packages\neo\rawio\maxwellrawio.py", line 113, in _parse_header
    signal_streams = np.array(signal_streams, dtype=_signal_stream_dtype)
ValueError: could not assign tuple of length 2 to structure with 3 fields."

Thanks a bunch for the quick reply!

michaelhadler, 2025-03-17:
I will now re-open the issue in Neo as requested by Zach :)