SpikeInterface/spikeinterface #3782: SpikeInterface wrapper for Kilosort4 does not invoke default Kilosort4 logging function

nikhilchandra, 2025-03-18:
The Kilosort4 API's entry point for the algorithm is a function called "run_kilosort()" located in kilosort/run_kilosort.py. 

The SpikeInterface wrapper for Kilosort4--located at src/spikeinterface/sorters/external/kilosort4.py--does not call "run_kilosort()" directly. Instead, it reproduces the individual function calls contained within run_kilosort(). 

One of the functions calls in the original run_kilosort() is setup_logger(), which is responsible for the generation of "kilosort4.log" in the ks4 output folder (the one that contains all the .npy files). This function call is missing from the SpikeInterface wrapper. 

Including this function call would make debugging various issues more convenient. Would it be possible to do so?

@h-mayorquin @alejoe91 @zm711 @cheydrick

zm711, 2025-03-18:
I think we thought this just required setting the verbose flag. (see #3768) I'm fine with having the `setup_logger()` in our wrapper. But I would prefer Alessio to comment on if he sees any problems--we had discussed making our own logger at some point, but never did it, so I'm never sure how loggers would play together.

nikhilchandra, 2025-03-18:
I see, so we just need to add the verbose flag to our SpikeInterface API call. Got it, thanks!

zm711, 2025-03-18:
Try that and see what happens. Alessio tweaked it recently. If it still doesn't work then we might need to play more with the logging.

cheydrick, 2025-03-21:
I'm hoping to have an opportunity to test this by early next week.

cheydrick, 2025-03-25:
Setting verbose=True in the call to run_sorter() does not result in Kilosort 4 generating a .log file.

cheydrick, 2025-03-25:
To be clear, setting verbose=True in run_sorter() does not result in Kilosort 4 including a file called kilosort4.log in the output folder (the one where all the .npy files get generated).

Passing verboe=True does cause Kilosort to emit more information to stdout.

closed 2025-05-06 by alejoe91