SpikeInterface/spikeinterface #3783: [Question] Annotating sorted spikes with channel positions

lcui809, 2025-03-18:
Hi there,

I have a list of sorted spikes and would like to annotate each spike with the channel position where it was originally generated. I have already computed the spike locations using sorting_analyzer, which has been very useful. However, after considerable effort, I still haven't figured out how to link each sorted spike to its source channel position.

I am using an HD-MEA from Maxwell. Any guidance would be greatly appreciated!

chrishalcrow, 2025-03-19:
Hi @lcui809, I thought we didn't have a method for this, but digging into the code, we do! It's not documented, so I'll add some documentation soon.

Turns out you can pass the `peak_channel` method when you compute spike locations. This will give you the location of the channel with the peak waveform on it. If you have a `sorting_analyzer` called `sa`, you can do this like so:

``` python
sa.compute({'spike_locations': {'method': 'peak_channel'}})
print(sa.get_extension('spike_locations').get_data())
>>> array([( 0., 20.), (20., 20.), ( 0.,  0.), ..., ( 0.,  0.), ( 0., 20.),
       ( 0., 20.)], dtype=[('x', '<f8'), ('y', '<f8')])
```

Hope that works!

zm711, 2025-03-19:
My deeper question is why do you care about the channel itself? One of the benefits of the fancier methods is to estimate the "true" position rather than just find the nearest channel which could be 10s of microns away. Is there a need for channel vs "estimated" true location which is likely more accurate unless a neuron is right on top of a channel?

yger, 2025-03-19:
Note sure to follow here. Internally, in SI, peaks are attached to a channel (because we need this information to extract sparse mask and do efficient computations within components), and spikes are not. However, given spikes, one can estimate their putative positions depending on various methods (grid/monopolar/barycenter), and I think this is more relevant at the end of the pipelines than working at the channel level. Otherwise, as @chrishalcrow found, we can get the spike locations at the channel levels, and not with any fancy inference scheme

lcui809, 2025-03-20:
Thanks, everyone, for answering my question — your insights are extremely helpful!

@chrishalcrow  Hi Chris, thanks again for resolving my question. I have a follow-up if I may.
Would it be possible to extract waveform features (e.g., peak-to-valley) or waveform metrics for all spikes? We're considering some downstream analyses, such as grouping spikes by frequency and extracting waveform features from different subgroups within the same unit. If this is feasible, could you please point me to the relevant NPY file where these metrics can be found?

chrishalcrow, 2025-03-21:
Hello - at the moment, we don't provide this functionality. So you'd need to make something custom. But we do have a framework which you could copy, although it is a bit complicated. If I were to try this, I would copy the `ComputeSpikeAmplitudes` class, and make something similar:

https://github.com/SpikeInterface/spikeinterface/blob/b31646140af4e0fd496b29cbb9a62b5ffa3c7313/src/spikeinterface/postprocessing/spike_amplitudes.py#L15

alejoe91, 2025-03-21:
@chrishalcrow let me jump in! We have a much more flexible framework for this in the [`sortingcomponents.features_from_peaks.py`](https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/sortingcomponents/features_from_peaks.py)

These are all PipelineNodes, and here's how you would compute the required features for all spikes:
```
from spikeinterface.sortingcomponents.features_from_peaks import compute_features_from_peaks

# analyzer is your analyzer
# we need to convert spikes to peaks
extremum_channel_inds = analyzer.get_template_extremum_channel(output="index")
spike_peaks = analyzer.sorting.to_spike_vector(extremum_channel_inds=extremum_channel_inds)
# this is a trick to assign a channel index to each spike

feature_list = [
    "amplitude",
    "ptp",
    "center_of_mass",
]
feature_params = {
    "amplitude": {"all_channels": False, "peak_sign": "neg"},
    "ptp": {"all_channels": False},
    "center_of_mass": {"radius_um": 120.0},
}
features = compute_features_from_peaks(recording, spike_peaks, feature_list, feature_params=feature_params, **job_kwargs)
```

It should be quite easy to implement other features, like peak-to-valley/halfwidth etc. @lcui809 do you want to give it a try?

However, these might be very noisy for individual spikes.

lcui809, 2025-03-23:
@alejoe91 @chrishalcrow 
Thank you so much for your guidance—it was extremely helpful and saved me a lot of time in building the pipeline code. I’ve incorporated Alessio’s suggestions, and the method now works perfectly. I’ve included my modified code below in case it’s useful to others. The calculations for 'peak_to_valley', 'peak_trough_ratio', 'half_width', 'repolarization_slope', and 'recovery_slope' have been added to features_from_peaks.py.

I may have one final question: as the next step, I’d like to group the spikes from a single unit, either by time (e.g., within a network burst or an interval) or by both time and frequency (e.g., high-frequency vs low-frequency spikes within the same burst). After grouping, I aim to average the waveforms within each group and compute the features of the averaged waveform.

Could you please recommend a good approach for isolating and averaging the waveforms of selected spikes?

from spikeinterface.core.node_pipeline import PipelineNode
import numpy as np
import scipy.stats

class PeakToValleyFeature(PipelineNode):
    def __init__(self, recording, name="peak_to_valley_feature", return_output=True, parents=None):
        PipelineNode.__init__(self, recording, return_output=return_output, parents=parents)
        self._dtype = np.float32

    def get_dtype(self):
        return self._dtype

    def compute(self, traces, peaks, waveforms):
        ptv = []
        for wf in waveforms:
            wf1d = wf[:, 0] if wf.ndim == 2 else wf
            trough_idx = np.argmin(wf1d)
            peak_idx = trough_idx + np.argmax(wf1d[trough_idx:])
            ptv.append((peak_idx - trough_idx) / self.recording.sampling_frequency)
        return np.array(ptv, dtype=self._dtype)

class PeakTroughRatioFeature(PipelineNode):
    def __init__(self, recording, name="peak_trough_ratio_feature", return_output=True, parents=None):
        PipelineNode.__init__(self, recording, return_output=return_output, parents=parents)
        self._dtype = np.float32

    def get_dtype(self):
        return self._dtype

    def compute(self, traces, peaks, waveforms):
        ptr = []
        for wf in waveforms:
            wf1d = wf[:, 0] if wf.ndim == 2 else wf
            trough_idx = np.argmin(wf1d)
            peak_idx = trough_idx + np.argmax(wf1d[trough_idx:])
            ptr.append(wf1d[peak_idx] / wf1d[trough_idx])
        return np.array(ptr, dtype=self._dtype)

class HalfWidthFeature(PipelineNode):
    def __init__(self, recording, name="half_width_feature", return_output=True, parents=None):
        PipelineNode.__init__(self, recording, return_output=return_output, parents=parents)
        self._dtype = np.float32

    def get_dtype(self):
        return self._dtype

    def compute(self, traces, peaks, waveforms):
        hw = []
        for wf in waveforms:
            wf1d = wf[:, 0] if wf.ndim == 2 else wf
            trough_idx = np.argmin(wf1d)
            trough_val = wf1d[trough_idx]
            threshold = 0.5 * trough_val
            pre = np.where(wf1d[:trough_idx] < threshold)[0]
            post = np.where(wf1d[trough_idx:] < threshold)[0]
            if len(pre) == 0 or len(post) == 0:
                hw.append(np.nan)
                continue
            cross_pre = pre[0] - 1
            cross_post = post[-1] + 1 + trough_idx
            hw.append((cross_post - cross_pre) / self.recording.sampling_frequency)
        return np.array(hw, dtype=self._dtype)

class RepolarizationSlopeFeature(PipelineNode):
    def __init__(self, recording, name="repolarization_slope_feature", return_output=True, parents=None):
        PipelineNode.__init__(self, recording, return_output=return_output, parents=parents)
        self._dtype = np.float32

    def get_dtype(self):
        return self._dtype

    def compute(self, traces, peaks, waveforms):
        slopes = []
        for wf in waveforms:
            wf1d = wf[:, 0] if wf.ndim == 2 else wf
            trough_idx = np.argmin(wf1d)
            times = np.arange(len(wf1d)) / self.recording.sampling_frequency
            return_idx = np.where(wf1d[trough_idx:] >= 0)[0]
            if len(return_idx) == 0:
                slopes.append(np.nan)
                continue
            end_idx = return_idx[0] + trough_idx
            if end_idx - trough_idx < 3:
                slopes.append(np.nan)
                continue
            res = scipy.stats.linregress(times[trough_idx:end_idx], wf1d[trough_idx:end_idx])
            slopes.append(res.slope)
        return np.array(slopes, dtype=self._dtype)

class RecoverySlopeFeature(PipelineNode):
    def __init__(self, recording, name="recovery_slope_feature", return_output=True, parents=None, recovery_window_ms=0.7):
        PipelineNode.__init__(self, recording, return_output=return_output, parents=parents)
        self.recovery_window_ms = recovery_window_ms
        self._dtype = np.float32

    def get_dtype(self):
        return self._dtype

    def compute(self, traces, peaks, waveforms):
        slopes = []
        for wf in waveforms:
            wf1d = wf[:, 0] if wf.ndim == 2 else wf
            peak_idx = np.argmax(wf1d)
            times = np.arange(len(wf1d)) / self.recording.sampling_frequency
            max_idx = int(peak_idx + (self.recovery_window_ms / 1000) * self.recording.sampling_frequency)
            max_idx = min(max_idx, len(wf1d))
            res = scipy.stats.linregress(times[peak_idx:max_idx], wf1d[peak_idx:max_idx])
            slopes.append(res.slope)
        return np.array(slopes, dtype=self._dtype)

_features_class.update({
    "peak_to_valley": PeakToValleyFeature,
    "peak_trough_ratio": PeakTroughRatioFeature,
    "half_width": HalfWidthFeature,
    "repolarization_slope": RepolarizationSlopeFeature,
    "recovery_slope": RecoverySlopeFeature,
})

zm711, 2025-03-26:
I think part of the team is way for a conference. I would ping this issue again in a few days :)

lcui809, 2025-03-26:
@zm711 Thanks a lot for letting me know. I have find the answer I need elsewhere. 

Please

closed 2025-03-26 by lcui809