SpikeInterface/spikeinterface #3789: What is the param "all_pcs" in sqm.pca_metrics.mahalanobis_metrics ?

BEGINRX, 2025-03-21:
I want to compute "isolation_distance" and "l_ratio". When I use the example `iso_distance, l_ratio = sqm.pca_metrics.mahalanobis_metrics(all_pcs=all_pcs, all_labels=all_labels, this_unit_id=0)
`
I don't know what the all_pcs is and how I can get it.

When I use
`qm_ext = sorting_analyzer.compute(input={"principal_components": dict(n_components=5),
                                "quality_metrics": dict(skip_pc_metrics=False)})
matrix = ext_pca.get_data()
`
the matrix is 3D. And I can see the 
`_possible_pc_metric_names = [
    "isolation_distance",
    "l_ratio",
    "d_prime",
    "nearest_neighbor",
    "nn_isolation",
    "nn_noise_overlap",
    "silhouette",
]` in pca_metrics.py. But I can't get the values of "isolation_distance" and "l_ratio". 
So how can I get the values of "isolation_distance" and "l_ratio" of all units?

alejoe91, 2025-03-21:
Hi @BEGINRX 

You need to get the `quality_metrics` extension, which returns a data frame, with the unit ids as indices and the quality metrics as columns.
Then you'll see the different metrics as columns:
```
ext_qm = sorting_analyzer.get_extension("quality_metrics")
qm_df = ext_qm.get_data()

display(qm_df)
print(qm_df["l_ratio"])
```

BEGINRX, 2025-03-21:
Should I update the version of spikeinterface? There is no "l_ratio" in qm_df.

[image]

alejoe91, 2025-03-21:
I see you have no PC metrics. Can you print this?
```
sorting_analyzer.get_extension("principal_components")
```

alejoe91, 2025-03-21:
Note, you need to compute quality metrics either at the same time or after computing PCA in order to get PCA metrics. If you first ocmputed wuality metrics and then principal_components, then the PCA metrics will not be available.

BEGINRX, 2025-03-21:
Thank you for your help! I got the 'isolation_distance' and 'l_ratio' after compute PCA. `sorting_analyzer.get_extension()` is useful! Thank you!
But I still wonder what the `all_pcs` is?

alejoe91, 2025-03-21:
The `sqm.pca_metrics.mahalanobis_metrics` is kind of internal :)

`all_pcs` is a flattened version of PC projections for unit neighbors, as defined here: https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/qualitymetrics/pca_metrics.py#L160C14-L164C49

BEGINRX, 2025-03-21:
I see. Thank you!

alejoe91, 2025-03-21:
I'll close the issue :)

closed 2025-03-21 by alejoe91