SpikeInterface/spikeinterface #3790: correcting motion/drift in 3D

chiyu1203, 2025-03-21:
Dear Spikeinterface community,
I use 128-channel 4-shank-3D probes to do acute (typically 1-2 hour) recording in head-fixed, walking insects. Each shank (60um width) is 150 um along anterior-posterior axis and/or 300 um medial-lateral axis away from other shanks, and embedded with 32 channels for 3 columns (the interval between columns is 18.5um) spanning around 315 um along the dorsal-ventral axis.

[image]

Following the [Handle motion/drift guide](https://spikeinterface.readthedocs.io/en/latest/how_to/handle_drift.html), I managed to detect peaks and estimate motion/drift in different methods. Since the span of the channels are much shorter than Neuropixels, I first tested different length of win_step_um and of win_scale_um (from 25um to 150 um). Then I focus on the analysis with DREDge because (a) It seems that DREDge and its ancestor detect more peaks than others 

[image]

(b) After viewing @cwindolf [wonderful talk](https://youtu.be/RYKHoipT-2A?si=aiw2GzXeqsCfrs3m) and the [paper](https://www.nature.com/articles/s41592-025-02614-5), I feel confident that this method would work well with short probes (imaging these 3D probes as a multi-shank Neuropixels UHD but with NP1-Like geometry)

[image]

However, after DREDge detected those peaks, it does not look like those drift were corrected. I wonder that is because (a) the drift is too small to correct or the correction is too small to view (b) the drift occurred at the 3rd dimension, which is not detected in the function?! If (b) is the case, is there anything I can help develop this new feature? or should I split the stacked probes into 2 portion and analyse them independently?

Below is the code I use to call the function. Feel free to let me know if you have any thought.
```
recording_corrected, motion, motion_info = spre.correct_motion(recording=recording_saved, preset=preset, folder=folder,overwrite=False, output_motion=True, output_motion_info=True, estimate_motion_kwargs={'win_step_um':win_um,'win_scale_um':win_um},**job_kwargs)
```

chiyu1203, 2025-04-03:
Hi!!! Thank you so much for the reply. Sorry, I only found out the "analyse motion shank-by-shank" idea in this issue #3458  after I post that question. And thank you for pointing out changing thresholds! I am doing the analysis again and will update what I get soon.

cwindolf, 2025-04-22:
Hi @chiyu1203 , that's really interesting, thank you for sharing! Shank 3 in particular is cool to look at -- it seems like different units are moving in different directions at t~1750, which is not something that can really be modeled by a motion field that any of these algorithms consider. Good luck with the longer probes, and feel free to reach out if you have further questions or issues.

closed 2025-04-22 by cwindolf

chiyu1203, 2025-04-23:
No problem! Thanks for documenting those parameters in the first place!
I think that t~1750 drift happened as tissue dried out, the shank shifted both horizontal and vertically. This is why I can imagine the difficulty about 3D motion estimation with these kind of multi-shank probes. On one hand, these probes are made flexible so that they do not break easily when the tissue move. On the other hand, this flexibility makes it difficult to estimate the drift/motion across shanks as the physical distance among shanks is never fixed.