SpikeInterface/spikeinterface #3796: about sortingview

bxy666666, 2025-03-23:
w2 = sw.plot_sorting_summary(sorting_analyzer, display=False, curation=True, backend="sortingview")
    plt.savefig(os.path.join(data_location, f'sorting_summary_{suffix}.png'))
    # 查看 SortingSummaryWidget 对象的所有属性和方法
    print(dir(w2))

    # 假设 handle_display_and_url 是用于生成和处理 URL 的函数
    # 生成 URL (这里假设 handle_display_and_url 是正确实现的)
    url = w2.url   # 确保你能从 w2 中获取 URL

    # 保存 URL 到文本文件
    url_file_path = os.path.join(data_location, f'plot_url_{suffix}.txt')
    with open(url_file_path, 'w') as f:
        f.write(f"URL: {url}\n")

    print(f"URL saved to {url_file_path}")                                                    
This is how 

[image]

I used before, but now I compeleted without this, How can I get url = w2.url with exited results?

alejoe91, 2025-03-26:
Hi @bxy666666 

There was a change in how sortingview works, so you need to change your login access. See a guide here: https://github.com/magland/sortingview

bxy666666, 2025-03-28:
Thanks , But now sortingview returns Internal Server Error. How can I solve this problem?

magland, 2025-03-28:
@bxy666666 could you share some more details about the error. Do you have a link?

magland, 2025-03-28:
@bxy666666 I found the problem and I think it should be fixed now.

ohkoibln, 2025-03-29:
> [@bxy666666](https://github.com/bxy666666) could you share some more details about the error. Do you have a link?
Yes, I got the link . But it seems to have no error now, thanks a lot for your helpful reply!

bxy666666, 2025-03-29:
> [@bxy666666](https://github.com/bxy666666) could you share some more details about the error. Do you have a link?

Yes, I got the link . But it seems to have no error now, thanks a lot for your helpful reply!

zm711, 2025-03-30:
Since this is fixed I'll close this issue! Thanks Jeremy!

closed 2025-03-30 by zm711