SpikeInterface/spikeinterface #3808: .ncs File Readability

DanielPerodin, 2025-03-26:
Hello,
I'm trying to use SpikeInterface to process neural data from Neuralynx .ncs files. For some reason the program does not recognize these .ncs files, giving an error that basically says it cannot find the files in the directory. I have checked that the directory was correct, and the files are not corrupted because they work in a different MATLAB program. Has anyone encountered any similar issues with .ncs files or heard of such problems? Many thanks.

zm711, 2025-03-26:
Hey @DanielPerodin, our neuralynx reading is done through Neo. Could we have you open an issue there? neuralynx processing is pretty complicated with different version etc. I would say we could trouble shoot over on [neo](https://github.com/NeuralEnsemble/python-neo/issues) so that we can figure out if it is a format that we don't yet support or if we need you to tweak how you're attempting to load. It would also be helpful if you post here or (ideally on Neo) your script so we see how you're trying to access this data!

DanielPerodin, 2025-03-26:
Hey @zm71, I will certainly open a an issue on Neo tomorrow with the script. Many thanks, we'll talk soon.

DanielPerodin, 2025-03-26:
> > Hey [@ZM71](https://github.com/ZM71), I will certainly open a an issue on Neo tomorrow with the script. Many thanks, we'll talk soon.
> 
> ?

sorry, the username I had to @ set for was missing a 1

zm711, 2025-04-11:
We will release a new Neo version within a day or two (Monday at latest) so we can close this :) The fix is in Neo 0.14.1.

closed 2025-04-11 by zm711