SpikeInterface/spikeinterface #3809: Duplicate spikes after merging units is not resolved

M-Hisham, 2025-03-26:
Hello,

I always run the function remove_duplicated_spikes() after sorting, however I think it only removes close spikes within the same unit (please correct me if wrong). 

The issue is when I merge two units, duplicated spikes are not resolved add artificial violations. the only I was able to fix is running  remove_duplicated_spikes() again after merging, and create a new spike Analyzer. 

is there a way I can avoid creating a new spike Analyzer, and only updates its sorting_base. I wonder if there is a better practice for this issue.

Thanks

zm711, 2025-03-26:
Are you using the `censor_ms` argument during merging with the sorting_analyzer? We can see in the docstring that setting a censor time should remove duplicated spikes within a merge
https://github.com/SpikeInterface/spikeinterface/blob/165035a8f73a436fc3a924bf60a456cfc85745e6/src/spikeinterface/core/sortinganalyzer.py#L1169-L1170

If you're trying that and it is not working then maybe we have a bug so we would need to troubleshoot. But I would start with that :)

M-Hisham, 2025-03-26:
Thank you very much. I didn't actually. this actually fix the problem

zm711, 2025-03-27:
Great! I'll close this then :)

closed 2025-03-27 by zm711