SpikeInterface/spikeinterface #3813: Spikeinterface to phy2 channel index question

Deborah-Chen, 2025-03-27:
Hi, I am running into an issue with channel indexing in phy2. We're using the Cambridge ASSY-156-F probe with 6 shanks and 64 channels, and sorting with KS4. I excluded dead channels with recording.remove_channels(). After opening the sorted results in phy2, I noticed that the channel numbers have been reordered as 0 to index-1. 

I understand that phy displays relative channel indices, but I'm unsure how to restore the original channel mapping. Is there a parameter in export_to_phy() that allows me to retain and display the original channel indexing in phy2?

My current parameters are
export_to_phy(sorting_analyzer=sorting_analyzer, 
              output_folder=output_folder / 'sorter_output' / 'phy_KS4', 
              remove_if_exists= True, 
              **job_kwargs)

Please let me know if I should provide more information. Thank you in advance! 

[image]

zm711, 2025-03-27:
I don't believe so. You could definitely find the extremum_channel for your units and then add that as a property on your sorting which should be written out to the `tsv` that phy uses to make clusterview. Maybe @alejoe91 has an easy script to do that around. I think @jakeswann1 may have also done something like this for some of his work? You've added columns for phy before right?

zm711, 2025-03-30:
@Deborah-Chen have you managed to try this and confirm it works?

Deborah-Chen, 2025-03-31:
@zm711 @jakeswann1 That worked perfectly, thank you both for your prompt response and help!

zm711, 2025-04-05:
Awesome. Feel free to open new issues if they come up!

closed 2025-04-05 by zm711

harshk95, 2025-07-20:
Hi, I just wanted to confirm somethings here. So, I am using an NPX2.0 probe and sorting by group with Kilosort4 in Spikeinterface. The 'original' channel number of the units are replaced upon export_to_phy, presumably, to new channel numbers which seem to be monotonically increasing as one moves through the shanks. I has already had the extremum channel appended to my properties and can confirm that the location seems to match the extremum channel, but of course there is a mismatch of the channel number in cluster_info.tsv and this extremum channel. 
I believe the same is the case for the export_to_IBLgui.
Thanks for the clarification.