SpikeInterface/spikeinterface #3816: `remove_channels` should either warn or raise an error when giving non-present channels

zm711, 2025-03-28:
Carry over from #3814.

### MRE
```python
>>> import spikeinterface.full as si
>>> recording , _ si.generate_ground_truth_recording([100], num_channels=64)
>>> rec = recording.remove_channels(1)
>>> print(rec.get_num_channels())
64

```
This will sometimes silently sometimes with a numpy warning do nothing. This is because the dtype of the channels is string, but I'm trying to remove an int. 

### Solution
I would argue that we should flat out error since it is too easy to ignore a warning if someone gives channels that can't be removed (either due to dtype or not existing). But at minimum we should warn the user that they are doing it wrong since nothing is happening.

@chrishalcrow.

alejoe91, 2025-03-29:
Thanks @zm711 

I agree this should raise an error!

chrishalcrow, 2025-03-30:
Agree on error: this should only be a problem with a user is inputting the channels to remove by hand (in contrast to using e.g. `detect_bad_channels` which will get the ids from the data), so I'd defo wanna be told firmly if I had gotten the ids wrong!

zm711, 2025-03-30:
Yep which I do all the times (although since I know why channels are strings I haven't made this mistake in the past. I just forgot whether our simulated was str or float). So I can imagine some users knowing they have bad channels and wanting to use it themselves. I actually think our wiring mechanism for when users know they have bad channels is a little too opaque and undocumented. Someone in my lab needed it once and it was a bit of a struggle to do honestly. I'll try to improve that docstring myself!

samuelgarcia, 2025-04-01:
+1 for raising error.

closed 2025-04-01 by alejoe91